JMRI/JMRI #810: Develop Release 4.3.4

bobjacobsen, 2016-02-15:
Issue for discussing release 4.3.4 contents, timing, etc.

mattharris, 2016-03-07:
We will be building this release this coming weekend.

Please could everyone verify and check the release note and update if required.

Also, please take a scan through your merged pull requests and make sure that they are assigned to the 4.3.4 milestone if not already done so.

Thanks.

mattharris, 2016-03-07:
I'll probably start the build process late afternoon/early evening Central European Time (CET) on Saturday.

mattharris, 2016-03-12:
It will now be tomorrow for starting with the build.

mattharris, 2016-03-13:
The release-4.3.4 branch has been created. 

From now on, please document your changes in the [jmri4.3.5.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.3.5.shtml) release note file.

Maintainers, please set the 4.3.5 milestone on pulls from now on.

Jenkins will be creating files shortly at [the secondary location](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.3.4/).

mattharris, 2016-03-13:
As may be clear, we're trying out the secondary Jenkins server for this build.

mattharris, 2016-03-14:
Closing as release has now been made

closed 2016-03-14 by mattharris