JMRI/JMRI #814: Best Programming Mode problem

mattharris, 2016-02-16:
Based on a jmriusers report, the programming mode default changes introduced in #541 seem to have a problem with the NCE-USB.

```
From: "Brian Gilhuly" <brianlgilhuly@...>

When I installed 4.3.3 on my RPi2 running updated Raspbian Jessie, it hangs
on start-up. All begins well; WiThrottle server starts, panel is loaded,
start-up script runs. Then nothing. The Panel Pro window never opens and the
java icon remains on its window tab.

If I re-install 4.3.2, everything works as normal.

Brian

Here is the console log:

2016-02-16 11:19:27,156 util.Log4JUtil INFO - This log is appended to file: /home/pi/.jmri/log/messages.log [main]
2016-02-16 11:19:27,164 util.Log4JUtil INFO - This log is stored in file: /home/pi/.jmri/log/session.log [main]
2016-02-16 11:19:27,227 apps.Apps INFO - PanelPro version 4.3.3-R965a469 starts under Java 1.8.0 at Tue Feb 16 11:19:27 EST 2016 [main]
2016-02-16 11:19:33,579 apps.Apps INFO - Starting with profile Welsh_Sidings.3dfcf14d [main]
2016-02-16 11:19:34,835 node.NodeIdentity INFO - Using jmri-000F600413E7-3dfcf14d as the JMRI Node identity [main]
2016-02-16 11:19:36,046 usbdriver.UsbDriverAdapter INFO - NCE USB /dev/ttyUSB0 port opened at 19200 baud [main]
2016-02-16 11:19:36,244 nce.NceConnectionStatus INFO - NCE EPROM revision = 7.3.1 [AWT-EventQueue-0]
2016-02-16 11:19:36,372 nce.NceProgrammer WARN - NCE USB-SB3/SB5/TWIN getSupportedModes returns no modes [main]
2016-02-16 11:19:38,470 configurexml.PerformFileModelXml INFO - Load file /home/pi/Documents/JMRI/Welsh Sidings.xml [main]
2016-02-16 11:19:43,397 configurexml.PerformScriptModelXml INFO - Run file /home/pi/Documents/JMRI/jython/jmri_defaults.py [main]
ScriptEngineManager providers.next(): javax.script.ScriptEngineFactory: Provider apple.applescript.AppleScriptEngineFactory not found
2016-02-16 11:19:43,536 script.JmriScriptEngineManager INFO - python 2.7 is provided by jython 2.7.0 [main]
2016-02-16 11:19:43,573 script.JmriScriptEngineManager INFO - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0 [main]
2016-02-16 11:19:46,652 withrottle.UserInterface INFO - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleGUIServer]
2016-02-16 11:20:18,198 configurexml.PerformScriptModelXml INFO - Run file /home/pi/Documents/JMRI/jython/Layout Startup.py [main]
2016-02-16 11:20:18,622 nce.NceProgrammer WARN - NCE USB-SB3/SB5/TWIN getSupportedModes returns no modes [main]
2016-02-16 11:20:18,632 nce.NceProgrammer WARN - NCE USB-SB3/SB5/TWIN getSupportedModes returns no modes [main]
2016-02-16 11:20:18,638 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception: java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
  at java.util.ArrayList.rangeCheck(ArrayList.java:638)
  at java.util.ArrayList.get(ArrayList.java:414)
  at jmri.jmrix.AbstractProgrammer.getBestMode(AbstractProgrammer.java:161)
  at jmri.jmrix.AbstractProgrammer.getMode(AbstractProgrammer.java:166)
  at jmri.jmrit.progsupport.ProgServiceModeComboBox.programmerSelected(ProgServiceModeComboBox.java:145)
  at jmri.jmrit.progsupport.ProgServiceModeComboBox.<init>(ProgServiceModeComboBox.java:128)
  at jmri.jmrit.progsupport.ProgServiceModeComboBox.<init>(ProgServiceModeComboBox.java:80)
  at jmri.jmrit.symbolicprog.tabbedframe.PaneProgAction.<init>(PaneProgAction.java:47)
  at jmri.jmrit.ToolsMenu.<init>(ToolsMenu.java:49)
  at jmri.jmrit.ToolsMenu.<init>(ToolsMenu.java:31)
  at apps.Apps.toolsMenu(Apps.java:666)
  at apps.Apps.createMenus(Apps.java:566)
  at apps.Apps.<init>(Apps.java:301)
  at apps.PanelPro.PanelPro.<init>(PanelPro.java:48)
  at apps.PanelPro.PanelPro.main(PanelPro.java:110)
[main]
2016-02-16 11:20:19,146 server.WebServer INFO - Starting Web Server on port 12080 [WebServer]
2016-02-16 11:20:19,459 server.WebServer INFO - Starting ZeroConfService _http._tcp.local for Web Server with properties {path=/, json=3.0} [WebServer]
```

mattharris, 2016-02-16:
Original message:

https://groups.yahoo.com/neo/groups/jmriusers/conversations/messages/125871

mattharris, 2016-02-16:
#816 has a first-pass at a fix for this. Now I don't have actual NCE hardware to check this on but have been able to get past the NPE with a dummy connection.

mattharris, 2016-02-16:
#816 now merged but this still needs more work.

dheap, 2016-02-16:
I'm not getting far with this one.

What I have found so far:
- Programming Track Mode is being enabled for the non Power Cab USB
connections and it should not be.
- It appears NceProgrammerManager.isGlobalProgrammerAvailable() should
control the visibility of Programming Track Mode and is not being invoked
early enough (actually rarely).
- Changing the return value from this one has no effect.

I compared SPROG Command Station mode because I have one and am somewhat
familiar with the code:
- SprogProgrammerManager.isGlobalProgrammerAvailable() is controlling the
visibility of Programming Track Mode and is being invoked as soon as the
connection is opened.
- Changing the return value from this one has exactly the expected effect.

But I'm having trouble tracing what is going wrong. Could someone else
please look at it?

Dave

KenC57, 2016-02-17:
Comments:

1.      Didn't know we had a NONE available for that, which then makes
sense.

2.      Looking at lines the rest of that file, I believe the PowerCab
should be returning DIRECT also, but it looks to me like only the 2006 is
returning with DIRECT.

3.      I think the 2004 also supported DIRECT, but would need somebody else
to concur on this.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

dheap, 2016-02-17:
The NONE is a good idea, but it doesn't solve the problem of why
NceProgrammerManager.isGlobalProgrammerAvailable() is not being invoked to
hide programming Track mode.

I deliberately introduced an error into both
NceProgrammerManager.isGlobalProgrammerAvailable() &
SprogProgrammerManager.isGlobalProgrammerAvailable() to get a trace and
compare the two.

As you will see, in the SPROG case isGlobalProgrammerAvailable() is
invoked just after the serial port is opened, while in the SB5 case,
isGlobalProgrammerAvailable() is not invoked, even though I opened a
roster entry in program Track mode. It wasn't invoked until I tried to
open Preferences. No wonder Program Track mode is not being hidden in the
Roster Frame.

Others more skilled at debugging than this novice may shed some light on
this.

Dave

2016-02-17 14:51:32,559 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-02-17 14:51:34,660 util.Log4JUtil                        INFO  - This log is appended to file: /Users/heap/Library/Preferences/JMRI/log/messages.log [main]
2016-02-17 14:51:34,662 util.Log4JUtil                        INFO  - This log is stored in file: /Users/heap/Library/Preferences/JMRI/log/session.log [main]
2016-02-17 14:51:34,669 apps.AppsBase                         INFO  - DecoderPro version 4.3.4ish-201602170338-heap-R5546656 starts under Java 1.8.0_71 at Wed Feb 17 14:51:34 AEDT 2016 [main]
2016-02-17 14:51:35,165 gui3.Apps3                            INFO  - Starting with profile SPROG_Command_Station.3e694e6c [main]
2016-02-17 14:51:35,500 node.NodeIdentity                     INFO  - Using jmri-70CD60AACEA6-3e694e6c as the JMRI Node identity [main]
2016-02-17 14:51:35,664 serialdriver.SerialDriverAdapter      INFO  - /dev/cu.usbmodemfd5431 port opened at 9600 baud, sees  DTR: true RTS: true DSR: false CTS: true  CD: false [main]
2016-02-17 14:51:35,678 jmrix.ConnectionConfigManager         ERROR - Unable to load [Element: <connection/>] into jmri.jmrix.sprog.sprogCS.configurexml.ConnectionConfigXml [main]
java.lang.ArrayIndexOutOfBoundsException: 10
	at jmri.jmrix.sprog.SprogProgrammerManager.isGlobalProgrammerAvailable(SprogProgrammerManager.java:52)
	at jmri.InstanceManager.setProgrammerManager(InstanceManager.java:630)
	at jmri.jmrix.sprog.SprogSystemConnectionMemo.configureManagers(SprogSystemConnectionMemo.java:175)
	at jmri.jmrix.sprog.sprogCS.SprogCSSerialDriverAdapter.configure(SprogCSSerialDriverAdapter.java:42)
	at jmri.jmrix.configurexml.AbstractSerialConnectionConfigXml.load(AbstractSerialConnectionConfigXml.java:110)
	at jmri.jmrix.ConnectionConfigManager.initialize(ConnectionConfigManager.java:77)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:259)
	at jmri.implementation.JmriConfigurationManager.lambda$initializeProvider$79(JmriConfigurationManager.java:255)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:254)
	at jmri.implementation.JmriConfigurationManager.lambda$initializeProvider$79(JmriConfigurationManager.java:255)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:254)
	at jmri.implementation.JmriConfigurationManager.lambda$load$78(JmriConfigurationManager.java:188)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:187)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:176)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:166)
	at apps.AppsBase.setAndLoadPreferenceFile(AppsBase.java:272)
	at apps.gui3.Apps3.setAndLoadPreferenceFile(Apps3.java:368)
	at apps.AppsBase.<init>(AppsBase.java:104)
	at apps.gui3.Apps3.<init>(Apps3.java:87)
	at apps.gui3.dp3.DecoderPro3.<init>(DecoderPro3.java:36)
	at apps.gui3.dp3.DecoderPro3.main(DecoderPro3.java:110)
2016-02-17 14:51:36,543 util.FileUtilSupport                  INFO  - File path program: is /Users/heap/jmri/ [main]
2016-02-17 14:51:36,546 util.FileUtilSupport                  INFO  - File path preference: is /Ours/Our Documents/Dave/Dropbox/JMRI/ [main]
2016-02-17 14:51:36,550 util.FileUtilSupport                  INFO  - File path profile: is /Users/heap/Library/Preferences/JMRI/SPROG_Command_Station/ [main]
2016-02-17 14:51:36,550 util.FileUtilSupport                  INFO  - File path settings: is /Users/heap/Library/Preferences/JMRI/ [main]
2016-02-17 14:51:36,550 util.FileUtilSupport                  INFO  - File path home: is /Users/heap/ [main]
2016-02-17 14:51:36,551 util.FileUtilSupport                  INFO  - File path scripts: is /Users/heap/jmri/jython/ [main]
2016-02-17 14:51:37,149 powerpanel.PowerPane                  WARN  - No power manager instance found, panel not active [main]
2016-02-17 14:51:37,150 throttle.PowerManagerButton           INFO  - No power manager instance found, panel not active [main]
[... cut ...]
	at java.lang.Class.newInstance(Class.java:442)
	at apps.gui3.TabbedPreferences.init(TabbedPreferences.java:161)
	at apps.gui3.TabbedPreferencesAction$2.run(TabbedPreferencesAction.java:98)
	at java.lang.Thread.run(Thread.java:745)
 [Thread-8]
2016-02-17 14:58:47,880 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: AA consecutive timeouts = 6 [Transmit]
2016-02-17 14:58:47,981 nce.NceConnectionStatus               WARN  - No response from NCE command station [Transmit]

dheap, 2016-02-17:
Sorry, my email attachments ended up in-line. Attached below

[SPROG-CS.txt](https://github.com/JMRI/JMRI/files/133666/SPROG-CS.txt)
[SB5.txt](https://github.com/JMRI/JMRI/files/133668/SB5.txt)

dheap, 2016-02-17:
The relevance of all this is that if you:
- Force the return value of
SprogProgrammerManager.isGlobalProgrammerAvailable() to true, the
Programming Track options appear in the Roster Frame.
- Force the return value of
SprogProgrammerManager.isGlobalProgrammerAvailable() to false, the
Programming Track options disappear from the Roster Frame.
This is desirable and correct behaviour.

However, if you:
- Force the return value of
NceProgrammerManager.isGlobalProgrammerAvailable() to true, the
Programming Track options appear in the Roster Frame.
- Force the return value of
NceProgrammerManager.isGlobalProgrammerAvailable() to false, the
Programming Track options still appear in the Roster Frame.
This is undesirable and incorrect behaviour.

Dave

bobjacobsen, 2016-02-17:
Sorry to just be coming to this. Would like to write a test for this, but not sure exactly where the problem lies.  Is this the failure:
*) PanelPro configured to NCE; NCE USB; (dummy port); V6.x.x; SB3
*) Save, restart

But then I can't get to "Service Mode Programmer" to get it to fail; that's greyed out in the Programmers menu. Alternately, if I open "Roster" (the DP3 main window), there's no service mode programming option.

How do you get to the failure?

mattharris, 2016-02-17:
@bobjacobsen If with the current HEAD, then it won't fail as my fix in #816 simply adds an entry `DefaultProgrammerManager.NONE` to the list returned by `getSupportedModes()` for this combination.

Without this addition, `getSupportedModes()` returned an empty list meaning that `getBestMode()` in `jmrix.AbstractProgrammer` would throw an NPE due to attempting to return the first item of the `getSupportedModes()` list.

I agree that we _should_ write a test for this but, right now, I'm not exactly sure _what_ to test.

Should we be allowing `getSupportedModes()` to return an empty list or should it __always__ return at least one entry (even if that entry is `DefaultProgrammerManager.NONE`?

If we can agree that there should __always__ be at least one entry in the list, then this can be tested.

rhwood, 2016-02-17:
If ```getBestMode()``` were abstract and non-null would that not solve the problem by requiring that it be overridden?

I'm not sure how you would test this without requiring that every subclass of ```AbstractProgrammer``` _also_ be tested, but this removes the need to require a difficult to enforce contract on ```getSupportedModes()``` (that the List be not empty).

dheap, 2016-02-17:
*) DecoderPro configured to NCE; NCE USB; (any existing serial port);
V7.x.x; SB5
*) Save, restart

Service Mode Programmer appears in both Roster Window and in menus. It
should not.

Dave

dheap, 2016-02-17:
> Sorry to just be coming to this. Would like to write a test for this, but
> not sure exactly where the problem lies. Is this the failure:
> *) PanelPro configured to NCE; NCE USB; (dummy port); V6.x.x; [...]

Agreed it is in PanelPro with these settings. Not so with DecoderPro

> Alternately, if I open "Roster" (the DP3 main window), there's no
>service mode programming option.

On my computer, there is a "Programming Track" radio button and a
"Programming Mode" selector, even with your settings. Neither should be
there (c.f. SPROS CS mode).

>How do you get to the failure?

There were two failures. The first reported was the OBE error in
getSupportedModes(). That has been patched. The second is the bogus
appearance of unsupported mode radio button and selector in Roster Window
(and menu item in DP) as documented above.

rhwood, 2016-02-17:
@dheap That's a different problem; ```ProgServiceModePane``` and whatever creates the menu items should not add a programmer button if the only available mode is ```DefaultProgrammerManager.NONE```, and the menu item should always appear, but be disabled if the only available mode is ```NONE```. I suspect that other systems that have already been returning a single mode of ```NONE``` also show the behavior you observed. Whatever is creating the GUI needs to allow for a list of supported modes that indicates that no supported mode is available.

rhwood, 2016-02-17:
It turns out that #816 is the first use of ```DefaultProgrammerManager.NONE``` outside of a single test, so maybe requiring that ```getSupportedModes()``` be not empty is not the right thing to do (if only because we then have to track down every use of ```getSupportedModes()``` that accepts its being null or empty and changing those).

dheap, 2016-02-17:
> @dheap That's a different problem; ProgServiceModePane and whatever
> creates the menu items should not add a programmer button if the only
> available mode is DefaultProgrammerManager.NONE, and the [...]

Program Managers do it differently. The SPROG CS hides or shows the
programmer button and mode menu based on the return value from
isGlobalProgrammerAvailable(). The NCE USB Program Manager is doing it
differently and getting it wrong. The issue seems to be in the
configureManagers() methods in the respective xxxxSystemConnectionMemo
classes.

dheap, 2016-02-17:
The problem of SB3/5 falsely showing Programming track Mode as being
available in the Roster Window predates #816
<https://github.com/JMRI/JMRI/pull/816> . Its just that the tracking of the
getSupportedModes() OBE brought it to light again. I've been meaning to
follow it up for quite a while. But I'm hitting a brick wall finding it.
> It turns out that #816 <https://github.com/JMRI/JMRI/pull/816>  is the first
> use of DefaultProgrammerManager.NONE outside of a single test, so maybe
> requiring that getSupportedModes()be not empty is [...]

mattharris, 2016-02-17:
Looking a bit deeper, there is a pattern that allows for an empty `getSupportedModes()` list:

[https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/AbstractProgrammer.java#L131]

specifically when trying to set the `ProgrammingMode` to `null`.

Now, is this pattern correct, or should we also allow for a `null` ProgrammingMode when `getSupportedModes()` returns a single entry `DefaultProgrammerManager.NONE`?

Note I did say that my #816 'fix' was just a first pass at this... :smiley:

KenC57, 2016-02-17:
I've also had the hint that the SPROG in command station mode doesn't
support a programming track either. So that configuration needs
testing/fixing for this same sort of issue. Slowly we track down the other
places that might throw the same errors.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

KenC57, 2016-02-17:
Bob,

 

My test case NCE is using a V7 USB with all jumpers on. I switch the config
side between the different command stations. But there was an update last
night that tried the quick patch by having the getBestMode() return a NONE.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

KenC57, 2016-02-17:
My take on this is we should add a test to all systems that empty isn't
right. The code should return at least NONE.

 

On the issue of when the isGlobal. returning true or false, that's another
issue of making sure each system is doing the right thing for what it
supports.

 

But doing both is the best answer to cover all bases.

 

And it is suspected that things like SPROG when in command station mode is
also a problem since it doesn't support a programming track in that mode. So
checking its version of getBest and isGlobal. is needed.

 

My time middle of the week is very limited. So reading and responding to
email is most of my time. I can run a few tests but that's the extent of my
time right now. Friday-Sunday has much more time.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

KenC57, 2016-02-17:
The other issue is that the AbstractProgrammer needs to catch the error if
an empty list is returned. Right now the 'Main' thread seems to get killed
and you get a hung session if this happens.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

mattharris, 2016-02-17:
@KenC57 I think that you're right - we should always return, at least NONE.

Also, to make the AbstractProgrammer a bit more robust _if_ an empty list is returned.

KenC57, 2016-02-17:
Matt,

 

And I think we saw the user found the 'quick fix' does get him past the
issue for now. But we did open a can of worms and Zymurgy's Law seems to
have taken over.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2016-02-17:
> On Feb 17, 2016, at 3:25 AM, Dave Heap <notifications@github.com> wrote:
> 
> *) DecoderPro configured to NCE; NCE USB; (any existing serial port);
> V7.x.x; SB5
> *) Save, restart
> 
> Service Mode Programmer appears in both Roster Window and in menus. It
> should not.

I’m not seeing this with the v4.3.3 code.  Not sure why I’m getting different results from everybody else, but since people are working on this already I’ll step back.

bobjacobsen, 2016-02-17:
> I'm not sure how you would test this without requiring that every subclass of AbstractProgrammer also be tested, but this removes the need to require a difficult to enforce contract on getSupportedModes() (that the List be not empty).

As a programming idiom, if you want to ensure that getSupportedModes() never returns an empty string:

- mark it “final” in the base implementation classes
- have it invoke another method to get the value to return, e.g. getSupportedModesImplementation()
- have it check the return value and error (log, throw, whatever) if it’s not acceptable. This is a programming error, not a run-time exception, so I’d throw a subclass of Error, not Exception, maybe “CoderError” (great name, though not really the right meaning) or better AssertionError

KenC57, 2016-02-17:
Bob,

 

Please try to stay in the loop on the ideas. As you can see this has a few
different issues being uncovered and some of this likely traces back to us
needing to be reminded about the original ideas of how things (should) work.

 

Big issue I see is why/how this can cause the main thread to die. That
leaves things like the receive/send threads and a few others out  there but
not always obvious to the user on how to kill and clean up. Also since it
leaves no user interface, how are they supposed to fix it??

 

Minor issue is when and how we should be handling things like when a system
doesn't support service or ops modes. Some things seem in conflict about
this.

 

Very minor issue is if early NCE systems supported direct programming mode
or not. Current code reads that only the 2006 and PowerCab do direct mode.
But my read of the sysman05 version seems to say they supported direct from
the beginning. You were around back then so I suspect you have a better feel
for this issue.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

dheap, 2016-02-17:
The SPOG in command station mode works fine. The Program Track button does not appear, neither does the Programming Mode selector.

They is how the SB5 connection should work.

> I've also had the hint that the SPROG in command station mode doesn't
> support a programming track either. So that configuration needs
> testing/fixing for this same sort of issue. Slowly we track down the other
> places that might throw the same errors.

dheap, 2016-02-17:
I'm working with JMRI master, not 4.3.3. Is anyone seeing the same as me?

> On 18 Feb 2016, at 2:07 AM, Bob Jacobsen <notifications@github.com> wrote:
> 
> > On Feb 17, 2016, at 3:25 AM, Dave Heap <notifications@github.com> wrote:
> > 
> > *) DecoderPro configured to NCE; NCE USB; (any existing serial port);
> > V7.x.x; SB5
> > *) Save, restart
> > 
> > Service Mode Programmer appears in both Roster Window and in menus. It
> > should not.
> 
> I’m not seeing this with the v4.3.3 code. Not sure why I’m getting different results from everybody else, but since people are working on this already I’ll step back.

dheap, 2016-02-17:
This is what I see with SB5 & SPROG CS connections respectively:
[image]
(SPROG-CS.png is exactly what we should be seeing.)
[image]
(SB5.png is **not** what we should be seeing.)

KenC57, 2016-02-17:
Maybe I don't remember right or maybe this is part of what's seen as a
conflict.

Does a connection have one 'programmer' which in turn support different ways
like ops or service, or does a connection have two different programmers,
one for ops, one for service??

Walking the code I also see a number of places where the code (mix of
programmer and NCE parts) that use depreciated calls for finding 'default'
things. So I don't know if that's also part of the issues we have that some
are still doing things 'the old way'.

That's about all the digging I've done for now. But here is what I have in
my code for 'fixing' this:

    public ProgrammingMode getBestMode() {
        List<ProgrammingMode> posibleModes = getSupportedModes();
        ProgrammingMode ret = jmri.managers.DefaultProgrammerManager.NONE;
        if (!posibleModes.isEmpty()) {
            ret = posibleModes.get(0);
        }
        return ret;
    }

I tried to setup a try/catch but it says there are no throwable exceptions
coming from that branch of code. Funny thing is my spellcheck in email
thinks programing is the right spelling, not 2 mm's.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

dheap, 2016-02-17:
The root cause of the problems reported by SB5 users in V4.3.3 is that the
NCE code is incorrectly configuring a Service mode programmer for the SB5
despite the fact that the SB5 hardware does not support it.

We have been sidetracked into fixing the OBEs, NPEs etc. in
getSupportedModes(), getBestMode() et al **caused** by trying to configure
an invalid service mode programmer that should never have been loaded.

Things are not going right in the NceSystemConnectionMemo.

If we look at a correctly-working example; in the SPROG CS code the
SprogCSSerialDriverAdapter calls
SprogSystemConnectionMemo.configureManagers(), which in turn calls
InstanceManager.setProgrammerManager(getProgrammerManager()), which in
turn calls SprogProgrammerManager.isGlobalProgrammerAvailable(). The
answer from isGlobalProgrammerAvailable() is the sole determining factor
in whether a service mode programmer is loaded or not. This all works
correctly.

If the NceSystemConnectionMemo were to call
NceProgrammerManager.isGlobalProgrammerAvailable() as part of its
configureManagers()and use the answer from isGlobalProgrammerAvailable()
to determine whether to set up a programmer, things would work properly;
The user would not see a bogus Programming Track Mode and Programming Mode
selector and the OBEs and NPEs we have been chasing and eliminating would
not have happened.

Unfortunately the NceSystemConnectionMemo.configureManagers() code does
not invoke isGlobalProgrammerAvailable() and I can't track where it is
loading the service mode programmer. It's not where it seems to be (lines
217-224). Something else is loading it regardless.

bobjacobsen, 2016-02-17:
> Unfortunately the NceSystemConnectionMemo.configureManagers() code does
> not invoke isGlobalProgrammerAvailable() and I can't track where it is
> loading the service mode programmer. It's not where it seems to be (lines
> 217-224). Something else is loading it regardless

Sorry I can't help directly, but I still can't make v4.3.3 fail. 

I suggest putting 

     new Exception("Traceback").printStackTrace();

in all the ctors that might be called when constructing a global programmer. Hopefully that will fire and show you where it's being done. 

Bob

bobjacobsen, 2016-02-17:
> My take on this is we should add a test to all systems that empty isn't
> right. The code should return at least NONE.

Code wins.

That having been said, I'm not a fan of NONE. 

The empty set is not the same as a sent containing the one element "empty". 

It seems easier to write GUI code that says "display elements; if there are no elements, disable yourself." Handling NONE requires a lots more special casing. 

In v4.3.3, I don't see a use of NONE except in its definition & a test class. Perhaps it's best to remove it.

dheap, 2016-02-18:
That worked, but untangling Service and Ops Mode programmers (one extends the other) so you can display only the Ops Mode programmer is getting beyond me.

But I have realised that NceProgrammer.getSupportedModes() should not return DefaultProgrammerManager.NONE but DefaultProgrammerManager.OPSBYTEMODE, not that it will fix our problem but it may be a start along the way.

> @bobjacobsen
> 
> I suggest putting 
> 
> new Exception("Traceback").printStackTrace();
> 
> in all the ctors that might be called when constructing a global programmer. Hopefully that will fire and show you where it's being done.

dheap, 2016-02-18:
No, that wasn't a good idea...

>But I have realised that NceProgrammer.getSupportedModes() should not
>return DefaultProgrammerManager.NONE but
>DefaultProgrammerManager.OPSBYTEMODE, not that it will fix our problem
>but it may be a start along the way.

KenC57, 2016-02-18:
Folks,

 

Clarify this: should the manager have one programmer manager and it services
both ops and service, or should there be two different managers (so you can
have different connections as defaults)??

 

I had always thought we had a singular manager for a connection and it did
both. But following some of the stuff when we split the concept of which is
default (in the default selector in preferences) I don't know if those
changes made the one aware of different the two different or what.

 

Dave,

 

Wouldn't that mean that almost all systems should reply with the ops byte
mode?? This is where I got the feeling this new code for 'best mode' was
among only the service mode programming. But I not sure what all was done
for that change of default stuff.

 

But my review of the stuff in the NCE connection memo, I don't like what I
see either around those line numbers. I'll try testing and confirm that
part. But the right answer gets back to my 'is that only the service mode'
or not.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

dheap, 2016-02-18:
Agreed. That's why I retracted it.

> Wouldn't that mean that almost all systems should reply with the ops byte
> mode?? This is where I got the feeling this new code for 'best mode' was
> among only the service mode programming. But I not sure what all was done
> for that change of default stuff.

dheap, 2016-02-18:
I think we should only have one manager per connection. Maybe the problem is in the code in the DP app and RosterFrame. It needs to separately check isGlobalProgrammerAvailable() and isAddressedModePossible() before it decides what to enable/disable (DP menu) or hide/unhide (RosterFrame).

It seems the PP app code gets it right (at least in the menus) so maybe we should look there and copy what it does.
> 
> Clarify this: should the manager have one programmer manager and it services
> both ops and service, or should there be two different managers (so you can
> have different connections as defaults)??

pabender, 2016-02-18:
On 02/17/2016 10:27 PM, Ken Cameron wrote:
> Clarify this: should the manager have one programmer manager and it 
> services
> both ops and service, or should there be two different managers (so 
> you can
> have different connections as defaults)??
There should only be ONE manager per connection type.

In the SPROG case, there is an enum that indicates whether the command 
station supports ops mode or service mode.  one of the two enum values 
can be passed to the constructor of the programmer manager.

In the Lenz case, there are command stations that support no programming 
(the Atlas Commander and Lenz Compact are examples), There are command 
stations that support service mode programming but not Ops Mode 
programming (The LH200, when used as a command station), and there are 
command stations that support everything. We handle the options 
completely based on the command station type ( in the Lenz Compact/Atlas 
Commander case, we don't even load the programmer manager...).

Paul

dheap, 2016-02-18:
We also need to fix NceProgrammerManager.isGlobalProgrammerAvailable() so
it only returns true for Power Cab, not all USB connections. (As well as
the Serial,Simulator and Network connections.)

KenC57, 2016-02-18:
Ok, so one program manager for the connection and then it deals with the
modes it does and doesn't support. Makes sense, but there does seem to be a
number of things that consider that a programmer existing means both service
and ops, even if it may only do one or the other.

 

At least that's my read so far. But I'm tired enough today.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

dheap, 2016-02-18:
I've checked again and I think it is probably OK the way it currently is.
> 
> We also need to fix NceProgrammerManager.isGlobalProgrammerAvailable() so
> it only returns true for Power Cab, not all USB connections. (As well as
> the Serial,Simulator and Network connections.)

dheap, 2016-02-18:
The main problem being RosterFrame. I did some proof of concept checking in this code and it seems quite feasible. But with all the multiple connections and selectors, I decided it is too complex for me to get my head around everything that has to do the "isAvailable..." checks and when do the disabling. Obviously the visibility of items has to change as the Programmer selection changes (to suit the capabilities of that particular Programmer).

Someone more familiar with the RosterFrame code would have to do it. I'm worn out chasing up this one and while I can see what needs fixing in general terms, I'm not up to trying to track all the nuances of RosterFrame.

> 
> Ok, so one program manager for the connection and then it deals with the
> modes it does and doesn't support. Makes sense, but there does seem to be a
> number of things that consider that a programmer existing means both service
> and ops, even if it may only do one or the other.

bobjacobsen, 2016-02-18:
Is it possible that we’ve lost the basic operation here?  It has been, and IMHO should remain, conceptually quite simple:

1) InstanceManager and the default-setting system provides access to a single default ProgrammerManager for global (unaddressed) and a single default ProgrammerManager for address (ops mode) programming.  (Can be the same Object, that doesn’t change this) Additional ones may also be available.  These have been registered by the configured systems during startup.

2) Any code (GUI or otherwise) that needs a programmer (of either global or addressed type) can (a) check if one is available via isGlobalProgrammerAvailable()/isAddressedModePossible()  or (b) just ask for the programmer.  If it finds is..Available() == false, or no programmer (a null return from the get..Programmer() call), then the code has to handle that:  a piece of GUI code disables, ident code skips doing ident, etc.

3) The ProgrammerManagers and Programmers have to meet a simple contract:  If the ProgrammerManager can’t return a working Programmer, it has to return is..Available()==false and a null when asked for the Programmer.  If they don’t meet that contract, they’re just broken and need to be fixed.

This leaves no use for “NONE” as a programmer mode. Not having NONE is a great simplification; the code doesn’t have to special case that, etc.

A programmer with only “NONE” doesn’t really meet its contract:  if you ask it to read, what’s it supposed to return?  And what’s the function of “NONE” mode if there are also others available?

Code wins, of course, so if people want to do something else they should do that.  But if you change the structure above, please change _everything_ consistently, including the JavaDocs in the jmri.* classes to describe how you’ve changed all the code.  Right now, they say that get..Programmer() returns null when a programmer isn’t available, so code has to handle that. If you want to switch to NONE, please change things so a programmer is _always_ returned, and add annotations so that FindBugs can check your work.

dheap, 2016-02-18:
See comments:

> Is it possible that we’ve lost the basic operation here? It has been, and IMHO should remain, conceptually quite simple:

I don't think I have.
> 1) InstanceManager and the default-setting system provides access to a single default ProgrammerManager for global (unaddressed) and a single default ProgrammerManager for address (ops mode) [...]

Correct.

> 2) Any code (GUI or otherwise) that needs a programmer (of either global or addressed type) can (a) check if one is available via isGlobalProgrammerAvailable()/isAddressedModePossible() or (b) just [...]

This is what is not happening in the case of RosterFrame at present and this what I am arguing should happen. RosterFrame does no interrogation of isAvailable...() but simply decides a multi-use programmer can always accept both. In the case of NCE, this is not true for some connection types.

> 3) The ProgrammerManagers and Programmers have to meet a simple contract: If the ProgrammerManager can’t return a working Programmer, it has to return is..Available()==false and a null when asked for the Programmer. If they don’t meet that contract, they’re just broken and need to be fixed.

That is working fine as far as the NCE programmer is concerned, but RosterFrame never asks and that is why we have a problem with the SB5.
> This leaves no use for “NONE” as a programmer mode. Not having NONE is a great simplification; the code doesn’t have to special case that, etc.
>
> A programmer with only “NONE” doesn’t really meet its [...]

I'm not a great fan of NONE. Note that NONE (previously a zero-length list) in the case of NCE does not indicate that Ops mode is unavailable. It only covers service mode.

It probably should have been returning null and then we probably would not have seen the OBEs.

Now it returns NONE, which allows the user to open a non-functional programmer and try doing things.

> Code wins, of course, so if people want to do something else they should do that. But if you change the structure above, please change _everything_ consistently, including the JavaDocs in the jmri.* [...]

As far as I am concerned, using NONE only masked two problems:
A) RosterFrame is not doing isAvailable() and hence allows access to dud programmers, leading to the original OBE.
B) NCE programmer should have been returning null, not an empty list.

I believe our priorities are to:
1) Fix RosterFrame so it interrogates isAvailable() before trying to set up any item in the frame. That will fix the bad display and probably also make the (NONE versus null versus empty list) return a moot point. This can safely be done **now** without 2) or 3).

2) Decide whether we want to use NONE or null as an empty return, but certainly not use null as a test of mode availability (we already have those methods).

3) Fix any non-conforming programmers returning empty lists to our final choice in 2).

Dave

mattharris, 2016-02-18:
I've been thinking more about this and I feel that there is not really a place for `NONE` as a programmer mode. Yes, I know that goes against what I was saying earlier but we're allowed to change our minds after thinking things through a bit more, yes?

My reasoning for using the `NONE` in the #816 fix was due to the fact that there was an expectation in `AbstractProgrammer.getBestMode()` that there would always be at least one entry in the programming mode list.

Now, logically, `AbstractProgrammer.getBestMode()` should only be called in the instance when we've already said that a programmer is available so, having an empty `getSupportedModes()` list is valid in those cases that do not have programmers available.

The #816 fix essentially masks a different problem in that we are erroneously suggesting that the NCE-USB SB5 combination does have the possibility of various programming modes available meaning `getBestMode()` was being called, when it is not, in fact, the case.

pabender, 2016-02-18:
Sent from my iPad

On Feb 18, 2016, at 3:24 AM, Dave Heap <notifications@github.com> wrote:
> As far as I am concerned, using NONE only masked two problems:
> A) RosterFrame is not doing isAvailable() and hence allows access to dud programmers, leading to the original OBE.
> B) NCE programmer [...]

As Bob mentioned earlier, { NONE } is not the same as {}. if a programmer truly supports no modes, An empty list is proper..... But in that case, there shouldn't be a programmer to ask what modes are supported.

> 3) Fix any non-conforming programmers returning empty lists to our final choice in 2).

It seems to me that the primary example of a non-conforming programmer is the NCE programmer.  Fix the is...Available() functions in NCE Programmer manager first, and then see what happens with the GUI.

Paul

bobjacobsen, 2016-02-18:
> As far as I am concerned, using NONE only masked two problems:
> A) RosterFrame is not doing isAvailable() and hence allows access to dud programmers, leading to the original OBE.

The choice is to use isAvailable _or_ to check for a null return from the get…Programmer call.  RosterFrame seems to be doing the second.  It’s getting a Programmer object, but that Programmer object is broken: It can’t program. The ProgramManager should never have provided it.

> B) NCE programmer should have been returning null, not an empty list.

Here are the things the NCE implementation should do:

i) If no service mode programming is possible (which as I understand it is the case we’re talking about), then the NCE ProgrammerManager should have isGlobalProgrammerAvailable() return false.

ii) If no service mode programming is possible (which as I understand it is the case we’re talking about), then the NCE ProgrammerManager should return null when asked for the global programmer.

In those cases, there should not _be_ an NCE Programmer (not ProgrammerManager) instance, so there’s no way to ask for Programmer.getSupportedModes()

In general, I believe that Programmer.getSupportedModes() should always return a list containing at least one valid mode, and the NONE mode should be removed.

> I believe our priorities are to:
> 1) Fix RosterFrame so it interrogates isAvailable() before trying to set up any item in the frame. That will fix the bad display and probably also make the (NONE versus null versus empty list) return a moot point. This can safely be done **now** without 2) or 3).

I have no problem with RosterFrame (which is very complicated code) checking before asking for a programmer.  

But as far as I can tell, it already properly handles the case when the ProgrammerManager returns null.  So proper coding of calls to isAvailable will just do the same thing.

Isn’t the problem that the NCE ProgrammerManager is returning a broken Programmer in a specific case?

(In the long run, changing RosterFrame to use the GUI widgets for programmer selection is probably a good idea, because that centralizes code for handling multiple programmers, etc. As they properly handle the case of one kind of programmer or the other not being present, that could be a good thing to do now instead of adding cals to is..Available to the existing code.  But, although I still can’t recreate the problem we’re talking about, I suspect that Single CV Programmer will show the same problem because the real issue is that the NCE ProgrammerManager isn’t living up to contract)

> 2) Decide whether we want to use NONE or null as an empty return, but certainly not use null as a test of mode availability (we already have those methods).

See above.

ProgrammerManager returns null for the Programmer request when there’s no possibility of programming.

> 3) Fix any non-conforming programmers returning empty lists to our final choice in 2).

I think the problem is the ProgrammerManager, not the Programmer.

KenC57, 2016-02-18:
Bob,

 

I think I see something that may have been cloudy for me before. See if this
sounds right.

 

1.      A connection will always have a programming manager (unless you have
neither ops or service modes).

2.      The programming manager will answer the isGlobal. or isAddr. and
getProgrammer. calls.

3.      The programming manager will only provide the programmer if the type
is supported (is whichever).

 

I think some of my thought has/had a confusion between having a programming
manager vs having a programmer. Does this sound better??

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2016-02-18:
> 1. A connection will always have a programming manager (unless you have
> neither ops or service modes).

Basically right. 

A wrinkle is that systems can (though not all yet do) separately register a global PM and an addressed PM with the InstanceManager. If the do that, the defaults for those can be set to different systems, eg SPROG for programming track, LocoNet for Ops mode. 

> 2. The programming manager will answer the isGlobal. or isAddr. and
> getProgrammer. calls.

Right, with the above caveat about specific types 

> 3. The programming manager will only provide the programmer if the type
> is supported (is whichever).

Right. 

> I think some of my thought has/had a confusion between having a programming
> manager vs having a programmer. Does this sound better??

Yes. 

The difference between access (and if need be creation) and functional use is why JMRI uses the ZzzManager vs Zzz object pattern so widely. It lets JMRI code self-assemble its references once configured & running.

dheap, 2016-02-18:
The the is...Available() functions in NCE Programmer manager are correct.

The GUI in RosterFrame is not using the the is...Available() functions at all!

> On 19 Feb 2016, at 12:14 AM, Paul Bender <notifications@github.com> wrote:
> 
> It seems to me that the primary example of a non-conforming programmer is the NCE programmer. Fix the is...Available() functions in NCE Programmer manager first, and then see what happens with the GUI.

dheap, 2016-02-18:
The NCE programmer was returning {} before Matt made his change!

> On 19 Feb 2016, at 12:14 AM, Paul Bender <notifications@github.com> wrote:
> 
> As Bob mentioned earlier, { NONE } is not the same as {}. if a programmer truly supports no modes, An empty list is proper..... But in that case, there shouldn't be a programmer to ask what modes are supported.

dheap, 2016-02-18:
I agree with NONE being removed. But why should the Programmer.getSupportedModes() return a valid mode (rather than {}) in the case that the hardware/connection type/firmware does not permit it.

That means getting to the end of our checks and saying; OK there are no modes available so we'd better tell a lie and return an invalid mode at random to satisfy the requirement that we must return a mode. That sort of constraint falls into the question category of "Have you stopped beating your wife?".

> On 19 Feb 2016, at 2:14 AM, Bob Jacobsen <notifications@github.com> wrote:
> 
> In general, I believe that Programmer.getSupportedModes() should always return a list containing at least one valid mode, and the NONE mode should be removed.

bobjacobsen, 2016-02-18:
> But why should the Programmer.getSupportedModes() return a valid mode (rather than {}) in the case that the hardware/connection type/firmware does not permit it.

I'm sorry that I'm not communicating clearly, but the paragraph above means you have not understood the main point. 

If the hardware provides no programming modes, there should not be a Programmer object from which to request modes. 

If there's no way to program, there is nothing for a Programmer object to represent, hence no such object should exist.

bobjacobsen, 2016-02-18:
> The the is...Available() functions in NCE Programmer manager are correct.
> 
> The GUI in RosterFrame is not using the the is...Available() functions at all!

Does the NCE ProgrammerManager return null from get..Programmer iff is..Available returns false?

KenC57, 2016-02-18:
I'm going to presume we rollback the NONE patch. I've not pulled that, so my
system is just before then.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

dheap, 2016-02-18:
You have communicated. But the convention that you are not going to ask for a mode is not known to the programmer. The code still exists and still has to have the logic for building a list. If that code ends up adding no elements to the list, you have an empty list.

If the Programmer Manager is in error and says a Global Programmer is available, hence allowing the Programmer object to be created, a call to getSupportedModes() will return an empty list.

I can't see how you can fudge the code in getSupportedModes() to deal with that case!

> On 19 Feb 2016, at 7:19 AM, Bob Jacobsen <notifications@github.com> wrote:
> 
> > But why should the Programmer.getSupportedModes() return a valid mode (rather than {}) in the case that the hardware/connection type/firmware does not permit it.
> 
> I'm sorry that I'm not communicating clearly, but the paragraph above means you have not understood the main point. 
> 
> If the hardware provides no programming modes, there should not be a Programmer object from which to request modes. 
> 
> If there's no way to program, there is nothing for a Programmer object to represent, hence no such object should exist.

KenC57, 2016-02-18:
The way I'm interpreting this whole thing is:

 

1.      The program manager will return a proper programmer, if one exists.

2.      If a programmer exists, it will have a best mode.

3.      For an ops mode programmer, it is always a OPS mode, I don't know of
any other for this type of programmer.

4.      For a service mode programmer, we get something like paged, direct,
etc.

 

??Do we have an example of a system that registers two program managers and
one that only uses one?? Seeing the examples will help.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

mattharris, 2016-02-18:
Randall's #833 is much nicer than my 'orrible hacky #816.

[4 more replies]

closed 2016-02-21 by bobjacobsen