JMRI/JMRI #824: Where is the "System connection" dropdown box populated?

pabender, 2016-02-17:
Previously, after you selected a manufacturer for the DCC system, the "System Connection" drop-down box was populated using the jmri.jmrix.ConnectionTypeList interface.

I've been looking at the Roco Z21 system today, and I noticed that the Z21XPressNetStream is showing as an available connection option, but this isn't actually a valid system independently (it is a system created automatically when you configure a Z21 connection.  It is not a connection the user can manipulate directly).  The Z21XPressNetStream is NOT listed in the RocoConnectionTypeList.  

What I'm really confused about right now is that the Z21XPressNetStream IS showing in the preferences, but the other stream protocols derived from AbstractStreamPortController are NOT showing in the preferences as choices.

Paul

rhwood, 2016-02-17:
The z21XNetConnectionConfig is listed in the RocoConnectionTypeList as the last item in that list, so it should be showing in combo box if I'm reading the code correctly.

rhwood, 2016-02-17:
The ```z21XNetConnectionConfig.name()``` method returns "Z21 XPressNet Stream".

pabender, 2016-02-17:
Ok, I must have missed that.  It certainly isn't supposed to be there....

closed 2016-02-17 by pabender