JMRI/JMRI #857: JMRI PanelPro is not using the jmri-<node>-<profile> profile.xml connection settings

dsand47, 2016-02-23:
I am using the new shared profile concept to share between Win7 and OS X.

This test used build #3077.  

Each system updates profile:ProfileConfig.xml as well as its own node specific profile.xml as expected.

The problem is that during JMRI start up, each system defaults to the current value in profile:ProfileConfig.xml.

Start PP using Windows after OS X.
[image]

After fixing Windows I returned to OS X.

Start PP using OS X after Windows.
[image]

Final Profile Structure:
[WaltM__USB.zip](https://github.com/JMRI/JMRI/files/141942/WaltM__USB.zip)

Dave Sand

rhwood, 2016-02-23:
This concept is not yet expected to be working, only the plumbing is in place for it to be written. See #69, which is the enhancement proposal and remains open.

closed 2016-02-23 by rhwood

Glasairman, 2016-05-21:
My connection always starts at speed 9,600 bps although the active profile says: (Version Umbuntu 16.04)

**<?xml version="1.0" encoding="UTF-8"?>
<auxiliary-configuration xmlns="http://www.netbeans.org/ns/auxiliary-configuration/1">
    <connections xmlns="http://jmri.org/xml/schema/auxiliary-configuration/connections-2-9-6.xsd">
        <connection xmlns="" class="jmri.jmrix.sprog.sprog.configurexml.ConnectionConfigXml" disabled="no" manufacturer="SPROG DCC" port="/dev/ttyACM0" speed="19,200 bps" systemPrefix="S" userName="SPROG">**
            <options/>
        </connection>
    </connections>
</auxiliary-configuration>

Glasairman, 2016-05-21:
<auxiliary-configuration xmlns="http://www.netbeans.org/ns/auxiliary-configuration/1">
    <connections xmlns="http://jmri.org/xml/schema/auxiliary-configuration/connections-2-9-6.xsd">
        <connection xmlns="" class="jmri.jmrix.sprog.sprog.configurexml.ConnectionConfigXml" disabled="no" manufacturer="SPROG DCC" port="/dev/ttyACM0" speed="19,200 bps" systemPrefix="S" userName="SPROG">
            <options/>
        </connection>
    </connections>
</auxiliary-configuration>

rhwood, 2016-05-21:
@Glasairman please open a new issue describing your issue. It does not seem to be related to the original issue, which is related to how settings are (or are not) copied between computers.