JMRI/JMRI #918: Change FindBugs annotations

mattharris, 2016-03-08:
Currently, we use annotations around the JMRI codebase to pass information to the FindBugs code analyser.

Until now, we've been using those provided by FindBugs in the `edu.umd.cs.findbugs.annotations` package.

In recent versions, FindBugs have deprecated the use of many of these annotations and recommending the use of annotations within the `javax.annotation` package.

This issue is created to discuss the options we have:

1. Do nothing to current codebase and continue to use `edu.umd.cs.findbugs.annotations` for new code
1. Do nothing to current codebase but use `javax.annotation` for new code
1. Change current (and future) codebase to use `javax.annotation`

bobjacobsen, 2016-03-08:
Generally, I favor moving forward for things like this.  It seems a targeted bulk-replace would be in order (choice 3).  The only tricky bit I see is NonNull vs Nonnull, but some quick grepping seems to indicate that a bulk replace would be OK for that. But perhaps I've overlooked something because:

From #912:
> But there do not always seem to be exact drop-in replacements for every annotation we currently use so would need a bit of work to move over to using javax.annotation in our case.

Which are those?  

A quick grep over just the import statements (not fully-qualified uses) finds these in the code:
````
edu.umd.cs.findbugs.annotations.CheckForNull;
edu.umd.cs.findbugs.annotations.CheckReturnValue;
edu.umd.cs.findbugs.annotations.DefaultAnnotation;
edu.umd.cs.findbugs.annotations.NonNull;
edu.umd.cs.findbugs.annotations.Nullable;
edu.umd.cs.findbugs.annotations.SuppressFBWarnings;
edu.umd.cs.findbugs.annotations.SuppressWarnings;
````

For ones that are equivalent, we could do the immediate bulk replace and change the [technical page](http://jmri.org/help/en/html/doc/Technical/FindBugs.shtml) to let people know about the new recommendation, leaving the others for working out later. Sort of choice 2.5. I'd be happy to do that, it should be pretty quick.

mattharris, 2016-03-08:
The main one that seems to be a potential issue is:

`edu.umd.cs.findbugs.annotations.DefaultAnnotation`

From the Javadoc:

> __Deprecated.__ - Use the JSR305 annotations instead. For example, you can use `ParametersAreNonnullByDefault` instead of `@DefaultAnnotation(NonNull.class)` so that method parameters are nonnull by default in the annotated element. You can also use `TypeQualifierDefault` in general to define your own annotation that specifies a default type qualifier.

We should also consider use of `edu.umd.cs.findbugs.annotations.SuppressFBWarnings` instead of `edu.umd.cs.findbugs.annotations.SuppressWarnings`.

Note: not all FindBugs annotations are deprecated - we do not need to replace _all_ of them.

mattharris, 2016-03-08:
@bobjacobsen before doing any bulk replace work, we should probably merge #917.

But I am in agreement with your option 2.5

mattharris, 2016-03-08:
#919 provides a first pass at this.

There _might_ be some stray unused imports as this was done via grep/sed on various combinations.

mattharris, 2016-03-08:
I've cleaned up most of the unused imports now and updated the technical page.

#919 now includes these changes.

bobjacobsen, 2016-03-08:
Ran FindBugs on the branch, looks good.

mattharris, 2016-03-08:
Shall we merge then?

mattharris, 2016-03-09:
#919 now merged.

bobjacobsen, 2016-03-09:
> Shall we merge then?

I think so.

mattharris, 2016-03-09:
Already did it this morning :-)

mattharris, 2016-03-09:
The [Jenkins FindBugs build](http://jmri.tagadab.com/jenkins/job/Development/job/Findbugs/59/findbugsResult/) after this commit has run successfully without any changes to the results.

So, looks like this has gone in OK and I didn't break anything.

bobjacobsen, 2016-03-19:
Much FindBugs progress in past week, so I'm closing this.

closed 2016-03-19 by bobjacobsen