JMRI/JMRI #921: LoadCommon() in ConnnectionConfigureXml causing issues with restoring connections.

pabender, 2016-03-09:
I've been investigating a couple of issues with RaspberryPi connections.

The immediate symptom I am seeing is that when you save a connection with "name" and prefix "n", it is restored with "name2" and prefix "n2".  (My configuration has both RaspberryPi GPIO and a standalone RFID connection defined, using /dev/ttyAMA0, and both connections exhibit this behavior.)

After checking all the things I normally check when this occurs, I've discovered that the call to 

loadCommon(shared,perNode,adapter) is causing the problem.  I have not traced the precise cause of the issue yet, but I know that commenting out the call to LoadCommon() corrects the symptom

This line was added back in September ( https://github.com/JMRI/JMRI/commit/096c0b5abcb1d73859d08b1113a6f03dc325e75d )

Paul

bobjacobsen, 2016-03-09:
Does this happen for all kind of connections, e.g. a LocoNet simulator connection?

rhwood, 2016-03-09:
If this did, we have more reporting about issues out of trunk from other developers or users (this change has been in since 4.1.x).

Because we have not heard more about this, I don't think this change is the real cause of the issue, but merely caused another issue to be exposed.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

pabender, 2016-03-09:
On 03/09/2016 10:32 AM, Randall Wood wrote:
> If this did, we have more reporting about issues out of trunk from other
> developers or users (this change has been in since 4.1.x).
>
> Because we have not heard more about this, I don't think this change is
> the real cause of the issue, but merely caused another issue to be exposed.

I agree that this change isn't the real cause.  I stated that commenting 
out the line removes the symptom.

I'm going to have to start running bisect on this to find out where the 
issue was introduced.

Paul

pabender, 2016-03-10:
On 03/09/2016 09:59 AM, Bob Jacobsen wrote:
>
> Does this happen for all kind of connections, e.g. a LocoNet simulator 
> connection?
>
>
I had not tried the LocoNet simulator, only an RFID connection and a 
RaspberryPi connection.

But trying LocoNet simulator just now, it does not exhibit the same issue.

Paul

pabender, 2016-03-10:
On 03/09/2016 11:28 AM, Paul Bender wrote:
> On 03/09/2016 10:32 AM, Randall Wood wrote:
>
> I agree that this change isn't the real cause.  I stated that commenting
> out the line removes the symptom.
>
> I'm going to have to start running bisect on [...]

It turns out the introduction of the loadCommon function, and it's 
interaction with the system connection memo, really did cause this 
issue. (It is ultimately the system connection memo that appends the 
numeric value automatically to a profile name and/or prefix, but only
at construction time.)

In addition to the RaspberryPi GPIO connection and the RFID connection 
(in a single profile, which was the original configuration showing the 
problem), I have also setup:
a LocoNet simulator connection,
an XPressNet simulator connection
a Lenz LIUSB connection.
an RFID connection by itself,
a RaspberryPi connection by itself,
a RaspberryPi connection and an XNetSimulator connection
a RaspberryPi connection and a LocoNetSimulator connection
a new profile with a RaspberryPi connection and the Rfid connection

The only one of these that fails to load correctly is the original 
profile.  On Tuesday night, loading the original profile was failing 
silently. On Wednesday night, loading the original profile was still 
failing, but there were null pointer exceptions thrown.

It looks like this was the same issue I was fixing in PR #926 and which 
Randall fixed in another way in PR #929.  My fix #926 doesn't solve the 
issue, but #929 appears to fix it correctly.

Paul

closed 2016-03-10 by pabender

rhwood, 2016-03-10:
I think ce45e1f70c2591f90201783450d61cbba26a266a introduced this issue because its that change that switched from using the shared configuration Element to using the perNode configuration Element.

Sorry about that.

commit 88104b19 (2016-03-10): Merge pull request #929 from rhwood/master

Fix for loading preferences from unconverted profile.
Fixes #921.