JMRI/JMRI #954: "Dummy" Layout Connection

k4kfh, 2016-03-15:
I'm developing [a software around JMRI](http://github.com/k4kfh/ZephyrCab) using WebSockets. It integrates closely enough with the roster that I can't really work on the program at all without a JMRI install running. Obviously I can just set up a JMRI install on my laptop, but how can I set up a true "dummy" layout connection? I thought that the "None" connection option was what I needed, but it doesn't pretend to be a layout. For example, when I send a command to turn the power on, JMRI drops the WebSockets connection and writes a big error about there being no power manager connection. I assume this is because of the connection being basically nonexistent. Does a true "dummy" connection function exist in JMRI? Is there something I can cobble together to make it work?

rhwood, 2016-03-15:
Setup a Loconet Simulator for a dummy layout.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

k4kfh, 2016-03-15:
@rhwood That worked perfectly, thank you!

closed 2016-03-15 by k4kfh