JMRI/JMRI #955: nbproject/ide-targets.xml missing but still referenced in project.xml

SteveRawlinson, 2016-03-16:
It looks like [this commit](https://github.com/JMRI/JMRI/commit/9eedbd9e99ffc5f9b1267cf8ab52feed1ee1d088) removed ide-targets.xml and also a reference to that file in project.xml but there is another reference to it at line 72. 

The result is that netbeans won't launch the debug target.

I don't know enough about netbeans to fix this quickly so I'm raising it here, perhaps @rhwood could take a look?

rhwood, 2016-03-16:
Sorry, got confused by the fact that “ant debug” (which is mapped to other actions) in JMRI does not run a debugger, but merely builds the JMRI classes.

I’ll get it fixed, but it will take a couple of days. In the meantime you can restore the ide-targets.xml file, but please don’t change the project.xml (that should get a debugger working again).

> It looks like this commit removed ide-targets.xml and also a reference to that file in project.xml but there is another reference to it at line 72.
>
> The result is that netbeans won't launch the [...]

Randall Wood
Alexandria Software
202.683.8604
randall.h.wood@alexandriasoftware.com
http://www.alexandriasoftware.com

mattharris, 2016-03-16:
@SteveRawlinson You can also run in debug mode if you set `debugger=true` in your `local.properties` file in the root of your checkout.

What this will then do is to launch the debugger when a regular target is executed (i.e. when 'Run' is chosen in NetBeans)

mattharris, 2016-03-16:
@SteveRawlinson Another option is to alter the target thus:

```
                <action name="debug">
                    <script>${ant.script}</script>
                    <target>decoderpro</target>
                    <property name="debugger">true</property>
                </action>
```

This will then launch the `decoderpro` and await for you to attach the debugger.

mattharris, 2016-03-16:
I'll take a look at automating this a bit further without having to bring back the `ide-targets.xml` file

SteveRawlinson, 2016-03-16:
Perfect Matt, thanks.

rhwood, 2016-03-16:
The ide-targets.xml introduces one ability we might want to retain: it has a netbeans specific XML entity that would automate attaching the debugger which could cause ant to fail elsewhere. That said, I'd like not to entirely reproduce another target, but to pass the debug port to a common target as a parameter.

mattharris, 2016-03-16:
#956 brings in an initial fix for this.

It doesn't automatically launch the debugger, rather sits and awaits for a debugger to be attached after launch.

SteveRawlinson, 2016-03-16:
... and that was the information I was missing (have to attach the debugger).

rhwood, 2016-03-19:
Fixed in #969.

closed 2016-03-19 by rhwood