JMRI/JMRI #973: AllTest logging ERROR messages

bobjacobsen, 2016-03-19:
AllTest is currently logging ERROR messages:

''''
    [junit] ERROR - Called with parameter "I". Please report to JMRI developers at https://github.com/JMRI/JMRI/issues/new [testShowAndClose] jmri.jmrix.DCCManufacturerList.?()
    [junit] java.lang.Exception
    [junit] 	at jmri.jmrix.DCCManufacturerList.getDCCSystemFromType(DCCManufacturerList.java:60)
    [junit] 	at jmri.util.ConnectionNameFromSystemName.getConnectionName(ConnectionNameFromSystemName.java:32)
    [junit] 	at jmri.jmrit.beantable.SensorTableAction.addPressed(SensorTableAction.java:120)
    [junit] 	at jmri.jmrit.beantable.AbstractTableAction$1$1.actionPerformed(AbstractTableAction.java:71)
''''

''''
   [junit] ERROR - Called with parameter "Internal". Please report to JMRI developers at https://github.com/JMRI/JMRI/issues/new [testShowAndClose] jmri.jmrix.DCCManufacturerList.?()
    [junit] java.lang.Exception
    [junit] 	at jmri.jmrix.DCCManufacturerList.getTypeFromDCCSystem(DCCManufacturerList.java:128)
    [junit] 	at jmri.util.ConnectionNameFromSystemName.getPrefixFromName(ConnectionNameFromSystemName.java:58)
    [junit] 	at jmri.jmrit.beantable.SensorTableAction.canAddRange(SensorTableAction.java:221)
    [junit] 	at jmri.jmrit.beantable.SensorTableAction.addPressed(SensorTableAction.java:143)
''''

''''
    [junit] ERROR - Called with parameter "Internal". Please report to JMRI developers at https://github.com/JMRI/JMRI/issues/new [testShowAndClose] jmri.jmrix.DCCManufacturerList.?()
    [junit] java.lang.Exception
    [junit] 	at jmri.jmrix.DCCManufacturerList.getTypeFromDCCSystem(DCCManufacturerList.java:128)
    [junit] 	at jmri.util.ConnectionNameFromSystemName.getPrefixFromName(ConnectionNameFromSystemName.java:58)
    [junit] 	at jmri.jmrit.beantable.SensorTableAction.canAddRange(SensorTableAction.java:221)
    [junit] 	at jmri.jmrit.beantable.SensorTableAction.access$000(SensorTableAction.java:34)
    [junit] 	at jmri.jmrit.beantable.SensorTableAction$2.actionPerformed(SensorTableAction.java:113)
''''

''''
    [junit] ERROR - Called with parameter "Internal". Please report to JMRI developers at https://github.com/JMRI/JMRI/issues/new [testShowAndClose] jmri.jmrix.DCCManufacturerList.?()
    [junit] java.lang.Exception
    [junit] 	at jmri.jmrix.DCCManufacturerList.getTypeFromDCCSystem(DCCManufacturerList.java:128)
    [junit] 	at jmri.util.ConnectionNameFromSystemName.getPrefixFromName(ConnectionNameFromSystemName.java:58)
    [junit] 	at jmri.jmrit.beantable.SensorTableAction.okPressed(SensorTableAction.java:172)
    [junit] 	at jmri.jmrit.beantable.SensorTableAction$1.actionPerformed(SensorTableAction.java:107)
''''

''''
    [junit] ERROR - Called with parameter "I". Please report to JMRI developers at https://github.com/JMRI/JMRI/issues/new [testShowAndClose] jmri.jmrix.DCCManufacturerList.?()
    [junit] java.lang.Exception
    [junit] 	at jmri.jmrix.DCCManufacturerList.getDCCSystemFromType(DCCManufacturerList.java:60)
    [junit] 	at jmri.util.ConnectionNameFromSystemName.getConnectionName(ConnectionNameFromSystemName.java:32)
    [junit] 	at jmri.jmrit.beantable.TurnoutTableAction.addPressed(TurnoutTableAction.java:779)
    [junit] 	at jmri.jmrit.beantable.AbstractTableAction$1$1.actionPerformed(AbstractTableAction.java:71)
''''

''''
    [junit] ERROR - Called with parameter "Internal". Please report to JMRI developers at https://github.com/JMRI/JMRI/issues/new [testShowAndClose] jmri.jmrix.DCCManufacturerList.?()
    [junit] java.lang.Exception
    [junit] 	at jmri.jmrix.DCCManufacturerList.getTypeFromDCCSystem(DCCManufacturerList.java:128)
    [junit] 	at jmri.util.ConnectionNameFromSystemName.getPrefixFromName(ConnectionNameFromSystemName.java:58)
    [junit] 	at jmri.jmrit.beantable.TurnoutTableAction.canAddRange(TurnoutTableAction.java:1366)
    [junit] 	at jmri.jmrit.beantable.TurnoutTableAction.addPressed(TurnoutTableAction.java:801)
''''

''''
    [junit] ERROR - Called with parameter "Internal". Please report to JMRI developers at https://github.com/JMRI/JMRI/issues/new [testShowAndClose] jmri.jmrix.DCCManufacturerList.?()
    [junit] java.lang.Exception
    [junit] 	at jmri.jmrix.DCCManufacturerList.getTypeFromDCCSystem(DCCManufacturerList.java:128)
    [junit] 	at jmri.util.ConnectionNameFromSystemName.getPrefixFromName(ConnectionNameFromSystemName.java:58)
    [junit] 	at jmri.jmrit.beantable.TurnoutTableAction.canAddRange(TurnoutTableAction.java:1366)
    [junit] 	at jmri.jmrit.beantable.TurnoutTableAction.access$500(TurnoutTableAction.java:52)
''''

''''
    [junit] ERROR - Called with parameter "I". Please report to JMRI developers at https://github.com/JMRI/JMRI/issues/new [main] jmri.jmrix.DCCManufacturerList.?()
    [junit] java.lang.Exception
    [junit] 	at jmri.jmrix.DCCManufacturerList.getDCCSystemFromType(DCCManufacturerList.java:60)
    [junit] 	at jmri.util.ConnectionNameFromSystemName.getConnectionName(ConnectionNameFromSystemName.java:32)
    [junit] 	at jmri.util.swing.BeanSelectCreatePanel.<init>(BeanSelectCreatePanel.java:90)
    [junit] 	at jmri.jmrit.beantable.signalmast.AddSignalMastPanel.<init>(AddSignalMastPanel.java:902)
    [junit] 	at jmri.jmrit.beantable.signalmast.AddSignalMastPanelTest.testDefaultSystems(AddSignalMastPanelTest.java:17)
''''

''''
    [junit] ERROR - Called with parameter "I". Please report to JMRI developers at https://github.com/JMRI/JMRI/issues/new [main] jmri.jmrix.DCCManufacturerList.?()
    [junit] java.lang.Exception
    [junit] 	at jmri.jmrix.DCCManufacturerList.getDCCSystemFromType(DCCManufacturerList.java:60)
    [junit] 	at jmri.util.ConnectionNameFromSystemName.getConnectionName(ConnectionNameFromSystemName.java:32)
    [junit] 	at jmri.util.swing.BeanSelectCreatePanel.<init>(BeanSelectCreatePanel.java:90)
    [junit] 	at jmri.jmrit.beantable.signalmast.AddSignalMastPanel.<init>(AddSignalMastPanel.java:902)
''''

''''
    [junit] ERROR - Called with parameter "I". Please report to JMRI developers at https://github.com/JMRI/JMRI/issues/new [main] jmri.jmrix.DCCManufacturerList.?()
    [junit] java.lang.Exception
    [junit] 	at jmri.jmrix.DCCManufacturerList.getDCCSystemFromType(DCCManufacturerList.java:60)
    [junit] 	at jmri.util.ConnectionNameFromSystemName.getConnectionName(ConnectionNameFromSystemName.java:32)
    [junit] 	at jmri.util.swing.BeanSelectCreatePanel.<init>(BeanSelectCreatePanel.java:90)
    [junit] 	at jmri.jmrit.beantable.signalmast.AddSignalMastPanel.<init>(AddSignalMastPanel.java:902)
    [junit] 	at jmri.jmrit.beantable.signalmast.AddSignalMastPanelTest.testSearch(AddSignalMastPanelTest.java:33)
''''

(Some seeming duplicates ommitted)

I think it's good to reduce spurious ERROR messages during testing; ideally, I'd like to get that down to none, so that an ERROR message is actually an error.

rhwood, 2016-03-19:
These all indicate that the Internal System Connection has no SystemConnectionMemo associated with it. This is an expected effect of not throwing an Exception when a SystemConnectionMemo does not exist in #960.

rhwood, 2016-03-19:
Actually there is an InternalSystemConnectionMemo class, I just don't know why it is not being used.

rhwood, 2016-03-20:
I've figured that there has been an InternalAdapter for about 12 years in the code, but that was simply never used. I don't know where it needs to be created.

bobjacobsen, 2016-03-20:
I thought this might be due to how the tests were set up, e.g. there was a missing part of the scaffold.  

But it might be more basic than that, as running PanelPro gives messages about "I" and "C" (for C/MRI).
````
     [java] 2016-03-20 06:32:49,445 jmrix.DCCManufacturerList             ERROR - Called with parameter "C". Please report to JMRI developers at https://github.com/JMRI/JMRI/issues/new [Listed Table Generation]
     [java] java.lang.Exception
     [java] 	at jmri.jmrix.DCCManufacturerList.getDCCSystemFromType(DCCManufacturerList.java:60)
     [java] 	at jmri.util.ConnectionNameFromSystemName.getConnectionName(ConnectionNameFromSystemName.java:32)
     [java] 	at jmri.jmrit.beantable.AbstractTableTabAction.createModel(AbstractTableTabAction.java:48)
````
````
     [java] 2016-03-20 06:32:49,463 jmrix.DCCManufacturerList             ERROR - Called with parameter "I". Please report to JMRI developers at https://github.com/JMRI/JMRI/issues/new [Listed Table Generation]
     [java] java.lang.Exception
     [java] 	at jmri.jmrix.DCCManufacturerList.getDCCSystemFromType(DCCManufacturerList.java:60)
     [java] 	at jmri.util.ConnectionNameFromSystemName.getConnectionName(ConnectionNameFromSystemName.java:32)
     [java] 	at jmri.jmrit.beantable.AbstractTableTabAction.createModel(AbstractTableTabAction.java:48)
````

This makes me think it's something internal to the system setup code.

rhwood, 2016-03-20:
There is definitely nothing wrong with the tests.

There are *specific* systems that are not registering SystemConnectionMemos with the InstanceManager, and these error messages are deliberately intended to identify those loudly, so we can identify which systems are not registering the SystemConnectionMemo.

Opening the Turnout table on a system with a LocoNet connection throws the error for the Internal connection only. Browsing that same table on a system with both a LocoNet and C/MRI system throws errors for the Internal and C/MRI connections, but again, not for the LocoNet connection.

So far, this identifies the Internal and C/MRI connections as not registering their SystemConnectionMemos when they are created, and those specific connection types need to be fixed to correctly silence these errors.

bobjacobsen, 2016-03-20:
> On Mar 20, 2016, at 6:59 AM, Randall Wood <notifications@github.com> wrote:
> 
> So far, this identifies the Internal and C/MRI connections as not registering their SystemConnectionMemos when they are created, and those specific connection types need to be fixed to correctly silence these errors.

Right.  The C/MRI one just isn’t registering in it’s ctor, got that working now (I think, still testing)

“Internal” is a bit messier.  There’s both a jmri.jmrix.internal system, but also separate Internal* managers that are created standalone.  Hope to investigate that later today.

rhwood, 2016-03-20:
I have a possible fix for the C/MRI in #988.

bobjacobsen, 2016-03-20:
> I have a possible fix for the C/MRI in #988.

Thanks. Working on possible fix for internal now.

bobjacobsen, 2016-03-20:
The internal case is a little more complicated than I thought.

There are two separate implementations for internal managers:
````
java/src/jmri/jmrix/internal/InternalTurnoutManager.java
java/src/jmri/managers/InternalTurnoutManager.java
````

The upper ones are the ones needed by the ConnectionManager.  The lower ones are automatically created by ProxyManagers; this is how historically the "internal" objects were added previous to real multi-system support.

To really fix this requires cleaning out all the lower-type managers.  That's actually not so hard _except_ that it's going to break other people's code (things like CATS).  So I'd like to think through some kind of deprecation strategy that will work. And that's the hard part.

More to follow.

bobjacobsen, 2016-03-20:
There's also this gem, from the initialization code in TurnoutOperationsManager:
````
     * There's a threading problem here, because this invokes gets the current
     * turnout manager, often the proxy manager, which in turn invokes
     * loadOperationTypes again. This is bad. It's not clear why it even works.
```

KenC57, 2016-03-21:
At least if we did find some proper solution that did require a change,
Rodney is likely able to work out and when the two packages can take the
step forward. So it would just become a timing and documentation issue.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2016-03-21:
Closed by #998 and #991

closed 2016-03-21 by bobjacobsen