JMRI/JMRI #987: Updated Signal Icons {from jmriusers}

mattharris, 2016-03-20:
The following Signal Icon updates should be incorporated
[PNB icons.zip](https://github.com/JMRI/JMRI/files/181271/PNB.icons.zip)

KenC57, 2016-03-20:
Comments from the user group thread mention that the sizes and shapes of the icons may need tuning. This comes from 'once upon a time' when it was critical to making icons rotate correctly. I don't know if that is still an issue. Also comments about some of the icons having mixed background colors. I was pointing out that having transparent backgrounds are the normal for icons in JMRI.

So these issues might need some work before rolling them into JMRI.
-ken c

mattharris, 2016-03-20:
@KenC57 Indeed. Will take these into account before merging.

pnbgit, 2016-03-20:
The backgrounds may well need tweaking, we used a solid background so the user could see the click-able area.  All icons that are different aspects of the same signal will be the same size, but sizes will vary from one signal type to another, as they were optimised for where they needed to fit on the panel.

pnbgit, 2016-03-20:
A further note, we used the black versions of the signal icons to indicate the unknown state, where the signals were represented by simple sensor icons. This might not be required when using them for signal heads or signal masts.

bobjacobsen, 2016-05-02:
Is this ready to merge? Or should we put a blue flag on it for more work?

mattharris, 2016-05-02:
Still needs more work - added `WIP` label.

mattharris, 2016-05-17:
Just received updated icons - will work on incorporating them

bobjacobsen, 2017-01-22:
Can we close this?

pnbgit, 2017-01-22:
I believe Matt wanted to make some modifications to the icons I provided, I don't know if this has been done yet. I can't see them in the latest JMRI release, so presumably they have not been merged yet.

mattharris, 2017-01-23:
Oh dear, it looks like I completely forgot about this.

Will need to look at it further to work out how far I got with it.

closed 2021-04-14 by stale[bot]