JMRI/JMRI #1007: Preserving layout editor loco block location between sessions

SteveRawlinson, 2016-03-22:
By associating memory variables with layout blocks you can display the location of locomotives as they move around the layout in the layout panel. These locations are not preserved between JMRI sessions. I am going to implement a (optional) means of saving this information (probably to xml files) on exit and reading those data on load.

I am planning to do this in a script which I'm using to automate my layout. With a bit more effort I could probably implement in the layout editor itself. 

While this is probably only a few extra lines of code, it will be a fair amount of work for me just getting my head around unfamiliar code so before I start I thought I'd test the waters and see if people think this would be useful.

mattharris, 2016-03-22:
This sounds a useful addition to me.

You might want to look at how I did a similar implementation for ID tag persistence.

SteveRawlinson, 2016-03-22:
Thanks Matt. If I created a LocoLocationXml class similar to your IdTagManagerXml does that sound like a reasonable place to start?

mattharris, 2016-03-22:
That sounds about right.

Also, look at the `ShutDownTask` registration in `init()` of the `DefautIdTagManager` to show how to get your store method registered to run at shutdown.

bobjacobsen, 2016-03-22:
See also http://jmri.org/help/en/html/doc/Technical/XmlPersistance.shtml for some background

kevin-dickerson, 2016-03-22:
On 22/03/2016 17:36, Steve Rawlinson wrote:
> By associating memory variables with layout blocks you can display the
> location of locomotives as they move around the layout in the layout
> panel. These locations are not preserved between JMRI [...]
Steve,

This information use to be saved between sessions in a xml file called 
"blockvalues.xml", this is/was done via the 
jmri.jmrit.display.layoutEditor.BlockValueFile.  It saved the value that 
was held against the block, rather than the memory location as not all 
blocks have or need a memory associated with them.  I had added a "Block 
Contents" option to the Layout Editor so that it was possible to do away 
with having to create and then associate a memory icon to display the 
contents of the block.

Regards
Kevin

SteveRawlinson, 2016-03-23:
Aha, thanks Kevin. That is indeed still working. I wondered why we had associated memory values when you can use the block contents directly. I think the LE help page needs to be updated to reflect that. I'll submit a PR later.

There is a slight difference between the memory label display and the block value: block value labels are invisible even in edit mode unless there is a value.

closed 2016-03-25 by SteveRawlinson