JMRI/JMRI #1014: Internal Managers are taking precidence.

rhwood, 2016-03-23:
See the following:
- https://sourceforge.net/p/jmri/mailman/message/34958585/
- https://sourceforge.net/p/jmri/mailman/message/34960302/
- https://sourceforge.net/p/jmri/bugs/549/

bobjacobsen, 2016-03-23:
@pabender 
> XPressNet is actually set as the default, and I tried changing it to internal and then changing it back, and that didn't help.

Thanks. That's probably an important clue. Could you check a couple more things? (Will be back to real computers late this evening, but trying to think this through now)  Does the Internal system show as a system in the Connections preferences? (E.g. is it just XPressNet, or both XPressNet and Internal?)  And when you open the Turnout Table and Sensor Table, that order do the systems appear in?

Thanks again, and sorry for the trouble.

pabender, 2016-03-23:
Sent from my iPad

On Mar 23, 2016, at 2:44 PM, Bob Jacobsen <notifications@github.com> wrote:

> @pabender
>
> XPressNet is actually set as the default, and I tried changing it to internal and then changing it back, and that didn't help.
>
> Thanks. That's probably an important clue. Could you check [...]
For the connections, only the system appears.  This is true whether I have an XPressNet connection or a Loconet connection defined.

Internal does appear on the defaults tab.

> And when you open the Turnout Table and Sensor Table, that order do the systems appear in?
> 
For the Turnout and Sensor tables, the order ( on a system with a Loconet simulator connection ) is:
All, Internal, LocoNet

Interestingly, the Light Table has:
All, Loconet, Internal

And the Reporter Table doesn't have any tabs ( which I don't think is right, because you can create both internal and LocoNet reporters.... )

Paul

rhwood, 2016-03-23:
It is possible that the DefaultsManager needs to get its list of connections from the ConnectionConfigManager instead of consulting the list of SystemConnectionMemos (I forget if that's a static field in the SystemConnectionMemo or handled by the InstanceManager). The ConnectionConfigManager knows only of the connections that are in the profile config, while the SystemConnectionMemo list contains all connections, including those a user did not configure.

Randall Wood
On the road

bobjacobsen, 2016-03-24:
This is pretty messy.  

The underlying problem is that we're gotten two conflicting bits of historical capability:
- internal managers for various things just appear:  You don't need to have an internal system configured to have IT1 or IS1 just work when you ask for it.
- You can (and people have) configure an explicit internal system, or even more than one: It's possible to have IS1 and I2S1 at the same time, or even JS1 referring to an internal Sensor.

This entangles the startup/initialization with the preferences in ways that I haven't yet figured out how to disentagle.

Maybe it's time to just fix it right, so I'm proposing a visible-to-users change. Please poke holes in it. If people on this thread are OK, I'll ask jmri-developers, and _maybe_ jmriusers about it.

- There's always an Internal system at the right side of the list of systems on the Preferences Connections pane.  It's there from a clean start, it stays at that side when you add a new one (e.g. last), and you can't delete it.
- You can add another Internal system if you want, which will appear somewhere else in the order.
- You can change the system letter/string of the special Internal system connection, but (again) you can't make it go away.
- Technically, this is moving the "I noticed you don't have an internal system so I'll create one" code into the preferences handling, and out of everywhere else (four or more places).

Note that this means that non-JMRI app users of JMRI, e.g. CATS et al, might need to adjust their own preferences and/or setup code to make sure the still have internal system and/or manager support.  That's a bit of a migration, but I'll put up a web page on how to do it.

Thoughts?

bobjacobsen, 2016-03-24:
One additional note on migration:
- Profiles without an internal system already explicitly configured will get on appearing on the right side of the list
- Profiles with an internal system on the right side of the connections list will keep it, no change
- Profiles with an internal system somewhere else in the list, but not on the right side, will get a new one on the right side with the first available I_n_ system string.

These will take effect automatically the 1st time (and every time) the profile is used with the new code.  They'll be stored when the profile is next stored. I'm not planning on doing any notification of these changes to the profile. 

Profiles written by the next JMRI release will run on prior (profile-supporting) JMRI versions, but the new Internal system will be present and might cause some user confusion. (It should still work as before, though there are no real guarantees if you go far enough back)

mattharris, 2016-03-24:
This seems a pragmatic solution to me.

The only 'gotcha' is the final scenario you mentioned that _might_ cause a bit of confusion. We'll just have to document that behaviour (and hope that it is actually read).

But, on the whole, this seems to make things a bit more robust and transparent.

pabender, 2016-03-24:
Sent from my iPad

On Mar 24, 2016, at 9:57 AM, Bob Jacobsen <notifications@github.com> wrote:

> Maybe it's time to just fix it right, so I'm proposing a visible-to-users change. Please poke holes in it. If people on this thread are OK, I'll ask jmri-developers, and maybe jmriusers about it. [...]
I have just one potential concern, how does this effect the first time startup wizard?  In other words, we need to detected that the current profile doesn't have any configuration and then create the new internal system, just to make sure the automatically created internal system isn't seen as a connection.  This is complicated to a certain degree because the configuration with only an internal connection is perfectly valid....

Paul

rhwood, 2016-03-24:
> Profiles written by the next JMRI release will run on prior (profile-supporting) JMRI versions, but the new Internal system will be present and might cause some user confusion. (It should still work as before, though there are no real guarantees if you go far enough back)

We can stop updating _ProfileConfig.xml_ if we want (and only write to _profile/profile.*_) This will ensure that older versions of JMRI only get the older config (which might have its own issues), but will also ensure that this change will not be reflected in older versions.

rhwood, 2016-03-24:
If a user wants the connection "None", but gets "None" and "Internal," how do we handle that? Note that the connection type "None" in the preferences UI is an internal connection.

Pugwash1, 2016-04-03:
getDefault in the instance manager returns the LAST item it has as the default. In my case the default is the first. So. changing return (T) l.get(l.size() - 1); to return (T) l.get(0); is my temporary work round. SHouldnt the default entry be flagged as the default rather than position based?

rhwood, 2016-04-03:
@Pugwash1 the InstanceManager is behaving precisely as designed and documented by returning the last item in the list. See the documentation comments immediate preceding that method in the source.

bobjacobsen, 2016-04-03:
Termporary workaround in #1108.   It's in the JMRI/JMRI repository as the bobj-internal-bypass branch and I'd appreciate if people could see if restores working throttles to their setup.

Pugwash1, 2016-04-03:
The temporary work around proposed only fixes the default throttle. The default programmer is also flawed. It would be nice to fix the underlying issue. There is a method for setting the defaults in the instance manager, but it is never called from anywhere that I can find. I can try digging further if someone can tell me where the profile is read and loaded. Right now I'm hunting and pecking.

bobjacobsen, 2016-04-04:
Does setting the default programmer in the preferences work?

If not, the same workaround might work: remove the section that creates and registers the program me when the internal system is creating, deferring until it's requested specifically.

Pugwash1, 2016-04-04:
You can fix the default programmer problem the same way. But you have to comment out two sections of code. One that creates as per normal and the other a place where it creates an internal one if one doesn't exist. After I followed these paths I thought it was easier to give in and just go with the flow and use the fix to pick the first rather than last. 

I can't check setting the default right now, sorry.

pabender, 2016-04-04:
On 04/03/2016 05:38 PM, Bob Jacobsen wrote:
>
> Termporary workaround in #1108 
> <https://github.com/JMRI/JMRI/pull/1108>. It's in the JMRI/JMRI 
> repository as the bobj-internal-bypass branch and I'd appreciate if 
> people could see if restores working throttles to their setup.
>
That fixes the throttles for me.

Now, I went ahead and repeated some tests with JMRI/JMRI master, using 
an XPressNet (LI100F) connection, I tried a few things under both 
PanelPro and DecoderPro, and the results were different.

The throttle works correctly on DecoderPro, but it uses the internal 
device on PanelPro.

The OpsMode Programmer works correctly on DecoderPro , but it uses the 
internal device on PanelPro.

The PowerControl works correctly on DecoderPro, but it uses the internal 
device on PanelPro.

The service mode programmer works correctly in both cases.

So something is broken badly, and it's broken in such a way that 
different applications
Except for the throttles, I get the same results on the 
bobj-internal-bypass branch.

Paul

rhwood, 2016-04-04:
apps.App (PanelPro) and apps.AppsBase (DecoderPro) start their managers in different order. Perhaps there is a clue in there?

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

Pugwash1, 2016-04-04:
Just change the add method of InstanceManager to always insert at the front. The setDefault method reorders stuff anyway. 
This works and appears rock solid. I thought about it. Since the setDefault method does a delete of the entry and add at the end we are already de-referencing the items, so this doesn't change anything. 

-        l.add(item);
+        l.add(0,item);

The best solution is to extend the ArrayList to include a setDefault() and getDefault() so that there is no chance of de-referencing. deleteItem method would do a logical not physical delete. With those changes defaults could be changed on the fly without restarts.

Just an Idea.
Steve G.

bobjacobsen, 2016-04-14:
This has too many tentacles to be handled before the next production release, so I've retagged it.  For a more immediate fix, please see #1108

bobjacobsen, 2016-07-03:
Removed 'afterNextProductionRelease' label as JMRI 4.4 is out.

pabender, 2017-06-17:
I'm not sure this is entirely related to this issue, but I just received a Roco Z21, and when setting up a new profile for the device, The default selections defaulted to the internal connections, not the new hardware connection.

rhwood, 2018-03-07:
I think this is fixed.

closed 2018-03-07 by rhwood