JMRI/JMRI #1023: WiThrottle throwing NameError

SteveRawlinson, 2016-03-24:
Running off the current master branch I can't use WiThrottle. System console has the following error immediately after I open the client on my phone:

2016-03-24 16:38:37,022 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception: java.lang.NoSuchMethodError: jmri.Turnout.getProperty(Ljava/lang/Object;)Ljava/lang/Object;
	at jmri.jmrit.withrottle.TurnoutController.filterList(TurnoutController.java:43)
	at jmri.jmrit.withrottle.AbstractController.buildList(AbstractController.java:57)
	at jmri.jmrit.withrottle.TurnoutController.sendList(TurnoutController.java:114)
	at jmri.jmrit.withrottle.DeviceServer.addControllers(DeviceServer.java:501)
	at jmri.jmrit.withrottle.DeviceServer.<init>(DeviceServer.java:154)
	at jmri.jmrit.withrottle.UserInterface.listen(UserInterface.java:281)
	at jmri.jmrit.withrottle.UserInterface$ServerThread.run(UserInterface.java:457)
 [WiThrottleGUIServer]

The line in question is:

Object o = t.getProperty("WifiControllable");

't' is a Turnout which inherits from NamedBean which certainly does have a getProperty(String) method. So I'd be interested to know what the problem is.

rhwood, 2016-03-24:
Try a clean & rebuild on your local working copy.

Your WiThrottle is looking for a method ```getProperty(Object)``` which does not exist anymore. The method ```getProperty(String)``` _does_ exist, which suggests that you have .class files that predate the change.

SteveRawlinson, 2016-03-24:
Thanks @rhwood that worked.

closed 2016-03-24 by SteveRawlinson