JMRI/JMRI #1043: Should TableSorter be replaced?

rhwood, 2016-03-26:
CI Tests often throw errors on [```jmri.util.com.sun.TableSorter```](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/util/com/sun/TableSorter.java). Are there any reasons that should not be deprecated in favor of [```javax.swing.table.TableRowSorter```](https://docs.oracle.com/javase/8/docs/api/javax/swing/table/TableRowSorter.html)?

```TableSorter``` is based on sample code that has been removed from Oracle's tutorial in favor of ```TableRowSorter```, which was added to Java in version 1.6.

bobjacobsen, 2016-03-26:
I think moving to TableRowSorter is a good idea in any case. As you point out, it's the recommended approach now.

I think it won't actually fix the underlying problem causing the CI failures (non-Swing-thread operations), but it maybe it'll reduce them, and that's a good thing.

rhwood, 2016-04-03:
```javax.swing.RowSorter``` and its subclass ```TableRowSorter``` have a more powerful sorting (and filtering) capability than the TableSorter, but this brings additional complexity when using sortable tables.
However, I am inclined not to attempt to hide this complexity because it brings additional capabilities that are made unusable by hiding it.

bobjacobsen, 2016-06-08:
The new [Deprecations](http://jmri.tagadab.com/jenkins/job/Development/job/Deprecations/61/warnings4Result/type.1926844441/) Jenkins job shows the references to the already-deprecated TableSorter class.

rhwood, 2016-08-28:
As of #1886, this is substantially complete. With the exception of the jmri.jmrit.operations package and public API that has been deprecated and is unused within the JMRI code (except by other deprecated APIs), TableSorter usage has been removed.

DanielBoudreau, 2016-08-28:
I tried to replace TableSorter but I had issues with either saving or loading the table column order.  I don’t think any of the other JMRI code is saving the user’s selected column order.

Dan

rhwood, 2016-08-28:
The RosterTable is saving the table state; some beans tables should be as well (most never did).

Note that if you do use the techniques in the RosterTable, you will need to setup the default column widths before loading them storage.

See the JTablePersistenceManager JavaDocs for details of how to save table state (note that it will not work with TableSorters)--essentially you set it up to save the table state, and it automatically saves it for you as needed.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

DanielBoudreau, 2016-08-28:
Thanks Randall,

I’ll take a look at RosterTable to see how column order is save and loaded.

Dan

randall.h.wood@alexandriasoftware.com<mailto:randall.h.wood@alexandriasoftware.com>
http://alexandriasoftware.com

DanielBoudreau, 2016-08-28:
A quick test of the RosterTable shows that the column order is saved/restored properly, but the column width isn’t saved/restored.  Was this done intentionally?  I want to do both.

Dan

http://alexandriasoftware.com

rhwood, 2016-08-28:
That's not intentional. I'm still figuring out how write all the unit tests.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

rhwood, 2016-08-29:
@DanielBoudreau #1887 should fix the issues you observed.

DanielBoudreau, 2016-08-29:
Your fix looks good.  Column order and width now saved and restored correctly.

Thanks,

Dan

rhwood, 2016-09-06:
This is done other than removing deprecated APIs (although that will have to wait until after 4.8 / 5.0).

closed 2016-09-06 by rhwood