JMRI/JMRI #1097: Loconet throttles have stopped working

SteveRawlinson, 2016-04-01:
I'm running a clean build from master as of last night. No loconet messages from JMRI throttles are making it onto the layout (or being seen by loconet monitor). Reverting to 4.3.4 fixes the problem.

Other loconet messages are getting out (signals and turnout motors are working) and throttle messages from the command station are working fine (and being seen by loconet monitor). Sensors are working. It's just messages from the JMRI throttles that are disappearing.

I'm using windows 7, a DCS50 Zephyr attached via a Locobuffer USB.

mstevetodd, 2016-04-01:
This sounds like very similar symptoms to #1014

SteveRawlinson, 2016-04-01:
You're right, I think it's the same issue. Closing this for now.

closed 2016-04-01 by SteveRawlinson