JMRI/JMRI #1111: Target engine (not) selected after positive Identification on Roster view

AlanUS, 2016-04-04:
In the current version of JMRI (4.3.4), from the Roster view (Decoder Pro), after a positive engine identification using the "Identify" button, the target loco is properly displayed in the lower part of the Roster window, **but ** the loco in _not_ selected in the list. Therefore any other action such as Program or Throttle results in a "no selection" message. One needs to manually select the engine.
It seems to me that in previous releases of JMRI (I couldn't tell which) the engine was properly selected in the list, which is what is expected from the Identify function.
Is this a regression or just my imagination?

rhwood, 2016-04-21:
I am unable to reproduce this problem in a build off master.

@AlanUS Can you verify this does/does not occur in the most recent nightly build (http://builds.jmri.org/jenkins/job/Development/job/Packages/lastBuild/)?

AlanUS, 2016-04-24:
@rhwood I'm away from home for another week, so will try next week end.

rhwood, 2016-05-02:
I am sometimes seeing a loss of selection in the Roster view, but have not yet identified a repeatable sequence to loose the selection.

bobjacobsen, 2016-05-03:
I think this is a "using Swing from the wrong thread" issue.  I caught it just once, and it looked like that. #1268 ensures that ident ends on the GUI thread, we'll see if that helps.

AlanUS, 2016-05-03:
I think that I have found a repeatable sequence to reproduce / remove the problem. I just tested with latest update from GIT repository, so the issue is still in as I write this.

1. In Roster view, move the "ID" column to any position other than first one (left-hand side).
Identification of loco provides a result in the Summary pane, but the identified loco is not selected in the Roster view.
Note that if a loco is already selected in the Roster view, the identification fails (nothing displayed in the summary pane) and the previously selected loco is deselected. A second hit on Identify will find the loco, in the summary pane, but still not select it.

2. Move the "ID" column back in first position (left-hand side) and things go back to normal: loco is properly identified in summary pane and selected in Roster list.

Hope that this will help finding the issue quickly.

bobjacobsen, 2016-05-04:
@AlanUS thanks!  I think the fix is in #1275

AlanUS, 2016-05-04:
Just tested #1275 
Seems to fix the issue. I couldn't reproduce it, and got the identified loco selected each time.
@bobjacobsen Thanks!

closed 2016-05-04 by AlanUS