JMRI/JMRI #1131: AppVeyor CI test will be offline, probably at least until April 11

bobjacobsen, 2016-04-08:
The AppVeyor CI tests are failing while trying to download a new copy of ant from Apache via http://archive.apache.org/dist/ant/binaries/apache-ant-1.9.6-bin.zip

That host is currently down:
````
This service is currently undergoing maintenance!

We are in the process of relocating this service. To save on the immense bandwidth that this service outputs, we have put it in maintenance mode, disabling all downloads for the next few days. We expect the maintenance to be complete no later than the morning of Monday the 11th of April, 2016.

With regards, the Apache Infrastructure Team.
````

This means the AppVeyor CI tests are going to show as failed until either that server comes back up in a few days, or we redo (or disable) our AppVeyor setup.

rhwood, 2016-04-09:
I’ve temporarily disabled the AppVeyor tests and will re-enable them on Monday.

> The AppVeyor CI tests are failing while trying to download a new copy of ant from Apache via http://archive.apache.org/dist/ant/binaries/apache-ant-1.9.6-bin.zip
>
> That host is currently down:
>
> This [...]

Randall Wood
Alexandria Software
202.683.8604
randall.h.wood@alexandriasoftware.com
http://www.alexandriasoftware.com

rhwood, 2016-04-10:
AppVeyor builds have been re-enabled and are confirmed to be working.

> 
> I’ve temporarily disabled the AppVeyor tests and will re-enable them on Monday.
> 
>> 
>> The AppVeyor CI tests are failing while trying to download a new copy of ant from Apache via http://archive.apache.org/dist/ant/binaries/apache-ant-1.9.6-bin.zip
>> 
>> That host is currently down:
>> 
>> This service is currently undergoing maintenance!
>> 
>> We are in the process of relocating this service. To save on the immense bandwidth that this service outputs, we have put it in maintenance mode, disabling all downloads for the next few days. We expect the maintenance to be complete no later than the morning of Monday the 11th of April, 2016.
>> 
>> With regards, the Apache Infrastructure Team.
>> 
>> This means the AppVeyor CI tests are going to show as failed until either that server comes back up in a few days, or we redo (or disable) our AppVeyor setup.
>> 
>> 
> 
> Randall Wood
> Alexandria Software
> 202.683.8604
> randall.h.wood@alexandriasoftware.com
> http://www.alexandriasoftware.com

Randall Wood
Alexandria Software
202.683.8604
randall.h.wood@alexandriasoftware.com
http://www.alexandriasoftware.com

closed 2016-04-11 by rhwood