JMRI/JMRI #1142: Block value / Reporter problem

dsand47, 2016-04-10:
The block value is updated when its associated reporter report changes even though the **Use Current?** option is not active.

Is this a bug or a misunderstanding on my part?

Currently using JMRI 4.3.4.

Dave Sand

bobjacobsen, 2016-04-10:
Not entirely sure what you're asking.  There's a note in the Block JavaDoc that says:
````
 * Optionally, a Block can be associated with a Reporter. In this case, the
 * Reporter will provide the Block with the "token" (value). This could be e.g
 * an RFID reader reading an ID tag attached to a locomotive. Depending on the
 * specific Reporter implementation, either the current reported value or the
 * last reported value will be relevant - this can be configured
````

That points to Reporter, which says in the summary:
````
 * A Reporter might also not be able to report all the time. The previous value
 * remains available, but it's also possible to distinquish this case by using
 * the getCurrentReport member function.
````

and has two relevant methods:
````
    /**
     * Query the last report. This will return a value even if there's no
     * current report available. If there is a current report, both this and the
     * current report will be equal. If nothing has ever been reported, this
     * will return a null object.
     */
    public Object getLastReport();

    /**
     * Query the current report. If there is no current report available (e.g.
     * the reporting hardware says no information is currently available) this
     * will return a null object.
     */
    public Object getCurrentReport();
````

I read all this to say that there are options _between_ reports, but that the value definitely should update when new info comes from the layout.  

Is that not what you're seeing?

(N.B. that "last" in "last report" in the Block JavaDoc is what's called  a "New York last":  It means the last one that's before now, not the one before that: It's _not_ "the one before the most recent one".  By "current" it means "right now".  So if "ABC" was reported at 1:30PM, and there's nothing available from the layout now at 1:40PM, then the last is "ABC" and the current is "" (nothing).  If "ABC" was reported at 1:30PM, and that's still available now, then both last and current are "ABC". )

dsand47, 2016-04-10:
There is an option, **Use Current?**, on the block table as well as the Edit dialog for a layout block that appears to control whether the block value should be updated by a change to the Reporter value.

[image]

Any changes to the Reporter value are copied to the block value regardless of the state of the option.

I have come up with another scenario which would explain the symptoms.  The **last** reported value is normally copied to the block value.  The **Use Current?** option is used to force the current value to be used even when it does not match the last report value.  If this is the case, then that explains my misunderstanding.

bobjacobsen, 2016-04-10:
Could you give a more detailed scenario of what you're seeing, please?  E.g. what happens when.  In particular, I don't understand what you're seeing as "the current value" as opposed to the "the last reported value".

"Any changes to the Reporter value are copied to the block value regardless of the state of the option." sounds exactly correct for either setting of that option.  The question is the value _between_ the changes.

More info:
The whole "current" thing in _Reporter_ is based on variations in hardware: some can and some can't be queried, so have values between reports and some don't.  If you specifically call the status method on a Reporter, it can or can't check with the hardware to see if there's a value right now.  The Reporter option is about how to handle all those possibilities, particularly if there isn't a value at that moment.  I don't know how Blocks would work with that:  They're certainly not calling the Reporter all the time to check the value over and over (at least, I hope not).

pabender, 2016-04-10:
Sent from my iPad

On Apr 10, 2016, at 6:34 PM, dsand47 <notifications@github.com> wrote:

> There is an option, Use Current?, on the block table as well as the Edit dialog for a layout block that appears to control whether the block value should be updated by a change to the Reporter [...]
When a new value is sent to a reporter object from the layout, then both the current report and the last report change.

When a reporter changes from a report to no report, the current report changes to "no report" from a report value, then the current report is set to "no report", but the last report value stays the same.

Use current forces the current report to be used, even if it shows no report.

Paul

dsand47, 2016-04-11:
Thanks to Bob and Paul, I now see that what appeared to me to be a bug was actually a lack of understanding of the data flow within the Reporter processes.

bobjacobsen, 2016-04-11:
Is there a way we could improve the phrasing on the window (or the code docs, though most people don't see those) to make confusion less likely for the next person?

dsand47, 2016-04-11:
When using the Layout Editor Block Edit function, something like the following might work:

Old:  Use the Current Report from the reporter as the value for the block
New:  Use the Current Report from the reporter instead of the last valid report

But even that might be confusing since most people will think of **last** as being **previous**.  Maybe **latest** would be better than **last**.  It also raises the question:  What does  **valid** mean?

The web page at http://jmri.org/help/en/package/jmri/jmrit/beantable/BlockEdit.shtml as well as the JMRI Block Table Help does a pretty good job of describing the process.  It is too bad that I did not see it earlier.

The JavaDocs for the **Block** class has a paragraph about the Reporter which talks about **Current** vs. **Last**, but it does not go into detail since that is Reporter implementation specific.

Dave Sand

closed 2016-05-01 by bobjacobsen