JMRI/JMRI #1164: XML check, validate not in DecoderPro(3) Action menu

bobjacobsen, 2016-04-14:
PanelPro and original-DecoderPro had "check xml file" and "validate xml file" in the Debug menu.  A recent jmriusers post noted that the instructions for creating a new decoder file refer to those:

````
JMRI: DecoderPro User Guide - Creating A Custom Decoder File http://jmri.sourceforge.net/help/en/html/apps/DecoderPro/CreateDecoder.shtml

JMRI: DecoderPro User Guide - Creating A Custom Decode... http://jmri.sourceforge.net/help/en/html/apps/DecoderPro/CreateDecoder.shtml
````

But there's no Debug menu in current DecoderPro (nee DP3), and the two items don't appear in the Action menu (which might be the logical replacement)

- [ ] Should those be there?
- [ ] If so, which menu should they appear in?

AlanUS, 2016-04-17:
As several items of the "debug" menu of PanelPro (and original DP) have been moved already to the "File" menu of current DP, I would vote for adding Check and Validate XML files in the "File" menu, as these are actions on individual files.

As several help pages have to be modified, referring to the original DP or the current DP (nee DP3) depending on context, can we agree what vocabulary to use, when a distinction is needed. 

I have used "legacy" DecoderPro (for original DP) and DecoderPro "Roster view" (for current DP nee DP3) in a recent update of the following page:
http://jmri.sourceforge.net/help/en/html/apps/DecoderPro/FileUpdate.shtml

Maybe would be better to refer to JMRI version (up to 3.x, from 4.x) instead.

rhwood, 2016-04-17:
I would not make any new references to the second generation DecoderPro (DecoderPro in JMRI versions 2.x and 3.x) in documentation, and refer only to the DecoderPro that is currently shipping and to the PanelPro that is currently shipping.

But I agree, these are generic file actions, and handling them should be under the File menu.

closed 2016-05-01 by bobjacobsen