JMRI/JMRI #1167: State issue when using JUnitUtil.initConfigureManager

pabender, 2016-04-15:
There's some kind of state contamination occurring in headlesstest that is masking an issuethat occurs when using JUnitUtil.initConfigureManager in tests.

If you use runtest to run a single test (or suite of tests) that calls the JUnitUtil.initConfigureManager, the following stack trace is produced:
java.util.ServiceConfigurationError: jmri.spi.PreferencesProvider: Provider apps.StartupActionsManager not a subtype
	at java.util.ServiceLoader.fail(ServiceLoader.java:239)
	at java.util.ServiceLoader.access$300(ServiceLoader.java:185)
	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:376)
	at java.util.ServiceLoader$LazyIterator.next(ServiceLoader.java:404)
	at java.util.ServiceLoader$1.next(ServiceLoader.java:480)
	at jmri.implementation.JmriConfigurationManager.<init>(JmriConfigurationManager.java:41)
	at jmri.util.JUnitUtil.initConfigureManager(JUnitUtil.java:190)	
        at jmri.jmrix.roco.z21.Z21XPressNetTunnelTest.setUp(Z21XPressNetTunnelTest.java:65)

If you run headlesstest (and I assume alltest, though I haven't done that yet) there is no error. 

In other words, if the test fails when run using runtest, it should also fail when run using headlesstest, but it's not, so there is some state left behind that hasn't been cleaned up when these tests start in headlesstest.  We need to find what that state is and devise a way to clean the state at the beginning of each unit test.

rhwood, 2016-04-15:
I don't think this is a state error—it appears the Java ServiceLoader (which performs its initialization before any ```main()``` method is run) is not functioning correctly in runtest.csh (BTW, its not a csh script, why the claim that it is?).

Are you certain that runtest.csh has the same class paths as ant?

rhwood, 2016-04-15:
Specificly, do you have a jmri.jar file in your class path when running runtest.csh?

rhwood, 2016-04-15:
Sorry, that first comment of mine is wrong - the ServiceLoader is lazy and creates providers on demand, but I think it does determine what to provide very early. It may have issues if the same class appears twice in the class path (in jmri.jar and java/classes).

pabender, 2016-04-15:
On Apr 15, 2016, at 5:57 AM, Randall Wood <notifications@github.com> wrote:
> Specificly, do you have a jmri.jar file in your class path when running runtest.csh?
> 
No, there is no jmri.jar file in the path.

Paul

pabender, 2016-04-15:
On Apr 15, 2016, at 5:08 AM, Randall Wood <notifications@github.com> wrote:
> I don't think this is a state error—it appears the Java ServiceLoader (which performs its initialization before any main() method is run) is not functioning correctly in runtest.csh (BTW, its not a csh script, why the claim that it is?).
> 
Bob will need to answer that question.
> Are you certain that runtest.csh has the same class paths as ant?
> 
Am I positive, no, but if there was something added to the ant class paths used for testing, they should have also been added to runtest.  That is a part of our testing infrastructure.

Paul

bobjacobsen, 2016-04-15:
> If you use runtest to run a single test (or suite of tests) that calls the JUnitUtil.initConfigureManager, the following stack trace is produced:
> java.util.ServiceConfigurationError: jmri.spi.PreferencesProvider: Provider apps.StartupActionsManager not a subtype

There's a class loader issue underlying this that I've never been able to track down. 

There's a workaround: add "-no loading" to each standalone test, see the bottom of http://jmri.org/help/en/html/doc/Technical/JUnit.shtml

I don't want to do that generally, because it prevents the reload-and-go JUnit feature from working right, which causes its own confusion. But I haven't been able to figure out which classes to add to the no-loader list.

pabender, 2016-04-15:
On 04/15/2016 09:14 AM, Bob Jacobsen wrote:
>> If you use runtest to run a single test (or suite of tests) that
>> calls the JUnitUtil.initConfigureManager, the following stack trace is
>> produced:
>> java.util.ServiceConfigurationError: jmri.spi.PreferencesProvider:
 >> Provider apps.StartupActionsManager not a subtype
>
> There's a class loader issue underlying this that I've never been able
> to track down.
>
> There's a workaround: add "-no loading" to each standalone test, see the
> bottom of http://jmri.org/help/en/html/doc/Technical/JUnit.shtml

The 4 tests that I'm working with now and seeing this on already have 
-noloading enabled:

     // Main entry point
     static public void main(String[] args) {
         String[] testCaseName = {"-noloading", 
Z21SimulatorAdapterTest.class.getName()};
         junit.swingui.TestRunner.main(testCaseName);
     }

I removed -noloading from the two test classes causing an issue right 
now (all in the jmri.jmrix.roco.z21 package) and that didn't change 
anything.

Paul

bobjacobsen, 2016-04-15:
@pabender:
> The 4 tests that I'm working with now and seeing this on already have 
> -noloading enabled:
> 
> 
> // Main entry point
> static public void main(String[] args) {
> String[] testCaseName = {"-noloading", 
> Z21SimulatorAdapterTest.class.getName()};
> junit.swingui.TestRunner.main(testCaseName);
> }

Could you point me to those in Git?  I don’t see them in JMRI/JMRI or pabender/JMRI master.  Would like to debug this a bit. Thanks.

bobjacobsen, 2016-04-15:
Nevermind, sorry for the confusion, found it in jmri.jmrix.roco.z21.simulator.Z21SimulatorAdapterTest

bobjacobsen, 2016-04-15:
Now it’s more of a mystery:

````
% ./runtest.csh jmri.jmrix.roco.z21.simulator.Z21SimulatorAdapterTest
ls: *.jar: No such file or directory
log4j:WARN No appenders could be found for logger (jmri.InstanceManager).
log4j:WARN Please initialize the log4j system properly.
log4j:WARN See http://logging.apache.org/log4j/1.2/faq.html#noconfig for more info.
(GUI shows all passed)

JMRI/projects/JMRI/ % git status
HEAD detached at pabender/master
````

So on my Mac (10.10.5) command line w Java 1.8.0_25 I don’t see the problem.  Is there perhaps code on your master that hasn’t been pushed?

pabender, 2016-04-15:
On Fri, Apr 15, 2016 at 9:58 AM, Bob Jacobsen <notifications@github.com>
wrote:

>
> So on my Mac (10.10.5) command line w Java 1.8.0_25 I don’t see the
> problem. Is there perhaps code on your master that hasn’t been pushed?
>
>
> The only things I have on my machine that are modified are a default.lcf
that adds debugging log information for a few things I've been working on
and a build.xml that sets the correct RxTx serial port string for my linux
machine.

Paul

rhwood, 2016-04-15:
Which test fails?

If you run it out of the junit4 branch in my fork, do the results change?

pabender, 2016-04-15:
On Apr 15, 2016, at 5:15 PM, Randall Wood <notifications@github.com> wrote:
> Which test fails?
> 
It isn't actually the tests that are failing.   

The error occurs when JUnitUtil.initConfigureManager is called in the setup routine.  Because the error occurs during setup, all test in the file fail.

The two files I am working with now are 

java/test/jmri/jmrix/roco/z21/Z21XPressNetTunnelTest.java

And

java/test/jmri/jmrix/roco/z21/simulator/Z21SimulatorAdapterTest.java

And no, removing the call to JUnitUtil.initConfigureManager is not an option.  Without that, there is a null pointer exception in the tests themselves because they construct objects that need the ConfigureManager.
> If you run it out of the junit4 branch in my fork, do the results change?
> 

I have not had time to check.  I might have time to try that after my daughter goes to bed tonight, but if I don't I probably won't get to it until Monday.

Paul

pabender, 2016-04-16:
On 04/15/2016 05:15 PM, Randall Wood wrote:
> If you run it out of the junit4 branch in my fork, do the results change?

Yes, the same error occurs on your junit4 branch, except that your 
branch only has one of the two test files (the 2nd was added to the 
master branch today).

Paul

pabender, 2016-04-16:
On 04/15/2016 06:58 PM, Paul Bender wrote:
> On Apr 15, 2016, at 5:15 PM, Randall Wood <notifications@github.com
> <mailto:notifications@github.com>> wrote:
>>
>> Which test fails?
>>
> It isn't actually the tests that are failing.
>
> The error occurs when JUnitUtil.initConfigureManager is called in the
> setup routine.  Because the error occurs during setup, all test in the
> file fail.
>
> The two files I am working with now are
>
> java/test/jmri/jmrix/roco/z21/Z21XPressNetTunnelTest.java
>
> And
>
> java/test/jmri/jmrix/roco/z21/simulator/Z21SimulatorAdapterTest.java

Casting a bit of a wider net here, running jmri.jmrix.PackageTest 
doesn't show a problem, so I did some investigation.

There are 4 files that have a call to initConfigureManager in jmri.jmrix 
and it's subpackages:

./roco/z21/Z21XPressNetTunnelTest.java
./roco/z21/simulator/Z21SimulatorAdapterTest.java
./rps/swing/polling/PollTableActionTest.java
./loconet/locomon/LocoMonPaneTest.java

When run individually, the first 3 show an error, but the 4th one does not.

Also, if the 4th test is run first, and then one of the other 3 is run, 
then there isn't an error.

So ./loconet/locomon/LocoMonPaneTest.java has a side effect that allows 
the tests to run cleanly.  I haven't found what that is however.  The 
LocoNet tests happen before either the roco or rps tests, so that 
explains why we don't see this in alltest.

There are also tests in the jmri package that use initConfigureManager, 
so one of those tests is allowing the Z21 tests to complete successfully 
on headlesstest, but I haven't tracked down which one yet.

Paul

pabender, 2016-04-19:
More work on this tonight.

I have eliminated the calls to JUnitUtil.initConfigureManager for test classes in which removing it had no effect on the tests.  That leaves us with categories of tests that use JUnitUtil.initConfigureManager.

The following test classes (paths relative to test/jmri/) fail when run independently:
./server/json/roster/JsonRosterSocketServiceTest.java
./server/json/roster/JsonRosterHttpServiceTest.java
./jmrix/roco/z21/Z21XPressNetTunnelTest.java (though I have a temporary workaround that eliminates the call to initConfigureManager()  right now).
./jmrix/roco/z21/simulator/Z21SimulatorAdapterTest.java
./configurexml/SectionManagerXmlTest.java
./configurexml/BlockManagerXmlTest.java

The following test classes succeed when run independently:
./jmrit/display/PanelEditorTest.java
./jmrit/display/SignalSystemTest.java
./jmrit/display/PositionableLabelTest.java
./jmrit/logix/NXWarrantTest.java
./jmrit/logix/LearnWarrantTest.java
./jmrit/logix/LogixActionTest.java
./jmrit/symbolicprog/CombinedLocoSelListPaneTest.java
./configurexml/LoadAndStoreTestBase.java (this one is actually a base class for several others).
./configurexml/LoadAndCheckTest.java

All of the tests succeed when run via headlesstest.

All the tests that fail do so with a stack trace like the one I noted at the beginning of this thread.

Also, just to verify this isn't a machine dependent failure, I did double check and make certain I can replicate this on multiple machines.  In my case it was two flavours of Linux (a Fedora machine and an Ubuntu machine).

I still don't understand why some test are failing and others are not however.  The setUp routine's for these all look similar to me.

Paul

pabender, 2016-04-20:
Ok, so I think I figured this out.

Tonight I went through the failing tests again, and double checked the "-noloading" option was set.  Once set, the tests (even the Z21 ones that started all this) started working again.

The reason I didn't catch this the other day is that I was running the package test for the Z21.  The package test did not have the "-noloading" option.  Once added, I can no run both the individual tests and the tests for the Z21 package without a failure. 

Paul

pabender, 2016-04-20:
closed via PR #1182

closed 2016-04-20 by pabender