JMRI/JMRI #1187: Hangs if DCC++ port not found

fiz1962, 2016-04-21:
On Windows 10 and PanelPro 4.3.4-Reb510be.  Connect Arduino DCC++ and configure as COM4.  Later connect Arduino as COM5.  Running PanelPro and selecting profile leads to dialog saying "COM4 not found while opening connection".  Clicking OK then hangs at splash with JMRI logo and text "Press F8 to disable Logixs".

rhwood, 2016-04-21:
Can you please attach a copy of C:\Users\your-user-name\JMRI\log\session.log to this issue? You may have to copy it to session.log.txt before you can attach it.

fiz1962, 2016-04-21:
At this point, JMRI is hung at the splash(?) screen with the Arduino plugged in a different USB port which comes up as COM5 rather than COM4.
=======================================
2016-04-21 10:51:40,100 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-04-21 10:51:40,229 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\fiz60_000\JMRI\log\messages.log [main]
2016-04-21 10:51:40,229 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\fiz60_000\JMRI\log\session.log [main]
2016-04-21 10:51:40,229 apps.Apps                             INFO  - PanelPro version 4.3.4-Reb510be starts under Java 1.8.0_77 at Thu Apr 21 10:51:40 CDT 2016 [main]
2016-04-21 10:51:50,222 apps.Apps                             INFO  - Starting with profile TrainClub.3ea46180 [main]
2016-04-21 10:51:50,574 node.NodeIdentity                     INFO  - Using jmri-645A04A6A748-3ea46180 as the JMRI Node identity [main]
2016-04-21 10:51:50,720 serial.DCCppAdapter                   ERROR - Serial port COM4 not found [main]
2016-04-21 10:51:50,722 configurexml.ErrorHandler             ERROR - COM4 not found while opening connection [main]
2016-04-21 10:51:53,818 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception: java.lang.NullPointerException
	at jmri.jmrix.dccpp.DCCppThrottleManager.<init>(DCCppThrottleManager.java:35)
	at jmri.jmrix.dccpp.DCCppSystemConnectionMemo.getThrottleManager(DCCppSystemConnectionMemo.java:98)
	at jmri.jmrix.dccpp.DCCppSystemConnectionMemo.get(DCCppSystemConnectionMemo.java:267)
	at jmri.managers.ManagerDefaultSelector.configure(ManagerDefaultSelector.java:154)
	at jmri.managers.ManagerDefaultSelector.initialize(ManagerDefaultSelector.java:205)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:259)
	at jmri.implementation.JmriConfigurationManager.lambda$3(JmriConfigurationManager.java:255)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(Unknown Source)
	at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:254)
	at jmri.implementation.JmriConfigurationManager.lambda$2(JmriConfigurationManager.java:188)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(Unknown Source)
	at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:187)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:176)
	at apps.Apps.<init>(Apps.java:276)
	at apps.PanelPro.PanelPro.<init>(PanelPro.java:48)
	at apps.PanelPro.PanelPro.main(PanelPro.java:110)
 [main]

rhwood, 2016-04-21:
@msunderwd Can you take a look at this?

msunderwd, 2016-04-21:
Yes. I am looking into it but haven't figured it out yet.

- Mark

fiz1962, 2016-04-21:
This may have been fixed in the current source code on github, but not yet in the latest Windows exe. When I download the source and run it in NetBeans, it continues and shows "DCC++: ERROR DCC++ SERIAL PORT USING COM4" on the main PanelPro page.  It no longer hangs.  Here are a few lines of the Output in NetBeans.  The uncaught exception is gone.

2016-04-21 13:58:15,083 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3edbb29a [main]
2016-04-21 13:58:16,637 node.NodeIdentity                     INFO  - Using jmri-D4BED94B59BF-3edbb29a as the JMRI Node identity [main]
2016-04-21 13:58:17,704 serial.DCCppAdapter                   ERROR - Serial port COM7 not found [main]
2016-04-21 13:58:17,708 configurexml.ErrorHandler             ERROR - COM7 not found while opening connection [main]
2016-04-21 13:58:24,091 jmri.InstanceManager                  ERROR - Should not store null value of type jmri.CommandStation [main]
java.lang.Exception: Traceback

rhwood, 2016-04-22:
> "DCC++: ERROR DCC++ SERIAL PORT USING COM4"

@fiz1962 Does this same error show in the development build downloaded from http://builds.jmri.org/jenkins/job/Development/job/Packages/3304/ ?

fiz1962, 2016-04-22:
This (above) build seems to work. If I have the DCC++ Arduino plugged in the correct serial port, it comes up.  If the DCC++ Arduino is NOT plugged in PanelPro still comes up, but has a message "DCC++: ERROR DCC++ SERIAL PORT USING COM3" in a red font, and the Preferences/DCC++ tab is red along with COM3 in red. I think this is the way it should behave.  Thanks

rhwood, 2016-04-22:
@fiz1962 thanks for testing; since that appears to be working correctly, I'm going to close this.

@msunderwd do we need language in the release notes concerning this?

closed 2016-04-22 by rhwood

msunderwd, 2016-04-22:
Randall,

Yes, a note in the Release Notes would be helpful. Just a comment that
"DCC++ crash when starting with the Base Station not attached has been
fixed." or similar language should suffice.

Unfortunately, I don't really have access to the repository right now.
Would you mind adding the comment?

Thank you,

Mark

rhwood, 2016-04-23:
> Yes, a note in the Release Notes would be helpful.
[…]
> Would you mind adding the comment?

Done.