JMRI/JMRI #1190: Startup exceptions

DanielBoudreau, 2016-04-22:
Every once in awhile I'm seeing exceptions when starting JMRI:

2016-04-21 20:26:24,189 apps.Apps                             ERROR - Error trying to setup preferences 9 [initialize preferences]
java.lang.ArrayIndexOutOfBoundsException: 9
	at java.util.ArrayList.elementData(ArrayList.java:418)
	at java.util.ArrayList.get(ArrayList.java:431)
	at javax.swing.JTabbedPane.getComponentAt(JTabbedPane.java:1224)
	at javax.swing.JTabbedPane.indexOfComponent(JTabbedPane.java:1719)
	at javax.swing.plaf.basic.BasicTabbedPaneUI$Handler.componentAdded(BasicTabbedPaneUI.java:3716)
	at java.awt.Container.processContainerEvent(Container.java:2260)
	at java.awt.Container.processEvent(Container.java:2231)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.Container.addImpl(Container.java:1136)
	at javax.swing.JTabbedPane.insertTab(JTabbedPane.java:724)
	at javax.swing.JTabbedPane.addTab(JTabbedPane.java:798)
	at javax.swing.JTabbedPane.add(JTabbedPane.java:882)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.newMessageTab(UserMessagePreferencesPane.java:170)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.setMinimumMessagePref(UserMessagePreferencesPane.java:87)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.<init>(UserMessagePreferencesPane.java:62)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:408)
	at java.lang.Class.newInstance(Class.java:438)
	at apps.gui3.TabbedPreferences.init(TabbedPreferences.java:160)
	at apps.Apps$3.run(Apps.java:386)
	at java.lang.Thread.run(Thread.java:745)
2016-04-21 20:26:24,189 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.ArrayIndexOutOfBoundsException: 9
	at java.util.ArrayList.elementData(ArrayList.java:418)
	at java.util.ArrayList.get(ArrayList.java:431)
	at javax.swing.JTabbedPane.getMnemonicAt(JTabbedPane.java:1246)
	at javax.swing.plaf.basic.BasicTabbedPaneUI.updateMnemonics(BasicTabbedPaneUI.java:523)
	at javax.swing.plaf.basic.BasicTabbedPaneUI.access$1300(BasicTabbedPaneUI.java:54)
	at javax.swing.plaf.basic.BasicTabbedPaneUI$Handler.updateHtmlViews(BasicTabbedPaneUI.java:3595)
	at javax.swing.plaf.basic.BasicTabbedPaneUI$Handler.componentAdded(BasicTabbedPaneUI.java:3716)
	at java.awt.Container.processContainerEvent(Container.java:2260)
	at java.awt.Container.processEvent(Container.java:2231)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.Container.addImpl(Container.java:1136)
	at javax.swing.JTabbedPane.insertTab(JTabbedPane.java:724)
	at javax.swing.JTabbedPane.addTab(JTabbedPane.java:798)
	at javax.swing.JTabbedPane.add(JTabbedPane.java:882)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.newMessageTab(UserMessagePreferencesPane.java:170)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.refreshOptions(UserMessagePreferencesPane.java:400)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.lambda$0(UserMessagePreferencesPane.java:57)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane$$Lambda$85/1702073703.propertyChange(Unknown Source)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at jmri.beans.Bean.lambda$6(Bean.java:149)
	at jmri.beans.Bean$$Lambda$17/1908143486.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:744)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

This isn't repeatable, and when this does happen I can't edit preferences.

Dan

DanielBoudreau, 2016-04-22:
Tried a couple of time and got this one:

2016-04-21 20:33:02,731 apps.Apps                             ERROR - Error trying to setup preferences -1 [initialize preferences]
java.lang.ArrayIndexOutOfBoundsException: -1
	at java.util.ArrayList.elementData(ArrayList.java:418)
	at java.util.ArrayList.get(ArrayList.java:431)
	at javax.swing.JTabbedPane.getTitleAt(JTabbedPane.java:1112)
	at javax.swing.plaf.basic.BasicTabbedPaneUI$Handler.updateHtmlViews(BasicTabbedPaneUI.java:3581)
	at javax.swing.plaf.basic.BasicTabbedPaneUI$Handler.componentAdded(BasicTabbedPaneUI.java:3716)
	at java.awt.Container.processContainerEvent(Container.java:2260)
	at java.awt.Container.processEvent(Container.java:2231)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.Container.addImpl(Container.java:1136)
	at javax.swing.JTabbedPane.insertTab(JTabbedPane.java:724)
	at javax.swing.JTabbedPane.addTab(JTabbedPane.java:798)
	at javax.swing.JTabbedPane.add(JTabbedPane.java:882)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.newMessageTab(UserMessagePreferencesPane.java:170)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.setMinimumMessagePref(UserMessagePreferencesPane.java:87)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.<init>(UserMessagePreferencesPane.java:62)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:408)
	at java.lang.Class.newInstance(Class.java:438)
	at apps.gui3.TabbedPreferences.init(TabbedPreferences.java:160)
	at apps.Apps$3.run(Apps.java:386)
	at java.lang.Thread.run(Thread.java:745)

DanielBoudreau, 2016-04-22:
And this one:

2016-04-21 20:34:22,588 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.newMessageTab(UserMessagePreferencesPane.java:230)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.refreshOptions(UserMessagePreferencesPane.java:400)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.lambda$0(UserMessagePreferencesPane.java:57)
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane$$Lambda$85/304127050.propertyChange(Unknown Source)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at jmri.beans.Bean.lambda$6(Bean.java:149)
	at jmri.beans.Bean$$Lambda$17/1071097621.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:744)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

bobjacobsen, 2016-04-23:
I've seen a couple of these now.  MacOS X 10.11.4, Java build 1.8.0_25-b17

It seems to be a simultaneous-notification problem. The best hint so far is Daniel's 2nd trace, which shows a non-AWT-EventQueue-thread access to Swing objects:  That can cause these kinds of conflicts.

apps.Apps line 386 is where the problem is coming from:
```` java
        /*Once all the preferences have been loaded we can initial the preferences
         doing it in a thread at this stage means we can let it work in the background*/
        Runnable r = new Runnable() {
            @Override
            public void run() {
                try {
                    InstanceManager.tabbedPreferencesInstance().init();
                } catch (Exception ex) {
                    log.error("Error trying to setup preferences {}", ex.getLocalizedMessage(), ex);
                }
            }
        };
        Thread thr = new Thread(r, "initialize preferences");
        thr.start();
````

That init() call is supposed to do initialization of Swing objects _only_, without any of them becoming visible.  There are a couple things that can go wrong:

- That code ends up doing setVisible(true)
- That code ends up attaching Swing objects to something that's already visible
- The objects that code is working on are setVisible(true) before this new thread is done with its work.

I don't know enough about how jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane works to know which of those is most likely.

rhwood, 2016-04-23:
I think that the culprit might be in ```jmri.util.ThreadingUtil.runOnGUIEventually()```:

```
        if (isGUIThread()) {
            // run now, despite the "eventually" in the name; just a simplification
            ta.run();
        } else {
            // dispatch to Swing
            javax.swing.SwingUtilities.invokeLater(ta);
        }
```

I think that logic to "run now" might be causing overlapping activities on the AWT thread, instead of merely queueing activity on that thread.

> ```
> at jmri.beans.Bean.lambda$6(Bean.java:149)
> at jmri.beans.Bean$$Lambda$17/1071097621.run(Unknown Source)
> ```

is a lambda passed as an argument to ```jmri.util.ThreadingUtil.runOnGUIEventually()```

bobjacobsen, 2016-04-23:
Not sure why that would have any effect (if it's already on the Swing thread, aren't the operations between Swing calls by definition?), but let's try that anyway.  See PR #1209.

rhwood, 2016-04-23:
Not calling ```invokeLater``` bypasses the [EventQueue](https://docs.oracle.com/javase/7/docs/api/java/awt/EventQueue.html) which may cause issues if an event triggers another event that preempts it before the original event is fully propigated. I recall seeing an issue (not within JMRI) about this once but can't find it now.

bobjacobsen, 2016-04-23:
> Not calling invokeLater bypasses the EventQueue which may cause issues if an event triggers another event that preempts it before the original event is fully propigated. 

No, it's still acting within the EventQueue; that's what the ````isGUIThread()```` check ensures.  

The more I look at @DanielBoudreau's first example, the more I think this is just a plain old multi-threading error that has nothing to do with Swing per se.  It shows two exceptions that were caught on two separate threads at the same millisecond, each of which had called the same method (in a singular object): ````UserMessagePreferencesPane.newMessageTab(UserMessagePreferencesPane.java:170)````.  The two sources are:
- apps.Apps$3.run(Apps.java:386) which invokes a separate thread to call apps.gui3.TabbedPreferences.init(TabbedPreferences.java:160) which works its way down.
- the Swing event queue which processes a Swing event (from where is not clear) which via (stack trace, read up)
```` java
    at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
    at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
    at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
    at jmri.beans.Bean.lambda$6(Bean.java:149)
    at jmri.beans.Bean$$Lambda$17/1908143486.run(Unknown Source)
````
invokes listeners that end up in the same code via (read up):
````
    at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.newMessageTab(UserMessagePreferencesPane.java:170)
    at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.refreshOptions(UserMessagePreferencesPane.java:400)
    at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane.lambda$0(UserMessagePreferencesPane.java:57)
    at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane$$Lambda$85/1702073703.propertyChange(Unknown Source)
````

So this seems like a basic multi-threading mistake, independent of Swing-specific semantics.  The simplest thing to do might be to not launch the 1st operation in a separate thread.

bobjacobsen, 2016-05-01:
This seems fixed, so closing this.  If it recurs, please reopen.

closed 2016-05-01 by bobjacobsen