JMRI/JMRI #1199: Shutdown task remains active on closed windows

DanielBoudreau, 2016-04-22:
When shutting down JMRI, windows that have already been closed continue to have their shutdown tasks active.  One example is the Trains window when operation files are modified.  Closing the Trains window causes the shutdown task to activate and prompt the user to save their files.  Unfortunately the shutdown task for the Trains window is activated again when JMRI is shutdown and the Trains window was previously closed.  Note that opening the Trains window more than once will cause the multiple instances of the shutdown task.  This a new bug that has been recently introduced.

rhwood, 2016-04-23:
It is, and has always been, the responsibility of the creator of a shutdown task to ensure it is deregistered if it should not be run.

The recent changes only reveal that:
1. the task is not being deregistered if inappropriate, or
2. the train window is not actually being closed after the task runs.

The recent changes fix a long standing bug where window closing events were *never* being triggered on OS X.

pabender, 2016-04-23:
Sent from my iPad

On Apr 22, 2016, at 6:48 PM, Daniel Boudreau <notifications@github.com> wrote:

> When shutting down JMRI, windows that have already been closed continue to have their shutdown tasks active. One example is the Trains window when operation files are modified. Closing the Trains [...]
While there may very well have been a change to the behavior, shutdown tasks ( and shutdown hooks ) really should be associated with shutting down the system, not closing a window.  

If you think about this from the model-view-controller point of view,  you are registering the shutdown hook with the view of the data, but you really want to register this with either the model or the controller ( probably the controller ).

Within a single JMRI instance, there really should only be one controller, but multiple views are to be expected.

Paul

DanielBoudreau, 2016-04-23:
From the OperationsManager.java written by R. Wood 2014:

 

   /**

     * Register the non-default {@link jmri.ShutDownTask}.

     *

     * Replaces the existing operations ShutDownTask with the new task. Use a

     * null value to prevent an operations ShutDownTask from being run when JMRI

     * shuts down. Use {@link #getDefaultShutDownTask() } to use the default

     * operations ShutDownTask.

     *

     * @param shutDownTask The new ShutDownTask or null

     */

    public void setShutDownTask(ShutDownTask shutDownTask) {

        if (InstanceManager.shutDownManagerInstance() != null) {

            if (this.shutDownTask != null) {

                InstanceManager.shutDownManagerInstance().deregister(this.shutDownTask);

            }

            this.shutDownTask = shutDownTask;

            if (this.shutDownTask != null) {

                InstanceManager.shutDownManagerInstance().register(this.shutDownTask);

            }

        }

    }

 

 

The “deregister” doesn’t seem to be working anymore.

 

Dan

DanielBoudreau, 2016-04-23:
Ignore my last comment.  The issue is a window that is already closed will get a call to handleModified() when shutting down JMRI,

DanielBoudreau, 2016-04-23:
Here's the stack track.  The operations window Trains is already closed:

TrainsTableFrame.handleModified() line: 565	
TrainsTableFrame(JmriJFrame).windowClosing(WindowEvent) line: 753	
AWTEventMulticaster.windowClosing(WindowEvent) line: 349	
TrainsTableFrame(Window).processWindowEvent(WindowEvent) line: 2058	
TrainsTableFrame(JFrame).processWindowEvent(WindowEvent) line: 297	
TrainsTableFrame(Window).processEvent(AWTEvent) line: 2017	
TrainsTableFrame(Component).dispatchEventImpl(AWTEvent) line: 4881	
TrainsTableFrame(Container).dispatchEventImpl(AWTEvent) line: 2292	
TrainsTableFrame(Window).dispatchEventImpl(AWTEvent) line: 2739	
TrainsTableFrame(Component).dispatchEvent(AWTEvent) line: 4703	
DefaultShutDownManager.lambda$0(Frame) line: 153	
611207429.accept(Object) line: not available	
Spliterators$ArraySpliterator<T>.forEachRemaining(Consumer<? super T>) line: 948	
ReferencePipeline$Head<E_IN,E_OUT>.forEach(Consumer<? super E_OUT>) line: 580	
DefaultShutDownManager.shutdown(int, boolean) line: 148	
DefaultShutDownManager.shutdown() line: 96	
AppsBase.handleQuit() line: 449	
Apps.handleQuit() line: 1113	
PanelPro(Apps).windowClosing(WindowEvent) line: 913	
AWTEventMulticaster.windowClosing(WindowEvent) line: 350	
AWTEventMulticaster.windowClosing(WindowEvent) line: 349	
AWTEventMulticaster.windowClosing(WindowEvent) line: 349	
JmriJFrame(Window).processWindowEvent(WindowEvent) line: 2058	
JmriJFrame(JFrame).processWindowEvent(WindowEvent) line: 297	
JmriJFrame(Window).processEvent(AWTEvent) line: 2017	
JmriJFrame(Component).dispatchEventImpl(AWTEvent) line: 4881	
JmriJFrame(Container).dispatchEventImpl(AWTEvent) line: 2292	
JmriJFrame(Window).dispatchEventImpl(AWTEvent) line: 2739	
JmriJFrame(Component).dispatchEvent(AWTEvent) line: 4703	
EventQueue.dispatchEventImpl(AWTEvent, Object) line: 746	
EventQueue.access$400(EventQueue, AWTEvent, Object) line: 97	
EventQueue$3.run() line: 697	
EventQueue$3.run() line: 691	
AccessController.doPrivileged(PrivilegedAction<T>, AccessControlContext) line: not available [native method]	
ProtectionDomain$1.doIntersectionPrivilege(PrivilegedAction<T>, AccessControlContext, AccessControlContext) line: 75	
ProtectionDomain$1.doIntersectionPrivilege(PrivilegedAction<T>, AccessControlContext) line: 86	
EventQueue$4.run() line: 719	
EventQueue$4.run() line: 717	
AccessController.doPrivileged(PrivilegedAction<T>, AccessControlContext) line: not available [native method]	
ProtectionDomain$1.doIntersectionPrivilege(PrivilegedAction<T>, AccessControlContext, AccessControlContext) line: 75	
EventQueue.dispatchEvent(AWTEvent) line: 716	
EventDispatchThread.pumpOneEventForFilters(int) line: 201	
EventDispatchThread.pumpEventsForFilter(int, Conditional, EventFilter) line: 116	
EventDispatchThread.pumpEventsForHierarchy(int, Conditional, Component) line: 105	
EventDispatchThread.pumpEvents(int, Conditional) line: 101	
EventDispatchThread.pumpEvents(Conditional) line: 93	
EventDispatchThread.run() line: 82

rhwood, 2016-04-23:
> TrainsTableFrame.handleModified() line: 565 
> TrainsTableFrame(JmriJFrame).windowClosing(WindowEvent) line: 753
> AWTEventMulticaster.windowClosing(WindowEvent) line: 349

The stack trace clearly shows that:
1. the window is open, and
2. that ```handleModified()``` is called by ```JmriJFrame.windowClosing(WindowEvent)```

If you don't want the window to be closed twice, ensure it is closed (not merely hidden) the first time.

DanielBoudreau, 2016-04-23:
The window was closed before shutdown. When I click on the main menu “Windows” only PanelPro appears.

 

Dan

rhwood, 2016-04-23:
> The window was closed before shutdown. When I click on the main menu “Windows” only PanelPro appears.

Visually closing a window is not the same as removing the window from the list of open windows that AWT maintains. It is possible (but not common) to merely hide a window when its closed, and this is what appears to be happening. However, since the TrainsTableFrame is set to dispose itself when closed (this is inherited from JmriJFrame and not overridden in OperationsFrame), you need to figure out what is preventing it from being disposed of when closed.

DanielBoudreau, 2016-04-23:
This is the code that is trying to close the Trains window when it is already closed.

 

          // close any open windows by triggering a closing event

            // this gives open windows a final chance to perform any cleanup

            if (!GraphicsEnvironment.isHeadless()) {

                Arrays.asList(Frame.getFrames()).stream().forEach((frame) -> {

                    // do not run on thread, or in parallel, as System.exit()

                    // will get called before windows can close

                    log.debug("Closing frame \"{}\"", frame.getName());

                    Date timer = new Date();

                    frame.dispatchEvent(new WindowEvent(frame, WindowEvent.WINDOW_CLOSING));

                    log.debug("Frame \"{}\" took {} milliseconds to close", frame.getName(), new Date().getTime() - timer.getTime());

                });

            }

            log.debug("windows completed closing {} milliseconds after starting shutdown", new Date().getTime() - start.getTime());

            // wait for parallel tasks to complete

rhwood, 2016-04-23:
Since there is *no* way to programmatically determine if a java.awt.Frame object is open or closed, and we *need* to ensure that all Frames get an opportunity to close, if you do not want a hidden window to be closed, ensure it is disposed of when closing it.

From the java.awt.Frame JavaDocs: "If the program doesn't explicitly hide or dispose the window while processing this event, the window close operation is canceled.”

There is no code path in the TrainsTableFrame that explicitly hides or disposes of the window, so the window is *not closed* as far as the JVM is concerned.

DanielBoudreau, 2016-04-23:
This line is processing all frames that have opened, it isn’t a list of open frames.

 

Arrays.asList(Frame.getFrames()).stream().forEach((frame) ->

 

I turned on the debug messages:

 

2016-04-23 09:47:54,122 managers.DefaultShutDownManager       DEBUG - Frame "frame1" took 36 milliseconds to close [AWT-EventQueue-0]

2016-04-23 09:47:54,122 managers.DefaultShutDownManager       DEBUG - Closing frame "frame2", title:  [AWT-EventQueue-0]

2016-04-23 09:47:54,122 managers.DefaultShutDownManager       DEBUG - Frame "frame2" took 0 milliseconds to close [AWT-EventQueue-0]

2016-04-23 09:47:54,122 managers.DefaultShutDownManager       DEBUG - Closing frame "frame3", title:  [AWT-EventQueue-0]

2016-04-23 09:47:54,122 managers.DefaultShutDownManager       DEBUG - Frame "frame3" took 0 milliseconds to close [AWT-EventQueue-0]

2016-04-23 09:47:54,122 managers.DefaultShutDownManager       DEBUG - Closing frame "frame4", title:  [AWT-EventQueue-0]

2016-04-23 09:47:54,122 managers.DefaultShutDownManager       DEBUG - Frame "frame4" took 0 milliseconds to close [AWT-EventQueue-0]

2016-04-23 09:47:54,122 managers.DefaultShutDownManager       DEBUG - Closing frame "frame5", title:  [AWT-EventQueue-0]

2016-04-23 09:47:54,122 managers.JmriUserPreferencesManager   DEBUG - Saving windowDetails element. [AWT-EventQueue-0]

2016-04-23 09:47:54,138 managers.DefaultShutDownManager       DEBUG - Frame "frame5" took 16 milliseconds to close [AWT-EventQueue-0]

2016-04-23 09:47:54,138 managers.DefaultShutDownManager       DEBUG - Closing frame "frame6", title: Turnouts [AWT-EventQueue-0]

2016-04-23 09:47:54,138 managers.JmriUserPreferencesManager   DEBUG - Saving settings element. [AWT-EventQueue-0]

2016-04-23 09:47:54,153 managers.JmriUserPreferencesManager   DEBUG - Saving windowDetails element. [AWT-EventQueue-0]

2016-04-23 09:47:54,169 managers.JmriUserPreferencesManager   DEBUG - Saving windowDetails element. [AWT-EventQueue-0]

2016-04-23 09:47:54,184 managers.DefaultShutDownManager       DEBUG - Frame "frame6" took 46 milliseconds to close [AWT-EventQueue-0]

2016-04-23 09:47:54,184 managers.DefaultShutDownManager       DEBUG - windows completed closing 219 milliseconds after starting shutdown [AWT-EventQueue-0]

2016-04-23 09:47:54,184 managers.DefaultShutDownManager       DEBUG - Shutdown took 219 milliseconds. [AWT-EventQueue-0]

2016-04-23 09:47:54,184 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]

DanielBoudreau, 2016-04-23:
The “list” in JmriFrame correctly shows which windows are open.

 

    public void dispose() {

        jmri.UserPreferencesManager p = jmri.InstanceManager.getDefault(jmri.UserPreferencesManager.class);

        if (p != null) {

            if (reuseFrameSavedPosition) {

                p.setWindowLocation(windowFrameRef, this.getLocation());

            }

            if (reuseFrameSavedSized) {

                saveWindowSize(p);

            }

        }

        log.debug("dispose \"{}\"", getTitle());

        if (windowInterface != null) {

            windowInterface.dispose();

        }

        if (task != null) {

            jmri.InstanceManager.shutDownManagerInstance().deregister(task);

            task = null;

        }

        synchronized (list) {

            list.remove(this);

        }

rhwood, 2016-04-23:
> The “list” in JmriFrame correctly shows which windows are open.

We cannot guarantee that all open java.awt.Frames are JmriJFrame descendants, but need to be able to close all java.awt.Frame objects.

I think that if TrainsTableAction did not permanently retain a reference to the TrainsTableFrame that it creates, this would not be an issue.

DanielBoudreau, 2016-04-23:
I open the “Turnouts” window and then closed it.  From the debug messages when shutdown JMRI:

 

2016-04-23 10:22:41,198 managers.DefaultShutDownManager       DEBUG - Closing frame "frame7", title: Turnouts [AWT-EventQueue-0]

 

Next test, I open and close the “Tornouts” window three times, and it appears three separate times:

 

2016-04-23 10:25:09,813 managers.DefaultShutDownManager       DEBUG - Closing frame "frame6", title: Turnouts [AWT-EventQueue-0]

2016-04-23 10:25:09,813 managers.JmriUserPreferencesManager   DEBUG - Saving settings element. [AWT-EventQueue-0]

2016-04-23 10:25:09,829 managers.JmriUserPreferencesManager   DEBUG - Saving windowDetails element. [AWT-EventQueue-0]

2016-04-23 10:25:09,845 managers.JmriUserPreferencesManager   DEBUG - Saving windowDetails element. [AWT-EventQueue-0]

2016-04-23 10:25:09,845 managers.DefaultShutDownManager       DEBUG - Frame "frame6" took 32 milliseconds to close [AWT-EventQueue-0]

2016-04-23 10:25:09,845 managers.DefaultShutDownManager       DEBUG - Closing frame "frame7", title: Turnouts [AWT-EventQueue-0]

2016-04-23 10:25:09,860 managers.JmriUserPreferencesManager   DEBUG - Saving settings element. [AWT-EventQueue-0]

2016-04-23 10:25:09,860 managers.JmriUserPreferencesManager   DEBUG - Saving windowDetails element. [AWT-EventQueue-0]

2016-04-23 10:25:09,876 managers.JmriUserPreferencesManager   DEBUG - Saving windowDetails element. [AWT-EventQueue-0]

2016-04-23 10:25:09,891 managers.DefaultShutDownManager       DEBUG - Frame "frame7" took 31 milliseconds to close [AWT-EventQueue-0]

2016-04-23 10:25:09,891 managers.DefaultShutDownManager       DEBUG - Closing frame "frame8", title: Turnouts [AWT-EventQueue-0]

2016-04-23 10:25:09,891 managers.JmriUserPreferencesManager   DEBUG - Saving settings element. [AWT-EventQueue-0]

2016-04-23 10:25:09,907 managers.JmriUserPreferencesManager   DEBUG - Saving windowDetails element. [AWT-EventQueue-0]

2016-04-23 10:25:09,907 managers.JmriUserPreferencesManager   DEBUG - Saving windowDetails element. [AWT-EventQueue-0]

2016-04-23 10:25:09,923 managers.DefaultShutDownManager       DEBUG - Frame "frame8" took 32 milliseconds to close [AWT-EventQueue-0]

2016-04-23 10:25:09,923 managers.DefaultShutDownManager       DEBUG - windows completed closing 292 milliseconds after starting shutdown [AWT-EventQueue-0]

2016-04-23 10:25:09,923 managers.DefaultShutDownManager       DEBUG - Shutdown took 292 milliseconds. [AWT-EventQueue-0]

2016-04-23 10:25:09,923 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]

rhwood, 2016-04-23:
Fine. Whatever. Do whatever you feel is appropriate as long as you can guarantee that that WindowClosing event has been called on *every* window (not just JmriJFrame descendants) before JMRI closes when the application receives a SIGTERM signal (there are normal application closing paths on POSIX-compliant computers that we cannot prevent that need to allow JMRI to close cleanly and correctly on this signal).

DanielBoudreau, 2016-04-23:
I’m only reporting the issue.  I didn’t create it. And I can easily fix the Trains window.  

 

Dan

pabender, 2016-04-23:
Dan,

On Apr 23, 2016, at 9:56 AM, Daniel Boudreau <notifications@github.com> wrote:
> The “list” in JmriFrame correctly shows which windows are open.

What Randall is pointing out is that there is a difference between what windows JMRI thinks are open and what windows the JVM thinks is open.

In order for the JVM to dispose of any object, it's reference count MUST be 0, so you need to make sure that when the window is closed any reference to that windows is also removed.

Windows that require special processing when closing, including removing shutdown hooks, need to handle the Window Closing event themselves, I.e. they need to override the behavior of the superclass.

References to the object need to be removed during the window closing event, unless you are just hiding the window to reopen later.  This is why Randall pointed out that the TrainsTableModel maintains a reference to the frame that contains it.

Paul

closed 2016-05-02 by rhwood