JMRI/JMRI #1216: DecoderPro hangs on quit

dheap, 2016-04-25:
This follows up from my recent comments on an issue raised on  jmrusers [Closing DecoderPro](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/127833)

Seems most likely to happen when an open window (usually a roster entry) is dirty. I counted 20 occurrences over 77 launches in my messages.log since April 15.

`Apps.AppsBase                         ERROR - Continuing after error in handleQuit [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.checkDirtyDecoder(PaneProgFrame.java:799)`

[session.log.zip](https://github.com/JMRI/JMRI/files/233796/session.log.zip)

dheap, 2016-04-25:
Sorry, this session log was after #1211 merged.

bobjacobsen, 2016-04-25:
````
java.lang.NullPointerException at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.checkDirtyDecoder(PaneProgFrame.java:799)
````

This is happening because ````symbolicprog.tabbedframe.PaneProgFrame.windowClosing(..)```` was called after ````dispose()```` had been called.

Not sure whether the problem was this call (e.g. too late) or that dispose had been called too early.

The traceback for this started with ````apps.AppsBase.handleQuit()````, which then worked down through the ````DefaultShutDownManager```` to some AWT window events to 
````awt.AWTEventMulticaster.windowClosing(..)```` finally to ````symbolicprog.tabbedframe.PaneProgFrame.windowClosing(..)````

rhwood, 2016-04-25:
#1222 fixes this in 4.3.5 and #1220 fixes this master.

> java.lang.NullPointerException at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.checkDirtyDecoder(PaneProgFrame.java:799)
>
> This is happening because [...]

Randall Wood
Alexandria Software
202.683.8604
randall.h.wood@alexandriasoftware.com
http://www.alexandriasoftware.com

rhwood, 2016-05-01:
Is this still reproducible?

bobjacobsen, 2016-05-02:
Current master of DecoderPro from Ant and Mac launcher seems to close just fine, with or without preferences open.

dheap, 2016-05-02:
I haven't encountered it since #1220 was merged.

> @rhwood Is this still reproducible?

rhwood, 2016-05-02:
Considering fixed.

closed 2016-05-02 by rhwood