JMRI/JMRI #1274: jmrit.Sound() does not report "file not found" condition

devel-bobm, 2016-05-03:
When a jython script executes 
`self.snd = jmri.jmrit.Sound("non-ExistantFile.wav")`
no indication is given that the file does not exist.  

Then, execution of
`self.snd.play()`
does not produce any sound.  No indication is given that there won't be any sound.

If nothing else, jmrit.Sound() could log an error in the console for the "file-not-found" condition.

bobjacobsen, 2016-05-04:
Does it put something in the log?

rhwood, 2016-05-04:
It doesn't log and per the [JavaDocs](http://docs.oracle.com/javase/1.5.0/docs/api/java/applet/Applet.html#getAudioClip(java.net.URL)), failure is deliberately silent. The call is in this [method](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/Sound.java#L98).

devel-bobm, 2016-05-04:
> -----Original Message-----
> dioClip(java.net.URL)> , failure is deliberately silent. The call is in this method

The particular file I had difficulty with was greater than 100000 bytes, so it was a "streaming" case, not a "Clip" case, account size of the file > 100000 bytes.  As such, my "issue" would apply to the various methods used in jmri.jmrit.Sound.StreamingSound(), including AudioSystem.getAudioInputStream(new File(_file))  .

Note that <https://docs.oracle.com/javase/8/docs/api/javax/sound/sampled/AudioSystem.html#getAudioInputStream-java.io.File-> reports that an exception should be thrown.  jmrit.Sound has this in a try/catch, and it ought to log an error.  I did not see any such an error in the log, and I had jmri.jmrit.Sound logging at the DEBUG level.

I would have expected such log.error() calls to show up in the log, regardless of log4j configuration, even when the java method was invoked from within a jython script.  Is that expectation incorrect?

rhwood, 2016-05-04:
The streaming case can only be called if the file actually exists (```jmrit.Sound#needStreaming(String)``` returns ```false``` for a nonexistent file).

devel-bobm, 2016-05-04:
Ok.

But the question is still open - do we really want a "silent failure" (pun intended), for either of these cases?  I think not.  

I spent several hours trying to figure out why the sound wasn't playing.  I re-worked my sound file several times.  Only after I hacked the source did I discover #1272 .  One simple log message could have saved me all that trouble...

bobjacobsen, 2016-05-04:
> But the question is still open - do we really want a "silent failure" (pun intended), for either of these cases? I think not. 

I may be misunderstanding, but I think the situation is that the Java sound support isn’t throwing any exceptions in the case you’re trying.  (AFAIK, JMRI logs all the exceptions from this code; let me know of a test case if that’s not true)

So what you’re asking for is a check (of something).  You’re welcome to add a check&throw for whether the file exists; is a regular file; is readable; has non-zero length; however far you want to go down the checking sequence. The underlying Java code won’t do that for you, but you can do it yourself.

In the long run, though, a better strategy might be to start getting rid of this class.  It’s based on An Ugly Hack of using the sun.applet.AppletAudioClip implementation, it uses deprecated OS support on Mac’s (you get a message that says “WARNING:  140: This application, or a library it uses, is using the deprecated Carbon Component Manager for hosting Audio Units. Support for this will be removed in a future release. Also, this makes the host incompatible with version 3 audio units. Please transition to the API's in AudioComponent.h.” at runtime), etc.  

Better would be to reimplement it in terms of the more modern Java APIs, or even sending uses to the code in the jmri.jmrit.audio package that already uses them.

devel-bobm, 2016-05-04:
I stumbled upon this issue thru following the code snippets found at JMRI Scripting FAQ page.  If jmri.jmrit.audio is a better solution, then it would be prudent to replace the FAQ snippets with code which uses the better mechanisms.

I cannot commit to any such re-work for the forseeable future.

rhwood, 2016-08-04:
#1738 fixes this by ensuring a ```NullPointerException``` is thrown when constructing a Sound object if the file cannot be found (this is because [```jmri.util.FileUtil#findUrl(String)```](http://jmri.org/JavaDoc/doc/jmri/util/FileUtil.html#findURL-java.lang.String-) method used to find the file returns null if the file cannot be found).

commit 69b691e2 (2016-08-04): Merge pull request #1738 from rhwood/new-sounds

Refactored jmri.jmrit.Sound
Fixes #1272
Fixes #1274
closed 2016-08-04 by rhwood