JMRI/JMRI #1297: DP not retrying when read/write fails

bobjacobsen, 2016-05-08:
(Noticed on JMRIusers, @pabender wrote [summary](https://sourceforge.net/p/jmri/mailman/message/35071473/) quoted below to start [discussion on jmri-developers](https://sourceforge.net/p/jmri/mailman/message/35071473/) about feature set)

DecoderPro is no longer retrying read/write operations that fail (red UNKNOWN status) during read/write operations.

````
On 05/07/2016 12:44 PM, Bob Jacobsen rgj1927@pacbell.net [jmriusers] wrote:
On May 6, 2016, at 11:31 AM, Paul Bender paul.bender@acm.org [jmriusers] <jmriusers@yahoogroups.com> wrote:

Now, we don't seem to retry anything on a failure, not even single CV reads.
That must be a bug in something quite specific, as I’m seeing DecoderPro retrying the operation when I force an error on single CV reads with a Digitrax PR3 standalone setup.  (The Single CV Programmer isn’t meant to do retries)

Could you give me the specifics to try to recreate that?  Does it leave a “red” DP entry, or is something else happening?

Yes, you end up with red entries on the programmer pane.  This is at 
least happening with a
data byte not found response (which triggers a 301 error on the front end).

The simplest way I've found to replicate this is to do a read full sheet 
on the basic pane with no locomotive present on the programming track.  
Here is the XPressNet log for a Lenz LZV100 (connected via an LI100F ) 
trying to read a Lenz Gold Mini decoder.

(trace elided, see email linked above)

If you look through the read requests above, you'll see there is exactly 
one read request for each of CVs:
1,29,17,18,19,105,106,8,7, and 128 (in that order).

This same operation used to trigger several read attempt cycles. Also, 
the "Data Byte Not Found" is not a fatal error because it can be caused 
by a locomotive moving on the programming track.
````

dheap, 2016-05-08:
Same on NCE and SPROG, so it is not system-specific.

I had to include retry code in the identify decoder code so we got reliable decoder identification with multi-CV variables. Likewise I was forced to add it to the new splitVal code I am working on, otherwise it was well-nigh impossible to read 4 to 28 CV variables as are found in ESU decoders (a retry of the whole 4 - 28 as happens with Read Changes would generally lead to a failure in a different component CV).

But my view is that it should be done at a lower level in the code, not in the individual VariableValue extensions code. Identify decoder does not even use VariableValue code (maybe it should) rather than duplicating code.

Dave

> @BobJacobsen (Noticed on JMRIusers, @pabender wrote summary quoted below to start discussion on jmri-developers about feature set)
> DecoderPro is no longer retrying read/write operations that fail (red UNKNOWN status) during read/write operations.

dheap, 2016-05-08:
Sorry, moved my comments back to jmri-developers.

> On 9 May 2016, at 7:27 AM, Dave Heap <dgheap@gmail.com> wrote:
> 
> @BobJacobsen (Noticed on JMRIusers, @pabender wrote summary quoted below to start discussion on jmri-developers about feature set)

closed 2021-04-14 by stale[bot]