JMRI/JMRI #1309: Scripts get wrong sensor type?

bobjacobsen, 2016-05-10:
Ron McKinnon reported a problem with a previously-working script where 
````
self.waitSensorActive(sensors.provideSensor("495"))
````
would immediately fall through.  Diagnosed to be that the ````sensors.provideSensor("495")```` call was return an Internal sensor object, not the required LocoNet object.  There was a LocoNet connection configured and working, defaults appeared to be set right.

Problem appears to be new in 4.3.5, likely associated with the various changes to how the Internal system is created.

For background, see the [jmri-developers thread](https://sourceforge.net/p/jmri/mailman/jmri-developers/thread/3aunib5nos14fpp41vs047r4ognbisfv5g%404ax.com/#msg35067594)

bobjacobsen, 2016-05-11:
Now seems to be related to startup order and when managers are created vs scripts run.  More discussion in thread.

rhwood, 2016-07-18:
Is this a symptom of #1014?

closed 2017-07-03 by bobjacobsen