JMRI/JMRI #1318: Create Release 4.3.7

bobjacobsen, 2016-05-16:
This is the last test release from HEAD of JMRI/JMRI master before the 4.4 production release, some time in June 2016.

mattharris, 2016-06-11:
The release-4.3.7 branch has been created. 

From now on, please document your changes in the [jmri4.3.7.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.3.7.shtml) release note file.

Maintainers, please set the 4.5.1 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [new server](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.3.7/)

mattharris, 2016-06-11:
This has now been released.

In addition, a [new branch has been created](https://github.com/JMRI/JMRI/tree/release-4.3.8-suggested-patches) `release-4.3.8-suggested-patches` which should be used when preparing any content for possible inclusion into the next release.

New commits on `master` will form the basis of 4.5.1.

closed 2016-06-11 by mattharris