JMRI/JMRI #1319: C/MRI update from Model Railroad Control Systems

bobjacobsen, 2016-05-16:
Chuck Cantina of Model Railroad Control Systems has contributed a major update to the C/MRI support. Parts are intended for their [product line](http://www.modelrailroadcontrolsystems.com/cpnode/), while other parts are general improvements. He provided some screen images which I've included below.

The code is based on JMRI 3.2.1 from 2012. Our code base has evolved a bit since then. Git will come to the rescue here.  I've started a "[MRCS-CMRI-update-from-3.2.1](https://github.com/JMRI/JMRI/tree/MRCS-CMRI-update-from-3.2.1)" branch at the 3.2.1 release point, merged Chuck's code, and committed his changes.  They compile and run headless test, with intermittent errors and failures that seem unrelated to the C/MRI code (our test reliability & coverage has also come a long way in four years!)  That branch is now at GitHub so people can check it out and try it. 

The next step is to merge forward to the current code.  That will require resolving some conflicts. I've put  the merge-to-current-master list at the bottom of this post; note in particular the bit about the rename list at the bottom. It's probably best to work on this in small slices.

The intervening production releases are:
- [ ] 3.4
- [ ] 3.6
- [ ] 3.8
- [ ] 3.10 (and 3.10.1)
- [ ] 4.0
- [ ] 4.2
- [ ] 4.4 (not yet, but we'll probably be there by the time this gets done)

If anybody wants to work on these migrations, please let me know, otherwise I'll probably take a shot at it intermittently over the next week or two, one production release at a time.

If you work on this branch, please make comments here or reference this Issue in your PRs so we can all stay in touch.

Bob

[image]

[image]

[image]

````
% git merge MRCS-CMRI-update-from-3.2.1
warning: Cannot merge binary files: resources/vsdecoder/example.vsd (HEAD vs. MRCS-CMRI-update-from-3.2.1)
warning: Cannot merge binary files: help/en/JavaHelpSearch/TMAP (HEAD vs. MRCS-CMRI-update-from-3.2.1)
warning: Cannot merge binary files: help/en/JavaHelpSearch/POSITIONS (HEAD vs. MRCS-CMRI-update-from-3.2.1)
warning: Cannot merge binary files: help/en/JavaHelpSearch/OFFSETS (HEAD vs. MRCS-CMRI-update-from-3.2.1)
warning: Cannot merge binary files: help/en/JavaHelpSearch/DOCS.TAB (HEAD vs. MRCS-CMRI-update-from-3.2.1)
warning: Cannot merge binary files: help/en/JavaHelpSearch/DOCS (HEAD vs. MRCS-CMRI-update-from-3.2.1)
Auto-merging xml/decoders/Zimo_Unified_software_MX69Vv31.xml
CONFLICT (add/add): Merge conflict in xml/decoders/Zimo_Unified_software_MX69Vv31.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX69Vv28.xml
CONFLICT (content): Merge conflict in xml/decoders/Zimo_Unified_software_MX69Vv28.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX69Sv31.xml
CONFLICT (add/add): Merge conflict in xml/decoders/Zimo_Unified_software_MX69Sv31.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX69Sv28.xml
CONFLICT (content): Merge conflict in xml/decoders/Zimo_Unified_software_MX69Sv28.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX69Lv31.xml
CONFLICT (add/add): Merge conflict in xml/decoders/Zimo_Unified_software_MX69Lv31.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX69Lv28.xml
CONFLICT (content): Merge conflict in xml/decoders/Zimo_Unified_software_MX69Lv28.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX695Vv30.xml
CONFLICT (content): Merge conflict in xml/decoders/Zimo_Unified_software_MX695Vv30.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX695Sv30.xml
CONFLICT (content): Merge conflict in xml/decoders/Zimo_Unified_software_MX695Sv30.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX695KNv30.xml
CONFLICT (add/add): Merge conflict in xml/decoders/Zimo_Unified_software_MX695KNv30.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX64Dv31.xml
CONFLICT (add/add): Merge conflict in xml/decoders/Zimo_Unified_software_MX64Dv31.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX64Dv28.xml
CONFLICT (content): Merge conflict in xml/decoders/Zimo_Unified_software_MX64Dv28.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX632v31.xml
CONFLICT (add/add): Merge conflict in xml/decoders/Zimo_Unified_software_MX632v31.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX632v28.xml
CONFLICT (content): Merge conflict in xml/decoders/Zimo_Unified_software_MX632v28.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX632Cv31.xml
CONFLICT (add/add): Merge conflict in xml/decoders/Zimo_Unified_software_MX632Cv31.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX632Cv28.xml
CONFLICT (content): Merge conflict in xml/decoders/Zimo_Unified_software_MX632Cv28.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX631v31.xml
CONFLICT (add/add): Merge conflict in xml/decoders/Zimo_Unified_software_MX631v31.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX631v28.xml
CONFLICT (content): Merge conflict in xml/decoders/Zimo_Unified_software_MX631v28.xml
Auto-merging xml/decoders/Zimo_Unified_software_MX631Cv31.xml
[... 285 lines cut ...]
CONFLICT (content): Merge conflict in java/src/apps/AppsBase.java
Auto-merging java/src/apps/ActionListBundle_it.properties
CONFLICT (content): Merge conflict in java/src/apps/ActionListBundle_it.properties
Auto-merging help/en/package/jmri/jmrit/operations/Operations.shtml
CONFLICT (content): Merge conflict in help/en/package/jmri/jmrit/operations/Operations.shtml
Auto-merging help/en/index.html
CONFLICT (content): Merge conflict in help/en/index.html
Auto-merging help/en/JavaHelpSearch/TMAP
CONFLICT (content): Merge conflict in help/en/JavaHelpSearch/TMAP
Auto-merging help/en/JavaHelpSearch/SCHEMA
CONFLICT (content): Merge conflict in help/en/JavaHelpSearch/SCHEMA
Auto-merging help/en/JavaHelpSearch/POSITIONS
CONFLICT (content): Merge conflict in help/en/JavaHelpSearch/POSITIONS
Auto-merging help/en/JavaHelpSearch/OFFSETS
CONFLICT (content): Merge conflict in help/en/JavaHelpSearch/OFFSETS
Auto-merging help/en/JavaHelpSearch/DOCS.TAB
CONFLICT (content): Merge conflict in help/en/JavaHelpSearch/DOCS.TAB
Auto-merging help/en/JavaHelpSearch/DOCS
CONFLICT (content): Merge conflict in help/en/JavaHelpSearch/DOCS
Auto-merging build.xml
CONFLICT (content): Merge conflict in build.xml
warning: inexact rename detection was skipped due to too many files.
warning: you may want to set your merge.renamelimit variable to at least 6487 and retry the command.
Automatic merge failed; fix conflicts and then commit the result.
````

pabender, 2016-05-20:
> The code is based on JMRI 3.2.1 from 2012. Our code base has evolved a bit since then. Git will come to the rescue here. I've started a "MRCS-CMRI-update-from-3.2.1" branch at the 3.2.1 release point, merged Chuck's code, and committed his changes. 

I have to ask, why was this based on 3.2.1, and not something more recent?

Do we know how many of the conflicts are due to formatting differences?  It might be a good idea to reformat the code in the branch before merging to see if that reduces the number of conflicts.

Paul

bobjacobsen, 2016-05-20:
> I have to ask, why was this based on 3.2.1, and not something more recent?

I think you'd have to ask Chuck about that. I don't think he's a GitHub subscriber. He's got a different development model. It works for him. 

The real issue is not so much this merge, but what happens in the future. That's not clear. 

> Do we know how many of the conflicts are due to formatting differences? It might be a good idea to reformat the code in the branch before merging to see if that reduces the number of conflicts.

Chuck changed only jmrix.AbstractMonFrame and files in the jmrix.cmri package & children. Those should be pretty straight-forward to merge. 

But there are lots of other conflicts appearing that are (I think) due to 3.2 being pre-Git, e.g. being migrated from SVN, and before we got the git control files and settings right.

pabender, 2016-05-20:
I just took a first stab at merging up to 3.4.

I'm hoping I didn't mess up any of the decoder files.  

There were a number of those with marked changes.  PR against the branch to be made in a minute.

bobjacobsen, 2016-05-20:
> I just took a first stab at merging up to 3.4.

Thanks!  How hard was it?

I was planning to do ````git checkout —theirs xml/; git add xml;```` et al.  What is more complicated than that?

Bob

pabender, 2016-05-20:
On May 20, 2016, at 4:50 PM, Bob Jacobsen <notifications@github.com> wrote:
> > I just took a first stab at merging up to 3.4.
> 
> Thanks! How hard was it?
> 
> I was planning to do ````git checkout —theirs xml/; git add xml;```` et al. What is more complicated than that?

For some reason, I couldn't get 'git checkout --thiers' to work.

I ended up using 'git mergetool' to resolve the conflicts.

Most of the conflicts were in properties files and decoder definitions.  If I did this right, I accepted the versions from 3.4 intact.

Most of the real conflicts were actually pretty straightforward.  In AbstractMonFrame, I did just add a couple of new panels to the window.  Chuck added two and there was one added between 3.2.1, so rather than trying to merge those, I just added all of them to avoid loosing functionality.

Paul

pabender, 2016-05-20:
Sent from my iPad

On May 20, 2016, at 4:29 PM, Bob Jacobsen <notifications@github.com> wrote:

> I think you'd have to ask Chuck about that. I don't think he's a GitHub subscriber. He's got a different development model. It works for him.
>
> The real issue is not so much this merge, but what [...]

While I welcome the contribution, I think we need to highly encourage outside contributors to commit using a recent release.

I had originally thought this might have been due to an OS incompatibility ( 3.2.1 was still Java 1.6 ) but we didn't switch off the Java 1.6 requirement until 4.0, so that isn't it.

> Chuck changed only jmrix.AbstractMonFrame and files in the jmrix.cmri package & children. Those should be pretty straight-forward to merge.
>
> But there are lots of other conflicts appearing that are [...]

Hopefully the merge I did already fixed most of those conflicts, or at least the real conflicts.

Paul

pabender, 2016-05-21:
I'm starting to look through the version 3.6 changes.

It looks like there was a bulk replace of "C/MRI" with CMRInet in the properties files that are under jmri.jmrix.cmri.  I may have inadvertently removed some of those changes with my previous commit (I hadn't noticed it, until I was confronted with lines that had that change plus special character changes in the non-english properties files).

Paul

bobjacobsen, 2016-05-21:
> It looks like there was a bulk replace of "C/MRI" with CMRInet in the properties files that are under jmri.jmrix.cmri

This is troubling. 

One of the very few exceptions to "code wins" is when the new code is subtractive: it takes something away. 

This does:
- Although a few do use the CMRInet terminology, many times more use the C/MRI terminology. This is true of the C/MRI discussion group, in the JMRI community, in Dr. Chubb's books, etc. 
- The change isn't complete: not just in the code, but more importantly in the documentation. Chuck provided no modifications to help files, the manuals, etc, so they'll be inconsistent with what's visible on the screen. 

How to proceed?

At a minimum, I think we should ask the user communities (JMRI _and_ C/MRI) via their mailing lists whether this is a good thing or not. Depending on the answer, we can figure out what's next.

pabender, 2016-05-21:
> At a minimum, I think we should ask the user communities (JMRI _and_ C/MRI) via their mailing lists whether this is a good thing or not. Depending on the answer, we can figure out what's next.

I think that is a good next step.

I'm not on the C/MRI mailing list.  Would you be able to send the request to both communities?

My current local branch has all the changes for 3.6 ready to go.  I'll hold off on pushing that until we have a decision here.

In my local repository, I went ahead and changed the properties files to CMRInet, but I can pretty quick do a bulk replace to put everything back to C/MRI.

Paul

rhwood, 2016-05-21:
The [NMRA CMRI SIG](http://nmra.org/sigs/cmri-sig) maintains the CMRInet layout control specification, so it might be an intentionally incomplete change from C/MRI (specification + implementing devices) to CMRInet (just the specification).

pabender, 2016-05-21:
On May 21, 2016, at 6:59 AM, Randall Wood <notifications@github.com> wrote:
> The NMRA CMRI SIG maintains the CMRInet layout control specification, so it might be an intentionally incomplete change from C/MRI (specification + implementing devices) to CMRInet (just the specification).
> 
We know where the change comes from, but the question really is should 30+ years of history be supplanted by terminology that is only a year or two old.  Chuck and Seth are big participants in the CMRInet movement.

At a minimum, we must be consistent in our presentation of information to users, so if we move forward with a change to CMRInet, it has to be universal across both the UI and the documentation.  The change was, as far as I can see, a blanket change only to properties files.

Paul

CCatania01, 2016-05-24:
I want to address some of the questions and concerns raised on this forum, by showing the timeline for events, which influenced the development work cycle for CMRInet support in JMRI.

December 2012 - JMRI production version 3.2 released.

May 2013 - In a telephone conference call with Seth Neumann and Chuck Catania, Bruce Chubb endorsed Open Source C/MRI on Arduino. 

May/December 2013 - Prototype cpNode system development.  CMRI protocol implemented per C/MRI User's Manual Version 3.0 protocol specification.  JMRI production version 3.2 used for CMRI protocol implementation on cpNode.  A few hand built proto systems sent to selected alpha testers, including one who uses VB exclusively with his layout.
 
January 2014 - Initial roll out of cpNode to Arduini Yahoo group.  Bare board buy by early adopters of cpNode.  Draft of cpNode Technical Guide sent out for review and comment.

April 2014 - MRCS established to promote Open Source hardware and software solutions for model railroad layout device management and control.
Progressive installation of 10 cpNodes installed at a client using Beta cpNodes and JMRI 3.2.  The layout is HO scale with 90+ blocks (DCCOD), Hare turnout control, C&O signals, and NCE for locomotive control.  Layout development is ongoing.  This installation was used to characterize C/MRI network performance for a total cpNode environment.  

May 2014 - Started work on changes to JMRI for cpNode support.  SVN download from SourceForge version 3.2.  Development environment dual core Intel iMac, OSX 10.6.8, NetBeans 7.2

July 2014 Cleveland NMRA - Demonstrated cpNode system in SIG room with JMRI 3.2.1'ish running some cpNode additions and network management functions.
An impromptu meeting in front of the MRCS table with Charlie Getz, Didrik Voss, Tony Koester, and Chuck Catania, a discussion of layout control standards ensued. Didrik Voss requested that a C/MRI protocol specification be written and submitted to the NMRA as a standard.  Chuck Catania was assigned the task to work with Bruce Chubb to create the document for submission to the NMRA.  
This collaborative work resulted in the establishment of a repository under the NMRA Standards section.  The C/MRI protocol specification became the Layout Control Specification LCS-9.10 CMRInet Protocol.  
Bruce Chubb wanted the slash "/" removed from C/MRI, and "CMRInet" used when describing or referring the CMRI protocol.  C/MRI or CMRI would remains as the designation for the hardware and software elements related to the Chubb Computer Model Railroad Interface.

March 2015 - CMRI SIG established in the NMRA with the CMRI_Users Yahoo group as the anchor for discussion, information, and decisions regarding CMRI.  All aspects of CMRI use, hardware, software, protocol extensions, and problems are directed through this forum. A CMRI Steering committee was formed and monitors activity on the CMRI_Users group.

Present - JMRI 3.2.1'ish with Java 1.6 support is intended to be a starting point for CMRI users who may want to migrate their existing layout control software to JMRI.  Typically, these users have older machines and operating systems and are not necessarily motivated to jump to the latest hardware or software offering.

The master transition plan is to migrate 3.2.1'ish to a 3.10.2 branch, establishing 3.10.2 as the CMRInet base.  Concurrently, a 4.x.x development branch with the CMRInet classes would be created, supporting Java 1.8.  4.x.x would stay as a development branch until such time as the supporting documentation, help files, and web pages can be updated.

The use of C/MRI titles in existing documents can remain unchanged as these refer to the Classic Chubb system.  The resource bundle changes, (swapping C/MRI,CMRI with CMRInet) apply within the network function and feature context.  If there is great objection to the GUI title changes, it is a simple matter to re-do the resource bundle files.  I will do that task if that is what is desired.
The bottom line motivation for this work is to provide a focused CMRI software environment for current and potential JMRI users.  Others can accomplish future extensions to LCS-9.10.x for BASIC, Visual Basic, and other CMRInet software implementations.  Visual Basic network management and error handling can be implemented using the JMRI CMRInet model.

Chuck Catania
Model Railroad Control Systems

pabender, 2016-05-24:
Sent from my iPad

On May 24, 2016, at 8:57 AM, Chuck Catania <notifications@github.com> wrote:
> Present - JMRI 3.2.1'ish with Java 1.6 support is intended to be a starting point for CMRI users who may want to migrate their existing layout control software to JMRI. Typically, these users have [...]
We understand that problem, but we made the decision more than 2 years ago that  we had to move forward in Java revisions to Java 8, because we were going to loose support for developing on 1.6.

Java 1.8 can still be installed on Windows XP, and that's the biggest group of older machines we are willing to do any support for.  ( If I remember correctly, almost any Intel Mac can be used for Java 1.8 as well, and Linux isn't really an issue.)
> The master transition plan is to migrate 3.2.1'ish to a 3.10.2 branch, establishing 3.10.2 as the CMRInet base. Concurrently, a 4.x.x development branch with the CMRInet classes would be created, [...]
The work that Bob and I are doing now with the code you gave us is to migrate the patch up to the current release.

I don't really like the idea of stopping to release a version 3.10.2.  Perhaps Bob has other thoughts on the subject.
> The use of C/MRI titles in existing documents can remain unchanged as these refer to the Classic Chubb system. The resource bundle changes, (swapping C/MRI,CMRI with CMRInet) apply within the [...]
The real problem I have is not that the resource bundles were changed, but the help files for version 3.2.1, which are a part of the source repository, were not updated to reflect the change.  That gives us a consistency problem, and that needs to be rectified at some point.  We can do that either by changing the resource files or by changing the help.

Paul

pabender, 2016-05-25:
After getting the branch up to version 3.6, it was just taking too much time to do this one release at a time, so I made the executive decision to just bring it even with master/origin.  PR #1360 brings the branch up to the current master/origin.

The PR DOES NOT pass AllTest yet, due in large part to the changes to AbstractMonFrame.  At least 4 Tests are failing for non-C/MRI monitors.  The code will have to be fixed before this can be merged into origin/master.

There may also have been other inadvertent errors added while doing the merge work.  That's one of the issues with merging code that came from a starting point so far back in the history history when other, conflicting, changes have been made to the code in the meantime.

 I would like to get the automated testing working on the PR before I merge it into the branch. 

Paul

CCatania01, 2016-05-25:
Paul,
The near term goal for me is to have a good representative JMRI version which will run with 1.6 and a developer release in the 4.3.x train which runs with 1.8.  This approach would cover most users, and provide a transition path for those CMRI users who want to migrate to, or start using JMRI.  Those CMRI users who are on the latest version, and update frequently, will be serviced by the great effort you are putting in.

I not sure there is a need to have all versions from 3.6 on be upgraded, unless once 3.6 can be built, it is a straightforward process to update the rest in that series.   It is your call as to the effort and direction which is needed.

I would like to get some guidance on what the documentation process entails.  I am committed to do that for the changes and features I have created.

Thank you,
Chuck

pabender, 2016-05-25:
Chuck,

On 05/25/2016 01:25 PM, Chuck Catania wrote:
> The near term goal for me is to have a good representative JMRI
> version which will run with 1.6 and a developer release in the 4.3.x
> train which runs with 1.8. This approach would cover most [...]

There's a tagged version of 3.6 with your changes on github right now.  
That does compile, but it's just the source repository tagged at that 
revision (so no binaries).

That said, One long-standing philosophy of the JMRI project is that we 
don't go back and add new features to previous releases.  We also don't 
make point releases for production code unless a very serious bug is 
found fairly shortly after the code was released, and in my 14 years 
working on the project, I can never remember us going back to make a 
point release on code once the next production release was made.

You are asking us to make a new point release based on 3.10.1, and there 
have been 2 intervening production releases. Back-porting new features 
to code released 17 months ago sets a very bad precedent about 
back-porting new features to other code we released in the past.  We 
just don't have the manpower to support adding all those newer features 
to old code, especially when the coding style and process has evolved so 
much (for the better) since the release was made.

I'm going to recommend highly that we continue to follow our own rules 
about point releases and production code.  This means any effort to try 
to release a 3.x.x version of this code should stop. I won't be putting 
any more effort into a release on that track, and I can't justify the 
man hours required for someone else to do that either.

> I would like to get some guidance on what the documentation process 
> entails. I am committed to do that for the changes and features I have 
> created.
>
>

The first step is to clone the repository branch that has the current 
version of the code in it so
that you can make updates to the branch.  You'll need to setup your 
repository so that you can merge to your fork of the branch and then 
create a pull request between your fork and the branch
(see: 
http://jmri.sourceforge.net/help/en/html/doc/Technical/getgitcode.shtml 
and http://jmri.sourceforge.net/help/en/html/doc/Technical/GitFAQ.shtml )

You need to update the documentation in the source repository under:

help/en/html/hardware/cmri/

to reflect the current state, and add documentation about the new 
tools.  (add pictures and new HTML files as required).

You also need to add and/or update tests for the code you have written.

the tests belong in

java/test/jmri/jmrix/cmri

At some point in the future (after the next production release) we 
willll merge the branch into master, but we have to get all the tests 
passing first.

Paul

pabender, 2016-05-26:
Ok, I've found the issue with why the PR was not passing tests.  It was something I added during the merge process due to a conflict, and not something that was in Chuck's original code submission.  I'll be pushing that up soon, so the branch will be passing tests in a couple of hours.

I did run into an issue is going to take more time to fix however.  There is now a C/MRI specific packet filter in the system independent AbstractMonFrame.  So fixing that is another thing on the ToDo list.  This isn't something that testing would have caught.

Paul

pabender, 2016-05-26:
PR #1365  cleans up the issues with the AbstractMonFrame.

We may want to think about the move from two buttons to start/stop logging to the one that chuck introduced.  I left the single button, but personally prefer two.

Paul

bobjacobsen, 2016-05-29:
@CCatania01 said:
> The near term goal for me is to have a good representative JMRI version which will run with 1.6 and a developer release in the 4.3.x train which runs with 1.8. This approach would cover most users, [...]

We've had the Java 1.6 vs 1.8 discussion, that decision has been made, and we're not going back. 

There are very few, perhaps even _zero_,  people who were running C/MRI via Visual Basic who can't install and use Java 1.8, since Java 1.8 runs quite well on Window XP or later.  (Yes, there are some pre-2008 Macs that can't easily run Java 1.8, but that's certainly not a large C/MRI BASIC community)

Sometime early this summer, we'll merge the branch that @pabender is working on. It'll appear in JMRI versions after that.  We'll maintain that code as natural part of how JMRI development works, and it'll just naturally keep working. 

Chuck, if that's not satisfactory, and you therefore want to create and distribute custom executables based on JMRI code, you're welcome to do that within the terms of [JMRI's license](http://jmri.org/COPYING).  Those terms are pretty simple. You should [read it yourself](http://jmri.org/COPYING), but basically you have to make clear that it's a modified version of JMRI, not JMRI itself; identify the modifications you've made and make those available; and that it's still under the same JMRI license terms.

GitHub makes it pretty straightforward to develop on separate branches. You create your own fork of the JMRI/JMRI repository and [work against that](http://jmri.sourceforge.net/help/en/html/doc/Technical/gitdeveloper.shtml). If you want to start with @pabender's [merge to JMRI 3.6](https://github.com/JMRI/JMRI/releases/tag/MRCS-CMRI-Update-3.6), you can do that instead of starting from the current head. Once you have something to release to users, we even have [instructions for creating release files](https://github.com/JMRI/JMRI/blob/master/scripts/HOWTO-distribution.md) in our usual easy-to-distribute form that you can use starting with your own repository.  

And, should you or your customers want to simultaneously have the features you've added, and all the other features that people add to JMRI (for an example of how much gets done every six months or so, see the list of changes in the [JMRI 4.2 release note](http://jmri.sourceforge.net/releasenotes/jmri4.2.shtml)), you can merge your work with JMRI's master branch and then [submit a pull request](http://jmri.sourceforge.net/help/en/html/doc/Technical/gitdeveloper.shtml#pull) to merge it back into JMRI. If you haven't been working along with us, there will be lots of changes that you'll have to handle to get a clean merge, instead of getting the benefit of working with the group.  But that's your choice.  If you propose a clean pull request, I'm sure it'll be merged in with thanks.

But we're not going back on the Java 1.6 vs 1.8 decision, and we're not going to start releasing older versions of the code under our name just because somebody doesn't like that decision.

CCatania01, 2016-05-29:
Bob,
Totally Agree.  

I will pick up Paul's 3.6 branch and make any updates to that for the near term.  There are a few fixes which need to be done, which would best be accomplished using my current development and test environment. 

I will then establish a system environment suitable for Java 1.8 and move all of my JMRI development to that platform.  Future work on CMRInet extensions and updates would be done in the current release branch. 

Chuck Catania

pabender, 2016-05-29:
Chuck,

On 05/29/2016 09:24 AM, Chuck Catania wrote:
> I will pick up Paul's 3.6 branch and make any updates to that for the
> near term. There are a few fixes which need to be done, which would
> best be accomplished using my current development and test [...]
You really need to setup a 1.8 development environment NOW.

There were some errors that you introduced in your code and some errors 
I introduced as part of the merge process.  Those errors have all been 
cleaned up in the current branch.

As Bob indicated, we're not going to take any more patches to code until 
the merge cleanly with head.  That means if you fix those in the 3.6 
version of this code, we won't accept those into any future release 
until you bring us something that works with head, which means YOU will 
be doing some work twice, because we aren't doing it for you.

Paul

bobjacobsen, 2016-07-03:
Removed 'afterNextProductionRelease' label as JMRI 4.4 is out.

closed 2017-10-29 by rhwood