JMRI/JMRI #1330: Improving JavaDoc and JavaDoc Process

bobjacobsen, 2016-05-20:
This is a master issue for a number of items around JMRI JavaDoc content.

By default, Java 1.8 interprets JavaDoc comments as HTML 4.01.  When JMRI was updated to use Java 1.8, that generated a lot of errors during JavaDoc processing, so we bypassed (most of) those with a ````-Xdoclint:all,-missing,-accessibility,-html,-syntax```` instruction to the JavaDoc compiler.

Still, the HTML nature can bite us, and @KenC57 noticed that the JavaDoc index had been corrupted by ````<s>```` appearing as text in a DCC++ JavaDoc comment, where it was interpreted as an HTML element.  That particular problem was fixed in PR #1327 but the general issue remains.

Here are the things that we need to do to get this back under control:

- [x] Fix extant problem with DCC++ (PR#1327)
- [ ] fix other existing JavaDoc errors and warnings
- [ ] Update recommendations on  [Documentation with JavaDoc and UML](http://jmri.org/help/en/html/doc/Technical/JavaDoc.shtml) page, see below
- [ ] Fix Jenkins to flag JavaDoc errors (maybe someday include in pre-merge GitHub CI)
- [ ] Tighten build warning options: Change flag & fix the bulk of the newly flagged errors and warnings
  - [ ]  ````html```` option - check that HTML is valid (initially 96 warnings and over a hundred errors)
  - [ ]  ````syntax```` option - check that JavaDoc syntax is valid (initially over 100 warnings and over 100 errors)

A few notes, also added to the [Documentation with JavaDoc and UML](http://jmri.org/help/en/html/doc/Technical/JavaDoc.shtml) page, follow.

If you want to include code in the comment (e.g. a code snippet), you can avoid trouble with < and & characters by including wrapping it with @code, as in {@code A & B < 0x20}.  If you’re using special characters, but don’t want it to be interpreted as code, you can use @literal as in {@literal ATT&T}.

The JavaDoc content is interpreted as HTML, so it should be valid HTML.  For example
````
 * Some text that forms a paragraph
 * <p>
 * Some more text. Note backslash on next line.
 * <p/>
 * Last text, start list
 * <ul>
 *    <li>List item
 * </ul>
````
Note that HTML 4.01 wants paragraph tags to be ended with ````</p>````` or```` </p>````, and that you can’t have a list inside a paragraph.

Some error messages and their translations:
 - malformed HTML; bad HTML entity: Probably use of &, < or > characters on the indicated line
 - bad use of ‘>’: Probably using ‘>’ as a character, e.g. A -> B. If so, replace with &gt; or wrap the line with @literal

Finally, note that this JavaDoc line is malformed:
````
 * @param foo
````
because it doesn’t include any explanatory text
````
 * @param foo Holds the latest Bar instance
````

We are currently suppressing the "accessibility" class of JavaDoc errors, which is 
mostly about providing extra information on tables and images, and the "missing" class of 
errors, which tags it if an @param or @return isn't provided in any places that could have 
received comments.  There are so many of these that it's not practical to tag them.

bobjacobsen, 2016-05-22:
Having trouble with getting Jenkins to show JavaDoc errors.  They errors are in the console log, but they don't show in the summary on the job page.  

To try to make this easier to debug, I created a specific project for testing:  http://jmri.tagadab.com/jenkins/job/Development/job/JavaDoc%20check/  It runs the ````ant javadoc```` job (and gets errors in the console), then (seems to be) set to scan for JavaDoc errors because it shows a graph, but the graph is always empty:

[image]

Help would be definitely appreciated!

bobjacobsen, 2016-05-23:
Summary of some work on the problem with JavaDoc not properly creating a Warnings Trend plot, having errors and warnings being listed, etc:  It's a Jenkins problem.  Apparently the JavaDoc scan in the compiler warnings plug-in wasn't updated when the Doclet output format was changed.

The actual code is in Jenkins GitHub [JavaDocParser.java](https://github.com/jenkinsci/warnings-plugin/blob/master/src/main/java/hudson/plugins/warnings/parser/JavaDocParser.java) and you can look at the regular expression used.  Note that it's a _Java_ regular expression, using some features to control group readout.

What we're getting (but not matching) in the console log:
````
  [javadoc] /var/lib/jenkins/jobs/Development/jobs/JavaDoc check/workspace/java/src/apps/Apps.java:683: warning: no description for @param
  [javadoc]      * @param wi
  [javadoc]        ^
  [javadoc] /var/lib/jenkins/jobs/Development/jobs/JavaDoc check/workspace/java/src/apps/AppsLaunchFrame.java:85: warning: no description for @param
  [javadoc]      * @param menuBar
  [javadoc]        ^
  [javadoc] /var/lib/jenkins/jobs/Development/jobs/JavaDoc check/workspace/java/src/apps/AppsLaunchFrame.java:86: warning: no description for @param
  [javadoc]      * @param wi
  [javadoc]        ^
  [javadoc] /var/lib/jenkins/jobs/Development/jobs/JavaDoc check/workspace/java/src/apps/AppsLaunchFrame.java:190: warning: no description for @param
  [javadoc]      * @param menuBar

  [javadoc] /var/lib/jenkins/jobs/Development/jobs/JavaDoc check/workspace/java/src/jmri/jmrit/withrottle/MultiThrottle.java:25: error: bad use of '>'
  [javadoc]  * <li> MT-L757<;>L757 On T throttle, remove loco L757. (Still has L1234)
  [javadoc]                  ^
  [javadoc] /var/lib/jenkins/jobs/Development/jobs/JavaDoc check/workspace/java/src/jmri/jmrit/withrottle/MultiThrottleController.java:26: error: malformed HTML
  [javadoc]      * Includes a separator - <;>
  [javadoc]                               ^
  [javadoc] /var/lib/jenkins/jobs/Development/jobs/JavaDoc check/workspace/java/src/jmri/jmrit/withrottle/MultiThrottleController.java:26: error: bad use of '>'
  [javadoc]      * Includes a separator - <;>
````

What does match (as seen in the "Generate Website Contents" job which was finding 5 warnings.  File "-:0", Line "-", Priority "Normal", Type "JavaDoc Tool", no Category)
````
  [javadoc] Building Context view for class jmri.util.JmriInsets
  [javadoc] Building Context view for class jmri.util.JmriJFrame
  [javadoc] javadoc: warning - dot: graph is too large for cairo-renderer bitmaps. Scaling by 0.495981 to fit
  [javadoc] Building Context view for class jmri.util.JmriJFrameAction
  [javadoc] Building Context view for class jmri.util.swing.sdi.JmriJFrameInterface
 
  [javadoc] Building Context view for class jmri.util.OrderedHashtable<K, V>.LocalEnumeration
  [javadoc] Building Context view for class jmri.util.OrderedProperties
  [javadoc] javadoc: warning - Warning: node c12312601, port p unrecognized
  [javadoc] Building Context view for class jmri.jmrit.ussctc.OsIndicator
  [javadoc] Building Context view for class jmri.jmrit.ussctc.OsIndicatorAction
````

Possible workarounds:
- Get Java 1.8 to emit the old format
- Create a custom console scanner in our Jenkins

Or, we might have to fix this...

mattharris, 2016-05-23:
I've added a custom parser that seems to do the trick - see:

http://jmri.tagadab.com/jenkins/job/Development/job/javadoc/13/warnings3Result/

mattharris, 2016-05-23:
Working through the first set of obvious errors at the moment

bobjacobsen, 2016-05-23:
I've got a bit more in the PR #1331 branch pushed, but not yet PR'd.

bobjacobsen, 2016-05-23:
Sorry, my mistake, those were merged already.

mattharris, 2016-05-23:
#1342 contains first pass, fixing the errors and obvious warnings

bobjacobsen, 2016-05-23:
Thanks for getting that filter to work!

How do we want to organize the CI tests around JavaDoc?  

I just looked at a few targets, and in the process noticed that the [Builds](http://jmri.tagadab.com/jenkins/job/Development/job/Builds/) target was doing the Ant "javadoc" target already, and the JavaDoc errors [were showing up as Java errors](http://jmri.tagadab.com/jenkins/job/Development/job/Builds/).

Do we want a separate JavaDoc target, with email, etc?  Or build it into some other(s), e.g. Builds?

mattharris, 2016-05-24:
Might need to look at modifying the javadoc parameters as it will currently stop reporting after 100 warnings are found.

Since we're on Java 1.8, we should be able to pass something like:

```xml
    <arg value="-Xmaxwarns"/>
    <arg value="200"/>
```

in the `javadoc` target to increase the number of output warnings from the default 100.

mattharris, 2016-05-24:
In #1350 I've increased the reported errors and warnings to 500.

mattharris, 2016-05-24:
> I just looked at a few targets, and in the process noticed that the [Builds](http://jmri.tagadab.com/jenkins/job/Development/job/Builds/) target was doing the Ant "javadoc" target already, and the [...]

>Do we want a separate JavaDoc target, with email, etc?  Or build it into some other(s), e.g. Builds?

I think that, for now, it's fine in the [Builds](http://jmri.tagadab.com/jenkins/job/Development/job/Builds/) target - it runs regularly enough to allow for issues to be seen quickly.

In addition, with the increased number of errors/warnings that should be visible once #1350 is merged, this should allow us to more quickly capture and resolve the 'low hanging fruit'

mattharris, 2016-05-24:
#1351 adds a (suppressed) definition in the `javadoc` target for the `@navassoc` tag used in the UML image generation. Without this, the regular JavaDoc tool throws an error due to the unknown tag.

bobjacobsen, 2016-05-26:
The vast majority of the remaining JavaDoc "syntax"-class errors are lines like:
````
    * @param p
````
which is [incomplete JavaDoc](http://jmri.sourceforge.net/help/en/html/doc/Technical/JavaDoc.shtml) because it doesn't contain explanatory text.

This regular expression (well, an extended regular expression to be precise) that matches those lines:
````
[[:space:]]*\*[[:space:]]*@param[[:space:]]*[[:alpha:]]*[[:space:]]*$
````
which makes it easy for me to do a bulk remove (almost all) of them. 

Should I do a bulk remove of them?

The only reason I can think of _not_ to is that somebody might be intending to go through and add a bunch of explanatory text Real Soon Now, and this would make less convenient.  But JavaDoc itself doesn't need these (it makes the parameter blocks from the actual parameters, not from empty @param statements) and the spec recommends against them because they just require maintenance and don't add value.

If the bulk remove should be done, @mattharris are you working on more JavaDoc updates? If so, I can wait until those have been merged so as not to cause conflicts, just let me know.

mattharris, 2016-05-26:
@bobjacobsen let's get rid of them - we can always go back through later and add documentation to them if deemed necessary.

#1361 has my latest updates - nothing more done since then.

pabender, 2016-05-26:
@mattharris @bobjacobsen 

I think somehow the parser for the Javadoc is causing an issue with the findbugs target.  It's reported failures the last 3 times it's run (builds 271,272,273).

Here's the relevant part of the build log:

`[FINDBUGS] Computing warning deltas based on reference build #270
ERROR: Step ‘Scan for compiler warnings’ aborted due to exception: 
java.io.IOException: Error: No warning parsers defined in the job configuration.
	at hudson.plugins.warnings.WarningsPublisher.perform(WarningsPublisher.java:297)
	at hudson.plugins.analysis.core.HealthAwarePublisher.perform(HealthAwarePublisher.java:68)
	at hudson.plugins.analysis.core.HealthAwareRecorder.perform(HealthAwareRecorder.java:259)
	at hudson.tasks.BuildStepCompatibilityLayer.perform(BuildStepCompatibilityLayer.java:78)
	at hudson.tasks.BuildStepMonitor$1.perform(BuildStepMonitor.java:20)
	at hudson.model.AbstractBuild$AbstractBuildExecution.perform(AbstractBuild.java:782)
	at hudson.model.AbstractBuild$AbstractBuildExecution.performAllBuildSteps(AbstractBuild.java:723)
	at hudson.model.Build$BuildExecution.post2(Build.java:185)
	at hudson.model.AbstractBuild$AbstractBuildExecution.post(AbstractBuild.java:668)
	at hudson.model.Run.execute(Run.java:1763)
	at hudson.model.FreeStyleBuild.run(FreeStyleBuild.java:43)
	at hudson.model.ResourceController.execute(ResourceController.java:98)
	at hudson.model.Executor.run(Executor.java:410)
Sending e-mails to: jmri-builds@lists.sourceforge.net daboudreau@hotmail.com paul.bender@acm.org rgj1927@pacbell.net mstevetodd@gmail.com
[BFA] Scanning build for known causes...
[BFA] No failure causes found
[BFA] Done. 0s
Finished: FAILURE`

mattharris, 2016-05-26:
@pabender I'm pretty sure it's down to a configuration change to the FindBugs job - there was a compiler warning scanner defined to run, but no actual parser linked to it.

This has since been changed - just waiting for the next FindBugs job to run, which I've just manually launched.

bobjacobsen, 2016-05-26:
Closing this, thanks everybody for all the help.  We've now got the Builds job on Jenkins giving us feedback with pretty extensive JavaDoc checking. Still lots of errors and warnings, but those will drop over time.

closed 2016-05-26 by bobjacobsen