JMRI/JMRI #1333: adding test coverage reports to the build

pabender, 2016-05-21:
I'm looking into adding test coverage reporting to the build process.

There's a package called JaCoCo ( http://eclemma.org/jacoco/ ) which includes ant targets
to build do code coverage analysis on the classes.  (eclemma is an eclipse plugin that makes uses of the jacoco library).

The code analysis is performed using a trace of testing.  It produces a nice HTML based report showing what classes have coverage and how much coverage they have.  You can also drill down into the classes to see what methods have tests.

In my local repository, I've modified the headlesstest and logalltest targets to generate the required trace file.  I've added a headless-coveragereport and alltest-coveragereport that depend, respectively, on the two testing targets to generate the actual report data.

(Here's an example report from the web: 
http://schnatterer.github.io/AntTestReporting/exampleReports/coverage/index.html 
instructions for how that was generated are here:
https://itaffinity.wordpress.com/2012/06/05/generating-and-customizing-junit-and-code-coverage-reports-with-ant/
)
Now, this is really most useful to us for alltest, because headlesstest misses all the GUI code.

So, now a couple of questions:
1) Where do we want the files to end up?  There is an execution trace file and a directory of reports produced as a result of the process.

2) Once I get this in place, it would be nice to have the results presented somewhere automatically, like the findbugs reports are, so we can reference these to find where tests are lacking.  I know the appveyer tests basically run alltest, but we don't appear to save anything off the filesystem.  Is it possible to get the jenkins server to run these?

Paul
P.S.
That wordpress site I liked to above also has some information on pretty formatting for the junit test results in [html(without frames)](http://schnatterer.github.io/AntTestReporting/exampleReports/junit/junit-noframes) [html (with frames)](http://schnatterer.github.io/AntTestReporting/exampleReports/junit/index.html) and [PDF](http://schnatterer.github.io/AntTestReporting/exampleReports/junit/junit.pdf).  It might be useful to get the HTML reports working, though it's pretty low priority.

rhwood, 2016-05-21:
Do you want to push them to http://coveralls.io?

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

mattharris, 2016-05-21:
We can publish via Jenkins - there is a JaCoCo plugin that should be
suitable.

I'll take a look at it.

pabender, 2016-05-21:
Sent from my iPad

On May 21, 2016, at 7:23 AM, Matthew Harris <notifications@github.com> wrote:

> We can publish via Jenkins - there is a JaCoCo plugin that should be
> suitable.
> 
> I'll take a look at it.

Ok.  I'll go ahead and push the changes up to github and create a pull request.

Paul

mattharris, 2016-05-23:
As this is now merged, I updated the job definition on Jenkins to now run from the `master` branch of the main repo.

[Build 4 (or later)](http://jmri.tagadab.com/jenkins/job/Development/job/JaCoCo/4/) are built from the main `JMRI/JMRI` repo.

pabender, 2016-05-23:
On 05/23/2016 03:45 AM, Matthew Harris wrote:
> As this is now merged, I updated the job definition on Jenkins to now
> run from the |master| branch of the main repo.
>
> Build 4 (or later)
> <http://jmri.tagadab.com/jenkins/job/Development/job/JaCoCo/4/> are
> built from the main |JMRI/JMRI| repo.
>

Cool.  Thanks.

Can you add one more exclude to the list of excluded classes.  I noticed 
that apps.tests.Log4jFixture.class is included, but that's really part 
of the test infrastructure, so it shouldn't be included
in the test coverage.

I'll get that added to the build.xml file as well.

Paul

pabender, 2016-05-24:
@mattharris or @bobjacobsen,
can one of you look at adding the new excludes from PR #1346  to the jacoco plugin on Jenkins?

I think those knock out most of the classes that don't match the **/*Test*.class pattern we of the testing infrastructure.

Paul

bobjacobsen, 2016-05-24:
Below is a list of all java/test .java files that don't end in either Test.java or Scaffold.java. Perhaps we could establish a convention that those names, plus maybe Demo.java (and perhaps Tests.java) are required, and simplify this a bit?

````
ButtonTestAction.java
CdiPanelDemo.java
CheckProgrammerNames.java
JUnitAppender.java
JUnitUtil.java
JmriServerParserTests.java
JsonMockConnection.java
LayoutManagerTest_AboutBox.java
LoadAndStoreTestBase.java
Log4JFixture.java
MockShutDownManager.java
MonitorFrameDemo.java
NetworkTreePaneDemo.java
OperationsSwingTestCase.java
OperationsTestCase.java
PackageDemo.java
SRCPClientParserTests.java
SRCPParserTests.java
SampleFactory.java
SamplePane.java
SchemaTestBase.java
SdiJfcUnitTests.java
SerialTurnoutTest1.java
SerialTurnoutTest2.java
SerialTurnoutTest3.java
SignalSystemTestUtil.java
SwingTestCase.java
TestBean1.java
TestConsistManager.java
TestTrafficController.java
TestUserPreferencesManager.java
TieToolFrameDemo.java
XmlLoadTests.java
````

bobjacobsen, 2016-05-24:
>can one of you look at adding the new excludes from PR #1346 to the jacoco plugin on Jenkins?

Done, I think, and [build number 10](http://jmri.tagadab.com/jenkins/job/Development/job/JaCoCo/10/) is running.

Are you sure you want **/*$*.class excluded?  AFAIK, that means that we're not checking coverage of any anonymous inner classes, whether from the java/src or java/test tree.

pabender, 2016-05-24:
Sent from my iPad

On May 24, 2016, at 1:43 AM, Bob Jacobsen <notifications@github.com> wrote:

> Below is a list of all java/test .java files that don't end in either Test.java or Scaffold.java.
> 
> Perhaps we could establish a convention that those names, plus maybe Demo.java (and perhaps Tests.java) are required, and simplify this a bit?
> 

The pattern is actually looking for *Test*.class and *Scaffold*.class, so that cuts a few more out of your list, leaving
> CdiPanelDemo.java
> CheckProgrammerNames.java
> JUnitAppender.java
> JUnitUtil.java
> JsonMockConnection.java
> Log4JFixture.java
> MockShutDownManager.java
> MonitorFrameDemo.java
> NetworkTreePaneDemo.java
> PackageDemo.java
> SampleFactory.java
> SamplePane.java
> TieToolFrameDemo.java
We already have Log4JFixture listed explicitly.  

Demo and JUnit we can definitely filter out with a pattern.  

Mock might be another good one to filter.  

The others I think we are better off listing explicitly ( that leaves 3 new explicit matches by my count, CheckProgrammerNames, SampleFactory and SamplePane.)

Paul

pabender, 2016-05-24:
> Demo and JUnit we can definitely filter out with a pattern.

We can't filter out Demo with a pattern after all. 

There are 2 java files matching "*Demo*" in the src tree:
./jmri/jmrit/operations/setup/LoadDemoAction.java
./apps/gui3/demo3/Demo3.java

Paul

bobjacobsen, 2016-05-24:
> There are 2 java files matching "Demo" in the src tree:
> ./jmri/jmrit/operations/setup/LoadDemoAction.java
> ./apps/gui3/demo3/Demo3.java

Demo3 can just go away. It's obsolete. 

More generally, I think we should have a naming convention around:
- *Test for JUnit tests
- *Demo for e.g. Popping a window for demos and manual tests
- *Scaffold, *Mock for test infrastructure objects

plus just a few JMRI infrastructure special cases. (Note those are only last words in the class name) That would be simple enough to document and even test at CI-time.

pabender, 2016-05-24:
On 05/24/2016 11:05 AM, Bob Jacobsen wrote:
>
>  > There are 2 java files matching "Demo" in the src tree:
>  > ./jmri/jmrit/operations/setup/LoadDemoAction.java
>  > ./apps/gui3/demo3/Demo3.java
>
> Demo3 can just go away. It's obsolete.
>

Ok.  Demo3 is actually referenced in apps.gui3.Apps3.java as being an 
example.

I don't see where anything else in the apps.gui3.demo3 package is 
referenced.

I'll go ahead and pull the package, and if it compiles, I'll push it up 
as a PR.

> More generally, I think we should have a naming convention around:
> - *Test for JUnit tests
> - *Demo for e.g. Popping a window for demos and manual tests
> - *Scaffold, *Mock for test infrastructure [...]

Ok, here are the current list of excludes I have:

         <exclude name="**/*Test*.class" />
         
         <exclude name="**/*Scaffold.class" />
         <exclude name="**/*JUnit*.class" />
         <exclude name="**/*Demo.class" />
         <exclude name="**/*Mock*.class" />
         <exclude 
name="jmri/jmrit/symbolicprog/tabbedframe/CheckProgrammerNames.class" />
         <exclude name="jmri/jmrix/openlcb/SampleFactory.class" />
         <exclude name="jmri/util/swing/SamplePane.class" />
         <exclude name="apps/tests/Log4jFixture.class" />

That does Test,Mock, and JUnit anywhere in the class name, but Scaffold 
and Demo only as the last word, and gives us 4 special cases.

Paul

mattharris, 2016-05-24:
As mentioned in #1354, build 12 or later of the [JaCoCo job on Jenkins](http://jmri.tagadab.com/jenkins/job/Development/job/JaCoCo/) has been updated with the new exclusion list.

pabender, 2016-05-24:
@mattharris 
I'm still seeing Log4JFixture in the coverage report.

The console output shows this for the exclusions:
[JaCoCo plugin] exclusions: [**/*Test*.class, **/*Scaffold.class, **/*JUnit*.class, **/*Demo.class, **/*Mock*.class, **/jmri/jmrit/symbolicprog/tabbedframe/CheckProgrammerNames.class, **/jmri/jmrix/openlcb/SampleFactory.class, **/jmri/util/swing/SamplePane.class, **/apps/tests/Log4jFixture.class]

Since the last 4 are specific classes, can we try removing the "**/" from the begining of just those 4.

Paul

pabender, 2016-05-24:
Oh, I see what's happening.  

A typo on my part.

We have Log4jFixture but it should be Log4JFixture (capital J)

Paul

pabender, 2016-05-24:
typo fixed in PR #1355

mattharris, 2016-05-24:
Ah, I'll change that in the job config.

And, BTW, if you want to explicitly write some 'code', use the backtick 'quote' mark '`' around what your writing vis:

`
[JaCoCo plugin] exclusions: [**/*Test*.class, **/*Scaffold.class, **/*JUnit*.class, **/*Demo.class, **/*Mock*.class, **/jmri/jmrit/symbolicprog/tabbedframe/CheckProgrammerNames.class, **/jmri/jmrix/openlcb/SampleFactory.class, **/jmri/util/swing/SamplePane.class, **/apps/tests/Log4jFixture.class]
`

pabender, 2016-05-24:
Thanks @mattharris

mattharris, 2016-05-24:
No probs. Jenkins now updated and new build launched

pabender, 2016-05-29:
I think we have this up and running pretty well now, so I am closing this isssue.

closed 2016-05-29 by pabender