JMRI/JMRI #1382: Update the -Xmx option value?

bobjacobsen, 2016-05-28:
A couple people have had large-operation problems that apparently were solved by increasing the max heap size (Xmx) from JMRI's default. We currently calculate that (for users using our released startup scripts). I've appended the specifics at the bottom of this, but roughly the calculation is "physical memory / 2, up to a max of 768MB". (Note: we use a simpler algorithm within Ant build.xml, setting the max to 640MB when running applications) 

If we're providing too small a Xmx size, we could calculate a larger one; either allowing a larger max (lots of machines are more than 1.5GB now), or allowing more than a half of the physical memory.

Another approach is to remove the option.  Long ago (up to Java 1.5?), the default size was 256MB. That's too small for JMRI, even then, and setting the option was needed. But maybe we could just allow Java to set the maximum to a default now?  It's hard to determine the Java 1.8 behavior from the documentation, as there are lots of garbage collectors, different defaults for different installs, etc.  Depending on where you look, we may already be doing something similar to the default [(1)](http://docs.oracle.com/javase/8/docs/technotes/guides/vm/gctuning/parallel.html#default_heap_size), [(2)](https://docs.oracle.com/javase/8/docs/technotes/tools/windows/java.html#BABHDABI),[(3)](http://docs.oracle.com/cd/E13150_01/jrockit_jvm/jrockit/jrdocs/refman/optionX.html#wp1035652), [(4)](https://docs.oracle.com/cd/E19159-01/819-3681/abeii/index.html). But it sounds like it's doing something at or below what we're doing with our startup scripts.

There is a risk to setting the value too high.  The JVM does not work well when paged. If Xmx exceeds _available_ physical memory, and Java _uses_ that space, then an OS low memory condition can bring the machine to a stop as it does continuous garbage collections.  I think avoiding that is the reason that Xmx is limited to a comparatively small fraction of available memory.

I recommend that in JMRI 4.5.1 we change this calculation to 
````
  Xmx = 192MB < (memory-256)*0.75 < 2GB
````

I'm starting the discussion now, so we have some time to consider this in advance, or in case somebody wants to change this before JMRI 4.4

---

Summary of current calculations

In scripts/AppScriptTemplate:
````
heap_size () {
    mem=640
    case "`uname`" in
        Linux*)
            mem=`cat /proc/meminfo | grep MemTotal | tr -d [:space:][:alpha:]:`
            mem=`expr $mem / 1024`
            ;;
        SunOS*)
            mem=`/usr/sbin/prtconf | grep Memory | /usr/bin/tr -dc '[0-9]'`
            ;;
        Darwin*)
            mem=`/usr/sbin/sysctl hw.memsize | tr -d [:alpha:][:space:].:`
            mem=`expr $mem / 1048576`
            ;;
        *)
            ;;
    esac
    if [ -z "$mem" ] ; then
        mem=640
    fi
    mem=`expr $mem / 2`
    if [ $mem -gt 768 ] ; then
        mem=768
    elif [ $mem -lt 192 ] ; then
        mem=192
    fi
    echo $mem
    return 0
}
````
and in scripts/WinInstallFiles/LaunchJMRI.nsi for Windows:
````
!define MINMEM     96                             ; Minimum memory in Mbyte
!define HIMAXMEM   768                            ; Highest Maximum memory
!define LOMAXMEM   192                            ; Lowest Maximum memory

 ; -- Get the memory status
  StrCmp ${ARCH_32BIT} $x64 Notx64
  Call GetSystemMemoryStatus64
  Goto CalcMem
  Notx64:
    Call GetSystemMemoryStatus
  CalcMem:
  System::Int64Op $4 / 1048576
  Pop $4
  DetailPrint "PhysicalMemory: $4m"
  ; -- Check that physical memory is >= MINMEM
  IntCmp $4 ${MINMEM} cmp_max cmp_min_lt cmp_max
  cmp_min_lt:
    ; -- If insufficient physical memory, stop with an error
    MessageBox MB_OK|MB_ICONSTOP "Not enough available memory to start. JMRI requires at least ${MINMEM} MBytes memory."
    Goto Exit
  cmp_max:
    ; -- Check if 1/2 physical memory is between LOMAXMEM and HIMAXMEM
    IntOp $CALCMAXMEM $4 / 2
    DetailPrint "Check if $CALCMAXMEMm is between ${LOMAXMEM}m and ${HIMAXMEM}m"
    IntCmp $CALCMAXMEM ${HIMAXMEM} cmp_done cmp_max_lt cmp_max_gt
  cmp_max_gt:
    ; -- 1/2 physical memory is greater than HIMAXMEM,
    ; -- so set to HIMAXMEM
    DetailPrint "Greater than HIMAXMEM: ${HIMAXMEM}m"
    StrCpy $CALCMAXMEM ${HIMAXMEM}
    Goto cmp_done
  cmp_max_lt:
    ; -- 1/2 physical memory is less than HIMAXMEM
    ; --
    DetailPrint "Less than HIMAXMEM: ${HIMAXMEM}m"
    DetailPrint "Check if $CALCMAXMEMm is less than ${LOMAXMEM}m"
    IntCmp $CALCMAXMEM ${LOMAXMEM} cmp_done cmp_use_lomax cmp_done
    cmp_use_lomax:
      DetailPrint "Use LOMAXMEM: ${LOMAXMEM}m"
      StrCpy $CALCMAXMEM ${LOMAXMEM}
  cmp_done:
  DetailPrint "MinMemory: ${MINMEM}m"
  DetailPrint "MaxMemory: $CALCMAXMEMm"

  StrCpy $OPTIONS "$OPTIONS -Xmx$CALCMAXMEMm"
````

rhwood, 2016-05-28:
While not questioning that the users experiencing issues had success by more than doubling the -Xmx value, do we feel a real need to more than double it for ourselves, or is there any sense that the users who did double it, experimentally set the values, or merely claimed a large space?

Do we have any sense that most JMRI users exclusively run JMRI on their computers and are not also running other processes (wittingly or not) that might cause an excessive -Xmx setting to cause swapping to occur (I'm thinking more about people running voice communications or remote session software)?

I'm more inclined to think that we should only set -Xmx if the results of
```
java -XX:+PrintFlagsFinal -version 2>/dev/null | grep MaxHeapSize | awk '{print $4}'
```
are less than 768 MB (i.e. use our own algorithm if the JVM has not set it above a certain size by default)

http://stackoverflow.com/a/13871564 led to the calculation and includes the start of a Windows calculation as well.

pabender, 2016-05-29:
Sent from my iPad

On May 28, 2016, at 5:24 PM, Bob Jacobsen <notifications@github.com> wrote:

> There is a risk to setting the value too high. The JVM does not work well when paged. If Xmx exceeds available physical memory, and Java uses that space, then an OS low memory condition can bring [...]
I've had issues with getting JMRI to run correctly on a Raspberry Pi.  I'll check and see what value I used for that 
after my daughter goes to bed tonight.  ( I think this was actually in the ant builds I had to make adjustments...., so
I could compile....)

Paul

pabender, 2016-05-29:
On 05/28/2016 08:15 PM, Paul Bender wrote:
> I've had issues with getting JMRI to run correctly on a Raspberry Pi. 
>  I'll check and see what value I used for that
> after my daughter goes to bed tonight.  ( I think this was actually in 
> the ant builds I had to make adjustments...., so
> I could compile....)
>
I just double checked, I had to set the compiler maximum memory size to 
1024m for JMRI to compile on a Raspberry PI Model B2.

Otherwise, it churns for about 45 minutes, and then it fails, running 
out of memory of course.

Paul

rhwood, 2016-05-29:
> I just double checked, I had to set the compiler maximum memory size to
1024m for JMRI to compile on a Raspberry PI Model B2.

Isn't that 100% of physical memory? What size swap did you have and do you have any sense of the size of tmpfs partitions (or how much virtual RAM was the machine using)?

pabender, 2016-05-29:
On May 29, 2016, at 6:12 AM, Randall Wood <notifications@github.com> wrote:
> I just double checked, I had to set the compiler maximum memory size to
> 1024m for JMRI to compile on a Raspberry PI Model B2.
> 
> Isn't that 100% of physical memory?
> 
Yes, it is.
> What size swap did you have and do you have any sense of the size of tmpfs partitions (or how much virtual RAM was the machine using)?
> 
Honestly, it's a default Raspbian image with oracle java installed.

I just checked the swap settings, and it uses a 100m swap file, which seems tiny to me.

The physical memory is also split between the CPU and GPU, so there isn't necessarily 1024m available, even with swap space.  I haven't adjusted that from the defaults.

I could probably use a smaller value, but I think I found something that worked, and didn't try to find an optimal value.

Paul

bobjacobsen, 2016-07-03:
Removed 'afterNextProductionRelease' label as JMRI 4.4 is out.

bobjacobsen, 2016-07-03:
@rhwood updated this for the POSIX (Mac, Linux) launchers in PR #1548.  Is something needed for Windows, or can this be closed?

rhwood, 2016-11-02:
Something is needed for Windows.

mattharris, 2016-11-28:
Pretty much got this covered now - not reading the current heap at the moment, rather doing the calculation always. PR to follow

mattharris, 2016-11-28:
#2610 should cover this for Windows

closed 2016-12-06 by bobjacobsen