JMRI/JMRI #1383: Selecting connection for startup actions when multiple of a connection

KenC57, 2016-05-28:
Currently it seems there is no way to pick which connection for a number of tools in the startup actions when there are more than one of the same type. Example is LocoNet Monitor, if you have more than one LocoNet, it will only open the default connection. I'm thinking we either need a way to pass arguments to this or that the list would show multiple (LocoNet Monitor, LocoNet 2 Monitor, LocoNet 3 Monitor) so you can pick which you want.

This gap extends to things like a programming connection tool and likely for all the tools listed in the connection that are offered in the startup actions.
-ken c

bobjacobsen, 2016-06-06:
> Currently it seems there is no way to pick which connection for a number of tools in the startup actions when there are more than one of the same type. Example is LocoNet Monitor, if you have more [...]

This might be better moved to jmri-developers to get a wider discussion.

The preferences UI is going to be key: _How_ to define which LocoNet should have a monitor window open.

rhwood, 2016-06-15:
The UI is pretty simple: in the editor for startup actions, a combobox with the valid connections for that action can be shown (and disabled if the action is not bound to a connection).

The hard part really is determining:
1. if an action needs to be bound to a connection
2. which connections an action can be bound to

I think that if every Action that needs to be bound to a SystemConnectionMemo implemented an interface that defined the following this could be done:
```
public SystemConnectionMemo getSystemConnectionMemo();

public void setSystemConnectionMemo(SystemConnectionMemo memo);

/**
 * Get a list of {@link jmri.jmrix.SystemConnectionMemo} subclasses that the
 * implementing class accepts.
 *
 * @return Array of SystemConnectionMemo subclasses.
 */
public Class<? extends SystemConnectionMemo>[] getSystemConnectionMemoClasses();
```

The first two methods are needed for the storage and retrieval of the system connection in the ConfigureXml classes for the startup actions, and the last method would be needed for the preferences panel to display the correct list of system connections to the user (this is important so that, for example, a user cannot connect a LocoNet Monitor to an NCE system).

rhwood, 2016-06-17:
#1520 provides a framework for making it possible to assign a connection on startup to connection dependent Actions. 

Actually making connection dependent Actions accept assigned connections is an exercise left to others who may be more familiar with which Actions need that information.

Actions that need to be connection-aware should implement the ```jmri.jmrix.SystemConnectionAction``` interface.

Startup buttons bound to a connection report that connection in a ToolTip, as do any actions listed in the Startup Items preferences.

Any issues with the implementation or user interface for assigning connections to actions, making users aware of which connection an Action is targeted against, should be addressed here or in a new issue on GitHub.

rhwood, 2016-07-18:
The interface [jmri.jmrix.swing.SystemConnectionAction](http://jmri.org/JavaDoc/doc/jmri/jmrix/swing/SystemConnectionAction.html) needs to be implemented for any startup actions that can be bound to a connection. Individual issues (preferably pull requests) should be opened for any startup action that needs this done to it.

closed 2016-07-18 by rhwood