JMRI/JMRI #1388: AppVeyor failures in Operations tests

rhwood, 2016-05-29:
We are getting intermittent, but frequent, failures at [AppVeyor](https://ci.appveyor.com/project/JMRI/jmri/history) with two of the tests introduced in #1381:
```
    [junit] Testcase: testNewFrameCreation(jmri.jmrit.operations.automation.AutomationTableFrameGuiTest):	FAILED
    [junit] expected:<1c[2]> but was:<1c[1]>
    [junit] junit.framework.ComparisonFailure: expected:<1c[2]> but was:<1c[1]>
    [junit] 	at jmri.jmrit.operations.automation.AutomationTableFrameGuiTest.testNewFrameCreation(AutomationTableFrameGuiTest.java:78)
    [junit] 	at junit.extensions.jfcunit.JFCTestCase$5.run(JFCTestCase.java:580)
    [junit] 	at junit.extensions.jfcunit.JFCTestCase$3.run(JFCTestCase.java:478)
    [junit] 	at java.lang.Thread.run(Thread.java:745)
    [junit] 
    [junit] 
    [junit] Testcase: testFrameCreation(jmri.jmrit.operations.automation.AutomationsTableFrameGuiTest):	FAILED
    [junit] null
    [junit] junit.framework.AssertionFailedError
    [junit] 	at jmri.jmrit.operations.automation.AutomationsTableFrameGuiTest.testFrameCreation(AutomationsTableFrameGuiTest.java:21)
    [junit] 	at junit.extensions.jfcunit.JFCTestCase$5.run(JFCTestCase.java:580)
    [junit] 	at junit.extensions.jfcunit.JFCTestCase$3.run(JFCTestCase.java:478)
    [junit] 	at java.lang.Thread.run(Thread.java:745)
```
Of the 11 AppVeyor builds following the #1381 merge, 6 failed on these two tests, 4 passed, and 1 failed for unrelated reasons.

bobjacobsen, 2016-05-29:
On the off chance this is a threading issue with the new operations automation support, PR #1395  adds use of "waitFor" for first test above and similar.  Not sure what's going wrong with the 2nd test, line numbers may have changed.

bobjacobsen, 2016-05-29:
Just noticed that the class names in those two failed tests are _not_ the same....  Sorry for any confusion.

bobjacobsen, 2016-05-30:
Looks like #1395 might have fixed these.  Let's keep an eye on it for a day or two to make sure.

rhwood, 2016-06-10:
This appears to have been fixed.

closed 2016-06-10 by rhwood