JMRI/JMRI #1410: Why is ActiveFlag an abstract class?

pabender, 2016-05-31:
I'm bulk writing tests to test the ActiveFlag classes in the various jmri.jmrix sub-packages.

One of the tests is failing (it happens to be the loconet one...) for some reason I can't explain (the rest work fine).

I was going to try making sure the classes were working with a fresh copy every time the tests run, but, the class is abstract, so it can be instantiated... 

Paul

pabender, 2016-05-31:
Except for the failing LocoNet test, the commit with all the ActiveFlag tests is in PR #1412.

The LocoNet test is the same as the ones uploaded in the PR, just referencing a different package.

Paul

bobjacobsen, 2016-05-31:
> Why is ActiveFlag an abstract class?

I think it's a hold-over from when somebody wanted to be able to build JMRI with only certain systems in the jar file.  It used reflection to create the menus, etc, that reflected the available (e.g. loadable, because in the class path) systems.

But JMRI's structure has changed a lot, and that's not needed now.  With the most recent architecture, you can look in InstanceManager to see if a system is available.

So it's somewhat obsolete, but it looks like ActiveFlag is still used for 

- letting the SignalHeadTable know whether the Grapevine system is active, so a special signal option can be shown - this can be done using InstanceManager instead
- letting the NCE consist support in operations know if NCE is configured - ditto

plus being used in jmrix.ActiveSystemsMenu for systems that haven't been fully migrated to the most recent multi-system architecture.  

Might be a good summer project to 
- deprecate ActiveFlag
- complete that migration for those systems that don't have it (e.g. still use ActiveFlag in ActiveSystemsMenu)
- remove ActiveFlag when possible (either at the end, or system by system)

pabender, 2016-05-31:
Probably not a summer project, but a good week or so.

I've closed my PR for the tests, and will start work on removing the ActiveFlag.

bobjacobsen, 2016-06-02:
The Test for LocoNet ActiveFlag might be failing because jmrix.ActiveSystemFlagTest uses jmrix.loconet as an example, probably messes with static state.

pabender, 2016-06-02:
I actually changed the reference in the test to use "jmri.jmrix.grapevine.ActiveFlag" instead of jmri.jmrix.loconet.ActiveFlag.  I haven't pulled the jmri.jmrix.grapevine.ActiveFlag yet.

Where are you seeing the test fail?  It succeeded locally and on appveyor from the branch in github.

Paul

bobjacobsen, 2016-06-02:
> Where are you seeing the test fail? It succeeded locally and on appveyor from the branch in github.

Not seeing it myself, just noticed the LocoNet reference while working on something else.

pabender, 2016-06-02:
Ok.  FYI, my branch, if you haven't seen it has been pushed to github, but no pull request yet.

https://github.com/pabender/JMRI/tree/RemoveActiveFlag

Paul

bobjacobsen, 2016-07-03:
Removed 'afterNextProductionRelease' label as JMRI 4.4 is out.

rhwood, 2016-07-18:
@pabender Is this complete?

pabender, 2016-07-18:
On 07/18/2016 04:56 PM, Randall Wood wrote:
> @pabender <https://github.com/pabender> Is this complete?
>

No.
I still have several systems to finish.

Paul

bobjacobsen, 2016-10-06:
@pabender Thanks for all the work in this area.

Looks like there's no longer any references to ActiveSystemFlag except itself:
```
% grep -rl ActiveSystemFlag java/src/
java/src//jmri/jmrix/ActiveSystemFlag.java
% grep -rl ActiveSystemFlag java/test
java/test/jmri/jmrix/ActiveSystemFlagTest.java
java/test/jmri/jmrix/PackageTest.java
```

Can we close this?

pabender, 2016-10-06:
No, because it's the ActiveFlag files we are cleaning up:

bash-4.3$ grep -rl ActiveFlag java/src/
java/src/jmri/jmrix/dcc4pc/ActiveFlag.java
java/src/jmri/jmrix/dcc4pc/serialdriver/SerialDriverAdapter.java
java/src/jmri/jmrix/oaktree/ActiveFlag.java
java/src/jmri/jmrix/oaktree/serialdriver/SerialDriverAdapter.java
java/src/jmri/jmrix/tmcc/ActiveFlag.java
java/src/jmri/jmrix/tmcc/serialdriver/SerialDriverAdapter.java
java/src/jmri/jmrix/maple/ActiveFlag.java
java/src/jmri/jmrix/maple/serialdriver/SerialDriverAdapter.java
java/src/jmri/jmrix/grapevine/ActiveFlag.java
java/src/jmri/jmrix/grapevine/nodeconfig/NodeConfigFrame.java
java/src/jmri/jmrix/grapevine/serialdriver/SerialDriverAdapter.java
java/src/jmri/jmrix/pricom/ActiveFlag.java
java/src/jmri/jmrix/rps/ActiveFlag.java
java/src/jmri/jmrix/rps/serial/SerialAdapter.java
java/src/jmri/jmrix/sprog/ActiveFlag.java
java/src/jmri/jmrix/sprog/ActiveFlagCS.java
java/src/jmri/jmrix/sprog/serialdriver/SerialDriverAdapter.java
java/src/jmri/jmrix/nce/ActiveFlag.java
java/src/jmri/jmrix/nce/networkdriver/NetworkDriverAdapter.java
java/src/jmri/jmrix/nce/usbdriver/UsbDriverAdapter.java
java/src/jmri/jmrix/nce/simulator/SimulatorAdapter.java
java/src/jmri/jmrix/nce/serialdriver/SerialDriverAdapter.java
java/src/jmri/jmrix/direct/ActiveFlag.java
java/src/jmri/jmrix/direct/serial/SerialDriverAdapter.java
java/src/jmri/jmrix/acela/ActiveFlag.java
java/src/jmri/jmrix/qsi/ActiveFlag.java
java/src/jmri/jmrix/qsi/serialdriver/SerialDriverAdapter.java
java/src/jmri/jmrix/bachrus/ActiveFlag.java
java/src/jmri/jmrix/bachrus/serialdriver/SerialDriverAdapter.java
java/src/jmri/jmrix/ActiveSystemFlag.java
java/src/jmri/jmrix/secsi/ActiveFlag.java
java/src/jmri/jmrix/secsi/serialdriver/SerialDriverAdapter.java
java/src/jmri/jmrix/ActiveSystemsMenu.java
java/src/jmri/jmrix/srcp/ActiveFlag.java
java/src/jmri/jmrix/srcp/networkdriver/NetworkDriverAdapter.java
java/src/jmri/jmrix/cmri/serial/ActiveFlag.java
java/src/jmri/jmrix/easydcc/ActiveFlag.java
java/src/jmri/jmrix/easydcc/networkdriver/NetworkDriverAdapter.java
java/src/jmri/jmrix/easydcc/serialdriver/SerialDriverAdapter.java

So there is still more work to do.

pabender, 2019-04-19:
I believe this issue is now resolved

closed 2019-04-19 by pabender