JMRI/JMRI #1417: Update Eclipse version we use for builds?

bobjacobsen, 2016-06-01:
Our Jenkins diagnostic builds run a step using the Eclipse compiler, as historically it generated good warnings.  It's the 2nd plot from the top [on the Builds job](http://jmri.tagadab.com/jenkins/job/Development/job/Builds/), among other places.

For reliability/consistency, we include that compile as a jar file in Git.  Currently, our README file says that it's (I haven't confirmed this by comparing files):
````
ecj.jar
- Eclipse compiler 3.5.1 from 
    - http://www.eclipse.org/downloads/download.php?file=/eclipse/downloads/drops/R-3.5.1-200909170800/ecj-3.5.1.jar
````
That's from September 2009. Eclipse has now released [up through version 4.5.1](http://mvnrepository.com/artifact/org.eclipse.jdt.core.compiler/ecj) of the compiler.

Do we want to update this?

I started looking into it because of a false-positive that it's throwing against java/test/jmri/server/json/roster/JsonRosterHttpServiceTest.java where it considers this to be a "could be null" error on the 2nd line:
````
        assertNotNull(exception);
        assertEquals(HttpServletResponse.SC_INTERNAL_SERVER_ERROR, exception.getCode());
````

There's an [Eclipse bug report](https://bugs.eclipse.org/bugs/show_bug.cgi?id=472618) against this, which shows fixed (last summer), and should be available in Eclipse 4.6 starting with the milestone 1 version.  They haven't fully released 4.6 yet (AFAIK), but there's a an (RC3 available)[http://download.eclipse.org/eclipse/downloads/drops4/S-4.6RC3-201605252000/]

An update is likely to show us new warnings (good, though it'll take some time to work through them), so I'd like to do the update sometime reasonable.  We could go to 4.5.1, on the principle of moving forward when we can, move to 4.6 when it's available (as it should fix these false positives at least), or move to a 4.6 release candidate before 4.6 final is available.

Thoughts?  

If there are no problems foreseen, I'll probably pull the 4.6RC3 and see how it behaves on a branch.

mattharris, 2016-06-08:
Bit late responding, but I think that this is a good idea. I also agree with the `afterNextProductionRelease` target.

As far as the version to use, I'd be tempted to go for 4.6RC3 (or even RC4 that I now see is available).

A quick look at their [overall release schedule](http://wiki.eclipse.org/Neon/Simultaneous_Release_Plan) and [sub-project specific schedule](http://www.eclipse.org/projects/project-plan.php?planurl=http://www.eclipse.org/eclipse/development/plans/eclipse_project_plan_4_6.xml) shows that it is expected to release 4.6 final later this month meaning that, when the time comes around to work on 4.5.1, the 4.6 final _should_ be available.

bobjacobsen, 2016-06-08:
Thanks.  

The 4.6RC3 compiler is on the [eclipse-compiler-update](https://github.com/JMRI/JMRI/tree/eclipse-compiler-update) branch if anybody would like to try it. It would be particularly useful if an Eclipse user could check that it's OK.

It seems to be working. The false-positive warnings have gone away.  With our current set of warning options, the total number of warnings drops to 34, all of which seem valid. 

Will update if there's a 4.6 final version before JMRI's 4.5.1 comes out.

bobjacobsen, 2016-07-03:
Removed 'afterNextProductionRelease' label as JMRI 4.4 is out.

bobjacobsen, 2016-07-03:
#1522 resolved this.

closed 2016-07-03 by bobjacobsen