JMRI/JMRI #1427: jmri.Address, jmri.TurnoutAddress, jmri.SensorAddress Obsolete?

pabender, 2016-06-03:
Going through some tests for XPressNet, I ran across a random call to create a jmri.TurnoutAddress object, which isn't used by XPressNet at all.

jmri.TurnoutAddress is supposed to be used to hold the user name and system name of a turnout.  This is based on jmri.Address.  This appears to only be used in tests.

I also found there is jmri.SensorAddress.  That apperas to have no uses at all.

Before I remove these bits of obsolete code, does anyone know of any legitimate uses of these in JMRI today?

Paul

bobjacobsen, 2016-06-03:
> Going through some tests for XPressNet, I ran across a random call to create a jmri.TurnoutAddress object, which isn't used by XPressNet at all.
>
> jmri.TurnoutAddress is supposed to be used to hold [...]

I haven’t checked the repository history, but I think those are associated with the jmri.layout effort.  That was an attempt to create a _single_ model for the overall layout:  A tree that contained everything, instead of separate collections of Sensors vs Turnouts vs SignalHeads with different subgroups for specific (physical) types.

Like other efforts of this type, it soon foundered on the problem of different people thinking about the logic of their layouts in different ways.  There’s just not enough gained by trying to force fit variation into a single model.

It might be best to go our usual “deprecated across at least one production release, and then delete when we remember to” route, but I think all that should be removed eventually.

pabender, 2016-06-03:
Ok.  PR #1428 takes care of the deprecation and removes the calls to create jmri.TurnoutAddress objects from the tests.

pabender, 2016-06-03:
Sometime after the next production release, there is one task remaining

- [ ] Remove jmri.Address, jmri.TurnoutAddress, and jmri.SensorAddress

bobjacobsen, 2016-06-03:
Also after the next production release:
- [ ] remove the jmri.layout package, deprecated in JMRI 4.3.5

pabender, 2016-06-22:
Bob,

What version should we target these classes for removal, in light of #1441?

Paul

bobjacobsen, 2016-06-22:
Although they (those classes and the layout package) have not been _marked_ as deprecated for long, I think it's OK to remove them for 4.5.1. They haven't had any real use in JMRI or any of the significant public interfaces. And in the unlikely event this breaks something, it's easy to put them back.

pabender, 2016-06-23:
Closed via #1560 and #1561

closed 2016-06-23 by pabender