JMRI/JMRI #1431: Problem with French help files in help/fr

bobjacobsen, 2016-06-04:
(See PR #1429 for initial changes)

Running the JHelpDev indexing tool on the French help files in the the help/fr directory gives several errors:

- an IOException due to not being able to write "Document"

- warnings about unrecognized (e.g. bad) encodings, like iso-iso-8859-1 (sic)

- warnings that otherwise valid encodings (iso-8859-1) are not supported

Those last two seem to have several things going on:

- If the file shows as stored with ISO encoding, even though it's plan text, that gives an error
- If it has a meta tag for encoding, ditto

The failure-to-write-Document error may be due to problems in the files:  Google found a mention of malformed `<title>` elements, but it could also be some of the `<meta>` tags for encoding.  

It's not clear that this is causing any runtime issues.

silverailscolo, 2016-06-12:
The encoding definition definitely has a mismatch with the contents:
[image]
Clicking Cancel on the first example file crashes TextWrangler, so we should fix this.

bobjacobsen, 2016-12-24:
#2753 makes sure that the map generation now works, but the indexing is still failing.

closed 2021-05-15 by stale[bot]