JMRI/JMRI #1435: junit.framework.Assert deprecated

DanielBoudreau, 2016-06-05:
Is there a reason we haven't changed all references of junit.framework.Assert to org.junit.Assert?

rhwood, 2016-06-05:
There is not a one-to-one mapping of the two classes, and really is no need to change anything in the tests unless you want to since the only required change between JUnit 3 and 4 has been made.

DanielBoudreau, 2016-06-05:
I’m only bring this up to get rid of the annoying deprecated annotations that now litter the test classes. Pretty easy to do a global replace, just wondering why no one has done it.

rhwood, 2016-06-05:
> I’m only bring this up to get rid of the annoying deprecated annotations that now litter the test classes. Pretty easy to do a global replace, just wondering why no one has done it.

```org.junit.Assert``` is not a straightforward replacement of ```junit.framework.Assert``` throughout our tests unless you also refactor every test to not extend ```junit.framework.TestCase```. I tried it, and ultimately did not commit those changes because I broke a significant number of tests doing so.

DanielBoudreau, 2016-06-05:
I was able to convert all of the operation tests.  Only issue was related to testing doubles, which the org.junit.Assert does not support.

pabender, 2016-06-05:
On Jun 5, 2016, at 12:34 PM, Daniel Boudreau <notifications@github.com> wrote:
> I was able to convert all of the operation tests. Only issue was related to testing doubles, which the org.junit.Assert does not support.
> 

My read of the current Javadoc for JUnit ( http://junit.sourceforge.net/javadoc/org/junit/Assert.html ) shows that assertEquals( double expected , double actual ) and asserEquals( string reason, double expected, double actual ) are both deprecated in favor of assertEquals( double expected , double actual, double delta ) and asserEquals( string message, double expected, double actual, double delta ) respectively.

So basically, the deprecation is telling us we need to include a tolerance for floating point values, which is a good thing.  There are actually very good technical reasons for this, most notably that floating point values on a computer are discrete representations of continuous real valued things.

Paul

bobjacobsen, 2016-06-05:
> My read of the current Javadoc for JUnit ( http://junit.sourceforge.net/javadoc/org/junit/Assert.html ) shows that assertEquals( double expected , double actual ) and asserEquals( string reason, [...]

(Exact comparisons can still be done with those new method signatures, AFAIK, just setting the delta to zero.)

> So basically, the deprecation is telling us we need to include a tolerance for floating point values, which is a good thing. There are actually very good technical reasons for this, most notably that floating point values on a computer are discrete representations of continuous real valued things.

Unlike some other languages, there are no platform dependencies in Java numerical calculations: If you do a calculation on one machine, you will get the same result on every machine, every time, forever.

So although there are good reasons for “within a certain accuracy” comparisons in operational code, particular code that’s using input from the real world, it’s often better to do exact comparisons in unit tests.   Once one starts accepting that unit tests (which should run the same way every time) are not exact, bad stuff can creep in. 

Yes, a refactoring that e.g. changes (A+B)*C to A*C+B*C might then fail, but IMHO that’s a good thing:  Understand why, fix the test value, and move on.

closed 2016-09-18 by DanielBoudreau