JMRI/JMRI #1439: XML Validation Error - Layout Editor Panel

dsand47, 2016-06-06:
When a Layout Editor panel contains sensor icons that have been converted to text and formatted, the xml validation process reports an error stating that attribute **hasBackground** is not allowed for element **sensoricon** .

Typical line:
`    <sensoricon sensor="S-LS627" x="128" y="320" level="10" forcecontroloff="false" hidden="no" positionable="true" showtooltip="true" editable="false" momentary="false" icon="no" text="27" size="10" style="1" red="255" green="255" blue="0" hasBackground="yes" redBack="0" greenBack="0" blueBack="0" margin="2" justification="centre" class="jmri.jmrit.display.configurexml.SensorIconXml">
`

The layout-config line:
`<layout-config xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://jmri.org/xml/schema/layout-2-9-6.xsd">
`

I am running JMRI 4.3.6.

Dave Sand

bobjacobsen, 2016-06-06:
Thanks for the detail.  Looks like a schema bug, will fix.  To make testing easier, could you attach that panel file (you'll have to give it a .txt extension for Github to take it)  Thanks again.

dsand47, 2016-06-06:
Here is the file.
[MMTRR Working Folder.xml.txt](https://github.com/JMRI/JMRI/files/301070/MMTRR.Working.Folder.xml.txt)

bobjacobsen, 2016-06-07:
PR #1449 should fix this.

commit a0c1105b (2016-06-07): Merge pull request #1449 from bobjacobsen/master

add hasBackground to icon schema Issue closes #1439
closed 2016-06-07 by bobjacobsen