JMRI/JMRI #1441: Removing deprecated code

bobjacobsen, 2016-06-06:
JMRI contains deprecated code, some dating back to 2.*.* releases.  Right after JMRI 4.4 is announced seems a good time to remove some of that.

Some things to consider:
- We don't want people using JMRI for their own projects to get too many surprises.  I've got a current copy of CATS to check against, and it (generally) doesn't seem to be using the really old stuff.  Are there other projects that should be kept in mind?

- Will it break older panel files?  The panel files contain class names, and we have quite a few migration classes still in place to read those:  c.f. [java/src//jmri/configurexml/DccSignalHeadXml.java](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/configurexml/DccSignalHeadXml.java) deprecated in JRMI 2.5.4.  Should we just mark these as kept to keep old (pre-JMRI version 3.*) files?  Or are those files so unlikely to be readable by JMRI 4.* that we should just keep telling people to migrate via e.g. reading and writing them with JMRI 3.something?

- How far forward to go? That's a question of time available, but also about our expectations of how long things should stay available. Is "Everything deprecated for at least 2 years" a valid plan?

rhwood, 2016-06-06:
Assuming that releases are versioned as x.y.z, I think that the removal plan should be depricated prior to version (x - 1).0 can be safely removed, and anything more recent should be retained.

BTW, Java 9 adds an optional forRemoval boolean to the Deprecated annotation to indicate that a deprecated element will/will not be removed in a future version. It would be nice to be able to use that.

bobjacobsen, 2016-06-06:
> Assuming that releases are versioned as x.y.z, I think that the removal plan should be depricated prior to version (x - 1).0 can be safely removed, and anything more recent should be retained.

Just for clarification:  I read that as deprecated-in-2.*.* can be removed after 4.0 (e.g. now), but that 3.*.* cannot.  Is that correct?

JMRI 4.0 was July 13, 2015
JMRI 3.0 was July 25, 2012
JMRI 2.14.1 was July 12, 2012

rhwood, 2016-06-07:
> Just for clarification: I read that as deprecated-in-2.*.* can be removed after 4.0 (e.g. now), but that 3.*.* cannot. Is that correct?

Yes, that is correct. I am proposing the removal of deprecated elements be tied to major releases (with the implicit assumption that changes to X are suggestive of some major internal or underlying change and are an opportune time to rework any user-generated content (e.g. scripts) that may depend on the deprecated element).

We might want to also remove elements introduced and deprecated within a single X release (i.e. introduced in 3.m and deprecated in 3.n), but without the @since JavaDocs, figuring that out could be time consuming.

pabender, 2016-06-07:
Since our practice has been that x.y.z with y even is a production release, I think we actually have two cases to consider:

1) if code is introduced and remove within a single test cycle (y is odd, and both x and y are the same for the version in which the code is introduced and removed) then we just remove the code, no deprecation required.

2) If code is released to the public in an even y production release, then it MUST be deprecated and available for at least two production releases.  In other words, code deprecated in (as an example) 3.5.4 would be eligible for removal in the 3.9.z development sequence (with the intervening 3.6 and 3.8 production releases).  That gives someone external using the code 18 months notice to stop using deprecated code.  (from the release of 3.6 to the release of 3.10).

I think that more or less matches with @rhwood's last statement:

> We might want to also remove elements introduced and deprecated within a single X release (i.e. introduced in 3.m and deprecated in 3.n), but without the @since JavaDocs, figuring that out could be time consuming.

Paul

Paul

KenC57, 2016-06-07:
Question:

 

Is there an easy way to let the remote user know their scripts or other
works are using a depreciated call?? Like logging it somehow so a user would
know. This could be enabled during the last period before removal. My
concern is those on the list and following things would know of what's
expiring. But somebody out there who is doing 'lone wolf' work (and there
does seem to be a number of those in this hobby) won't know until it stops
working when the depreciated code is removed.

 

I'm not sure how possible this is to do. But that's one thing that would
help a lot of the transitions as we go around removing things.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2016-06-07:
> Is there an easy way to let the remote user know their scripts or other works are using a depreciated call??

The Java compiler can tell you if you're invoking deprecated code at compile time. (Eventually, I'd like to get us to the point where we watch that via Jenkins for the JMRI code)

I don't know of a way to do that easily, e.g. automatically, for scripts in Python et al.  We can add logging code to the deprecated methods but that's (1) manual & can't be counted on and (b) will other not be seen by script writers.

bobjacobsen, 2016-06-07:
Are "deprecations within the Java code (and perhaps scripts)" and "deprecations that affect reading the XML files" conceptually different? And, if so, are there additional categories?

For example, consider:

- Deprecations around loading XML files wait until the second major release:  JMRI 4.* doesn't have to load files written by JMRI 2.*, but does need to be able to load files written by any JMRI 3.*  (I'm mostly talking about panel/configuration files; not sure if I'm considering preferences setup files or not)

- Code deprecations that don't affect loading should wait for N years, which in practice is the 2*Nth production release.  (I'd lean toward N=2, two years and 4 production releases, but I'd be happy with other numbers too)

- Code changes with high user impact, e.g. that are used in scripts, should get noticed for a specific time which might be longer than N.  (But if that's too long, you're really just making the problem larger:  More people will be writing new code with the deprecated method than will be migrating away from it.)

It seems to me that most users are more affected by things outside our code, e.g. loading files, than by things inside the code.  Only a small fraction of the community writes Java code with JMRI, and the number writing complicated scripts isn't much bigger.  Hence the idea of having a clear, easy to remember way to tell if a file should be loadable or not.

KenC57, 2016-06-07:
Bob,

 

Good point about scripts/xml vs code. I'd say the shorter period is ok for
code since anyone writing against the code should be using tools able to
point out use of depreciated calls.

 

Reading generated stuff (xml) I could see replacement of the depreciated
code with conversion code so we'd have the option to insure anything written
is using the new code but also to alert the user that they should do a new
save, so it has the new calls.

 

The back of my mind is the scripts. AFAIK, you don't have an easy way to
know you are making calls to depreciated code when you run a script. XML we
can have tools that check any calls for being ok and not depreciated (I
think) so just like the XML check, it could spot calls and see if any are
depreciated. It's how to make it easy to know something should be recoded
that seems where we miss things.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2016-06-07:
Re scripts:  I think the most useful thing there would be better ways to know that we haven't broken the sample scripts, e.g. having tests over them.  They contain most of the common idioms that people use for JMRI script programming.  If we find that deprecating & eventually removing a method would cause any of those to break, we can do various things to cope with that.

KenC57, 2016-06-07:
I would agree that what's in the delivered scripts is a good selection of
calls and I suspect very few calls are used that aren't in one of those
scripts.

 

-ken c

bobjacobsen, 2016-06-07:
There's now a Jenkins job showing deprecation warnings in our code:  Places where we use a deprecated class, method, etc.  See the [summary](http://jmri.tagadab.com/jenkins/job/Development/job/Deprecations/) and [most recent detailed results](http://jmri.tagadab.com/jenkins/job/Development/job/Deprecations/lastBuild/warnings4Result/); click the "Type" tab to see it organized by the deprecated reference.

rhwood, 2016-06-19:
The branch JMRI/JMRI:remove-deprecations removes all classes deprecated in JMRI 2.x except:
- 5 classes whose deprecation was changed to 4.5.1 (see #1533).
- ```jmri.jmrix.loconot.LnSecurityElementManager``` required by the following non-deprecated classes:
  - ```jmri.jmrix.loconet.AspectGenerator``` (no further dependencies within JMRI)
  - ```jmri.jmrix.loconet.configurexml.LnSecurityElementManagerXml``` (no further dependencies with JMRI)
- ```jmri.jmrix.loconet.SecurityElement``` required by the following non-deprecated classes:
  - ```jmri.jmrix.loconet.AspectGenerator``` (no further dependencies within JMRI)
  - ```jmri.jmrix.loconet.configurexml.LnSecurityElementManagerXml``` (no further dependencies with JMRI)

When should ```AspectGenerator``` and LnSecurityElementManagerXml``` be marked deprecated as of?

```jmri.jmrix.loconet.AbstractAlmImplementation```, whose sole subclass has been deprecated since 2.13.5 is not deprecated. When should it be deprecated as of?

bobjacobsen, 2016-06-21:
`jmri.jmrix.loconet.AbstractAlmImplementation` and the various pieces around `jmri.jmrix.loconet.SecurityElement` are the stubs for a not-shipped-with-JMRI tool that configured some CML Electronics boards.  I've sent email to ask the status of that, and to see if I can get a copy of the source code (so I can see if it's even still working with recent JMRI versions).  No reply as yet.  

I think the appropriate thing to do is to deprecate all the non-deprecated parts of this as of 4.5.1.  Will create a PR to do that on the JMRI/JMRI:remove-deprecations branch shortly.

rhwood, 2016-06-21:
The release notes for JMRI 2.10 state that Loco Tools is "removed and no longer supported" as of that version. Should it also be deprecated (it relies on an instance() method deprecated in 2.9.5), or since its removal was announced 5 years ago (and it was removed for Windows users then), should it just be removed?

devel-bobm, 2016-06-21:
> -----Original Message-----
> m3edJ32Aks5qODDngaJpZM4IvJeo> .

Be aware - CML, the business, has changed hands recently, so your e-mail contacts might not be current. - see the www.cmlelectronics.co.uk web site.

bobjacobsen, 2016-06-22:
Now have more info.  `jmri.jmrix.loconet.SecurityElement`, `jmri.jmrix.loconot.LnSecurityElementManager`, `jmri.jmrix.loconet.AspectGenerator` and `jmri.jmrix.loconet.configurexml.LnSecurityElementManagerXml` can go, as there's no longer an external dependency.

But, in other news, `jmri.jmrix.loconet.AbstractAlmImplementation` should stay for at least a little longer, as it may be a useful part of developing support for the new Digitrax DCS240.

I'll put this in a PR for the `JMRI/JMRI:remove-deprecations branch`.

bobjacobsen, 2016-07-03:
Removed 'afterNextProductionRelease' label as JMRI 4.4 is out.

bobjacobsen, 2016-07-07:
#1624 is related to this.

closed 2016-07-09 by bobjacobsen