JMRI/JMRI #1472: Need User Libraries folder

rhwood, 2016-06-10:
A user asked in a phone call today if we could add a separate directory for user contributed JARs from the program installation directory.

I think this would be a good idea, and suggest that we place this in the portable path ```preferences:lib``` (```C:\Users\<user>\JMRI\lib on Windows```, ```~/.jmri/lib on Linux```, and ```~/Library/Preferences/JMRI/lib``` on OS X by default).

I am comfortable writing the *nix and OS X launchers to do this, but would need assistance doing this on Windows.

There are some questions that should be answered first:
- Should we really do this?
- Should user-provided JARs be before or after JMRI shipped JARs in the class path (should user-provided JARs be able to override JMRI content, or merely augment JMRI content)?

mattharris, 2016-06-10:
_If_ we choose to do this, I would suggest that user-provided JARs take precedence over JMRI shipped JARs, otherwise there could be confusion.

Now we could add a command-line switch to influence this behaviour.

mattharris, 2016-06-10:
Oh, and I'm happy to work on the Windows launcher for this.

bobjacobsen, 2016-06-10:
I think this is a good idea, especially on Mac and Linux where installation is replacing a directory. 

I think in-front-of-shipped is more powerful, but more  dangerous, than behind-shipped. If we're adding a capability, we should make it powerful/flexible. 

It would be good to have the contents of this e.g. added to the context report so people helping debug can see it. (Or is the full classpath already there?)

Maybe add an INFO level message at startup, too? Don't like those for the majority of users who run vanilla configs, but perhaps we could show the local part of the classpath when it exists.

mattharris, 2016-06-10:
Complete classpath [would appear to be reported](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/mailreport/ReportContext.java#L109)

Possibly a good idea to add an `INFO` message, but not entirely sure _how_ we would do that, other than by parsing the classpath each time.

rhwood, 2016-06-10:
Actually, looking at how the NetBeans Platform (not the IDE, but the development harness) does this, they take two arguments when running the application:
```
  --cp:p <classpath>    prepend <classpath> to classpath
  --cp:a <classpath>    append <classpath> to classpath
```

This leads me to think we want to use those command line options and to allow them to be set in a configuration file (```preferences:jmri.conf``` in our portable scheme?). This will *require* that users explicitly list their jar files in the order they are to be prepended or appended to the standard JMRI classpath, which I think is a significant benefit for troubleshooting this.

pabender, 2016-06-10:
On 06/10/2016 12:39 PM, Bob Jacobsen wrote:
> It would be good to have the contents of this e.g. added to the 
> context report so people helping debug can see it. (Or is the full 
> classpath already there?)
I'm with Bob on this.

We've already seen a case this year where someone installed a newer 
version of a library we use (pi4j) because the new version does 
something the old one doesn't, and then complained that JMRI didn't 
work... it turned out the API changed in the version, and our code won't 
work with it, but the new version is only available as a snapshot, and 
hasn't had an official release yet (I looked two days ago...).

Paul

rhwood, 2016-06-10:
Let me further clarify the alternate proposal:

- The JMRI launchers will parse a Java properties file at ```preferences:jmri.conf``` that contains at least the property ```default_options``` (it may be empty, and if nonexistent, should be treated as empty). All properties will be surrounded by quotes (so the file can also be parsed by a POSIX shell script).
- ```default_options``` will contain command line arguments as if written on the command line and command line arguments will be processed after default_options.
- In addition to existing options, the options ```--cp:p <classpath>``` and ```--cp:a <classpath>``` will be used to prepend or append elements to the classpath and can be accepted multiple times. These options can be specified on the command line or in the ```default_options``` property of ```jmri.conf```.

(This is how the NetBeans Platform Harness handles this problem; I am suggesting this particular method because although its been on hold, I am again working on porting JMRI to the NetBeans Platform.)

bobjacobsen, 2016-06-10:
I think building in an options mechanism, and using it for this, sounds good.  Can we make the same thing for all the various ways we launch?  The launchers we distribute for Mac OS X / Linux / Windows; from Ant; from NetBeans and Eclipse (I guess that last one is OK).

rhwood, 2016-06-10:
- The launchers for OS X/Linux are not an issue for me to do.
- The Windows launcher appears to already have much of this in place (I can't really confirm, but looking at its code, it appears to).
- We may need to handle ant/NetBeans (NetBeans relies on ant) differently—ant breaks JVM arguments, classpath, and command line arguments into separate elements, so we might want to stick with a different, but similar mechanism to the OS X/Linux/Windows launchers that resides in ```local.properties``` (we might also want to ship with an ignored, but copyable jmri.conf file in program: as a starting point for users creating their own).
- As far as I understand, there is no configured command in Eclipse to run JMRI from within the IDE, and ant is relied upon for that.

DanielBoudreau, 2016-06-11:
Randall,

 

I’m not completely sure I understand what you mean by this: “As far as I understand, there is no configured command in Eclipse to run JMRI from within the IDE, and ant is relied upon for that”.

 

I launch PanelPro within Eclipse by using “Run Java Application” command and selecting “apps.PanelPro.PanelPro”.  Not sure if that meets the definition of a “configured command”, but I don’t think it uses ant.

 

Dan

rhwood, 2016-06-11:
According to [Running Ant buildfiles](http://help.eclipse.org/mars/topic/org.eclipse.platform.doc.user/tasks/tasks-ant-running.htm), ant targets (or external tools) can be setup and persisted, and these appear to be the only way to run a Java program from within eclipse that can add JARs not in JMRI to the execution environment without the risk of modifying a file under revision control (like .classpath). Although Eclipse will happily run JMRI without ant, neither [Launching a Java program](http://help.eclipse.org/mars/topic/org.eclipse.jdt.doc.user/tasks/task-launching_java_program.htm) or [Setting execution arguments](http://help.eclipse.org/mars/topic/org.eclipse.jdt.doc.user/tasks/tasks-executionArgs.htm) suggest we can provide a mechanism native to Eclipse for configuring the execution environment in Eclipse as configurable or persistent as ant or another external tool.

Randall

DanielBoudreau, 2016-06-11:
Can one of these two Eclipse run configuration tabs do what we need?

 

http://help.eclipse.org/mars/index.jsp?topic=%2Forg.eclipse.jdt.doc.user%2Ftasks%2Ftasks-java-local-configuration.htm

 

 

*	The Classpath tab defines the location of class files used when running or debugging an application. By default, the user and bootstrap class locations are derived from the associated project's build path. You may override these settings here.
*	The Source tab defines the location of source files used to display source when debugging a Java application. By default, these settings are derived from the associated project's build path. You may override these settings here.

rhwood, 2016-06-11:
I've updated the title and description of this issue to reflect the course of discussion.

rhwood, 2016-06-11:
Although we can setup custom launchers within Eclipse that have the effect of allowing an Eclipse user to add custom parameters and classpaths to the launched process, Eclipse provides no way for us to store, under revision control, launchers that can take parameters from a file stored outside revision control.

Eclipse Mars does, however, include an internal ant implementation, and launchers for ant builds using the standard build.xml can be stored under revision control, but it appears that Eclipse users will have to manually add this to their favorites to retain this in menus. (Eclipse will use maven as a "native", not "external" build tool, which I think means that if we switched to maven from ant, Eclipse would "just work" the same way NetBeans and IntelliJ do, but that's outside the scope of this issue.)

rhwood, 2016-07-02:
Other than the Windows launcher, this is done.

rhwood, 2016-11-02:
@mattharris Will you be able to do this for the Windows launcher?

mattharris, 2016-11-02:
@rhwood Yes, I'll take a look.

bobjacobsen, 2016-11-28:
Once this is in place on all three platforms, it might be helpful to add a mention to

http://jmri.sourceforge.net/help/en/html/doc/Technical/plugins.shtml

That page has fallen a bit behind in general, but I don't know enough about the new extension mechanisms to do a good job fleshing it out more. Or are those already documented somewhere else? If so, maybe we should remove that "plugin" discussion here.

mattharris, 2016-11-28:
Got in-place the `--cp:a=[CLASSPATH]` and `--cp:p=[CLASSPATH]` option processing.

Still working on the `%userprofile%\JMRI\jmri.conf` parsing.

mattharris, 2016-11-28:
#2610 should cover these

closed 2016-11-29 by rhwood