JMRI/JMRI #1473: Redo libUSB support

bobjacobsen, 2016-06-10:
JMRI uses version 0.5.7 of libusb, including associated .jar, .dll, .so, .jnilib files.

This is old, and doesn't work so well. The JMRI code just pops a demo & navigation window; most of the JMRI connection is actually making sure that the lib is available for a very few scripting uses, e.g. for the RailDriver driver.

As part of the effort to get the ReDAC locomotive controller working, this needs to be updated to a more recent version.  This _will_ break some existing user scripts: The APIs changed from libusb 0.5.7 to now.  But there seems to be no way to have both present and available in JMRI. I don't see a way to do any kind of multi-year migration/deprecation.

I propose changing it early (as early as possible) in the JMRI 4.5.*/4.6 cycle so that people have some time to modify any local scripts before RMI 4.6.

bobjacobsen, 2016-07-03:
Removed 'afterNextProductionRelease' label as JMRI 4.4 is out.

rhwood, 2016-08-20:
Are you considering the http://usb4java.org libusb implementation?

rhwood, 2016-08-20:
Note that IIF the Windows JMRI launcher gets classpath modification support (#1472), any new libusb API could coexist with the existing API if a user wanted to "re-add" the older API support.

commit f639a589 (2017-07-15): Merge pull request #3679 from rhwood/master

New USB library.
Fixes #1473
closed 2017-07-15 by rhwood