JMRI/JMRI #1478: Create Release 4.3.8

mattharris, 2016-06-11:
This is the next test release, building on 4.3.7 (not from HEAD of master). It will form the basis of the Production Release towards the end of the month.

Only minor changes and bug fixes to be considered.

rhwood, 2016-06-13:
Are the bug fixes in #1489 too large a change for 4.3.8?

mattharris, 2016-06-13:
I would suggest that this is OK to consider for 4.3.8 - releasing 4.4 without this _could_ cause us some issues.

mattharris, 2016-06-17:
4.3.8 is scheduled for this weekend - do we still want to go with this or postpone a bit?

bobjacobsen, 2016-06-17:
> 4.3.8 is scheduled for this weekend - do we still want to go with this or postpone a bit?

I’m in favor of sticking to the 4.3.8/4.3.9/4.4 schedule. 

It’s true that there are additional changes that could go into that sequence so they’ll be in 4.4.  But that’s _always_ true.  A quick & by-the-book sequence gets 4.4 out the door while letting us get back to normal test releases. That in turn helps keep alive some of the other things that people are working on.

rhwood, 2016-06-17:
I'm also in favor of the weekend.

That said I have two questions:
- Is #1514 to large to pull back into 4.3.8 or 4.3.9?
- When we write PRs against the 4.3.8 branch, can we merge them ourselves, or should we wait for the releaser to review and merge them?

mattharris, 2016-06-18:
I've started on this (the eagle-eyed amongst you will have noticed that the PRs against `release-4.3.8-suggested-patches` have been merged and will soon be merged into `release-4.3.8`).

I'll be working further on this later today due to family commitments.

If there are any other items for consideration in 4.3.8, please submit PRs against `release-4.3.8-suggested-patches`.

Thanks.

bobjacobsen, 2016-06-18:
When will it be OK to start merging "afterNextProductionRelease" items into master?  

The plan for 4.4 is late June (notionally June 26), and 4.5.1 would be a couple weeks later (notionally July 8)   

Also, do we want to move 4.5.1 up a bit, so that it's created before the NMRA meeting (which is quite early this year: July 3 to 10)?

If 4.5.1 stays where it is, there's time to merge major updates (deprecations, etc) between them. We can hold those until then, but some of the larger ones are starting to show conflicts even now.  On the other hand, we don't want to make it hard to include fixes in 4.3.9 (and most people, including myself, find it hard to start developing things like that against an older branch point than the current master).

mattharris, 2016-06-19:
Now released. @bobjacobsen can I have a think about this and get back to you later?

bobjacobsen, 2016-06-19:
> can I have a think about this and get back to you later?

Absolutely!  You're doing the release work, you decide how best to do it.

mattharris, 2016-06-22:
@bobjacobsen As mentioned in https://github.com/JMRI/JMRI/pull/1555#issuecomment-227849672, I think that we can go-ahead with commencing the merge of `afterNextProductionRelease` items into `master`.

For management of fixes to 4.3.x/4.4 series, we will need to do this starting from the `release-4.3.9-suggested-patches` branch right now as opposed to `master`, merging those across into `master`.