JMRI/JMRI #1485: Reduce warning verbosity when loading roster entries with ESU decoders

bobjacobsen, 2016-06-12:
From @dheap:
````
The log.warn() at line 233 of java/src/jmri/jmrit/roster/LocoFile.java from this commit:
<https://github.com/JMRI/JMRI/pull/1378/commits/b0863742efa26a4b8804ddb72ba54a4b50081ee8> causes 5,120 warning messages to be displayed in the System Console every time a user opens a roster entry based on an ESU V4.0 or Select definition.

The reason for the warnings is that just over two years ago commit <https://github.com/JMRI/JMRI/commit/82255767c8b3be48bb5e08024bed5b2ea2ae0681> moved the creation of the 5,120 single-bit Function Map variables (spread over 640 CVs) from XML to code (java/src/jmri/jmrit/symbolicprog/FnMapPanelESU.java) thereby eliminating 20,000 lines of XML code in the decoder definition and reducing definition file load time from an unacceptable several minutes to a still-slowish 10 seconds or so.

(FnMapPanelESU.java also takes care of loading the variable with the saved value from any stored CV value as each variable is created.)
````

The ESU variables are created here:
````
java/src//jmri/jmrit/symbolicprog/FnMapPanelESU.java:                        String name = "ESU Function Row " + Integer.toString(iRow + 1) + " Item " + Integer.toString(item + 1);
````

Design issues:
 - The warning messages comes from jmri.jmrit.roster.LocoFile; it's not wise to couple jmrit.roster and jmrit.symbolicprog through testing for specific text.  
- That text isn't (yet) internationalized.
- The warning comes from text in a file _not_ present as a class in the current VariableTableModel, so we can't defer that to the VariableTableModel contents per se.

My first thought it to provide plug-in handling for this warning so that the jmri.symbolicprog invoker can provide a lambda to filter these.  That'll have to be aware of the special ESU subclasses, but at least that's localized to a single subclass.

rhwood, 2016-06-12:
Edited description so links work.

dheap, 2016-06-12:
> @bobjacobsen
> That text isn't (yet) internationalized.

We don't internationalise the equivalent (and more complex) " controls output " variable names in FnMapPanel.java either, nor would I see it as appropriate to do so.

See http://jmri.sourceforge.net/help/en/html/apps/DecoderPro/FnMapping.shtml

dheap, 2016-06-12:
- In both the FnMapPanel and the FnMapPanelESU cases, the text is **never** seen by the user.
- In the FnMapPanel case, the text is **only** seen by the the definition writer.
- In the FnMapPanelESU case, the text is **never** seen, except in the roster entry XML file.

> @dheap
> 
>> @bobjacobsen
>> That text isn't (yet) internationalized.
> 
> We don't internationalise the equivalent (and more complex) " controls output " variable names in FnMapPanel.java either, nor would I see it as appropriate to do so.

PR #1631 (2016-07-09): Reduce warning verbosity when loading locomotive file with ESU V4 decoders
Fixes #1485
closed 2016-07-09 by bobjacobsen