JMRI/JMRI #1540: Create Release 4.3.9

mattharris, 2016-06-19:


mattharris, 2016-06-24:
Will look to start the builds for this later today.

mattharris, 2016-06-25:
I'm a little bit behind schedule for this, but first candidate builds should be expected within about 1 hour or so

mattharris, 2016-06-25:
First files for JMRI 4.3.9 are available:

http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.3.9/

Feedback appreciated! I'd like to release these later today.

Note: this should be the final release of this development series -
unless any showstopping issues are experienced with this, this will
become production release 4.4.

silverailscolo, 2016-06-25:
Test installing latest build on macOS / OS X 10.11.5 succesful. However, came across an error opening the LRoute table, see PR #1583

mattharris, 2016-06-25:
Build 3 will contain the fix in #1583

DanielBoudreau, 2016-06-25:
Downloaded and installed OK on win7.  Loaded panel using NCE simulator, tested turnouts and routes, all good.

 

Dan

 

http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.3.9/

AlanUS, 2016-06-26:
Build 3. Installed and tested OK in Win7 and Win10, with Sprog IIv3.
Identified loco, read decoder, run loco.

mattharris, 2016-06-26:
Thanks for the testing.

Closing this as this is now release.

closed 2016-06-26 by mattharris