JMRI/JMRI #1550: LocoNet Monitor throwing NPEs

rhwood, 2016-06-22:
Helping a user debug a script, we are getting this error:

```
2016-06-21 20:50:44,830 LnScanner                             INFO  - msg is B2 1E 6F 3C [AWT-EventQueue-0]
2016-06-21 20:50:44,831 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrix.loconet.locomon.Llnmon.format(Llnmon.java:590)
	at jmri.jmrix.loconet.locomon.Llnmon.displayMessage(Llnmon.java:3357)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:186)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:204)
	at org.python.core.PyObject.__call__(PyObject.java:478)
	at org.python.core.PyObject.__call__(PyObject.java:482)
	at org.python.core.PyMethod.__call__(PyMethod.java:141)
	at org.python.pycode._pyx7.displayMsg$22(<script>:690)
	at org.python.pycode._pyx7.call_function(<script>)
	at org.python.core.PyTableCode.call(PyTableCode.java:167)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:138)
	at org.python.core.PyFunction.__call__(PyFunction.java:413)
	at org.python.pycode._pyx7.message$21(<script>:679)
	at org.python.pycode._pyx7.call_function(<script>)
	at org.python.core.PyTableCode.call(PyTableCode.java:167)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:307)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:198)
	at org.python.core.PyFunction.__call__(PyFunction.java:482)
	at org.python.core.PyMethod.instancemethod___call__(PyMethod.java:237)
	at org.python.core.PyMethod.__call__(PyMethod.java:228)
	at org.python.core.PyMethod.__call__(PyMethod.java:218)
	at org.python.core.PyMethod.__call__(PyMethod.java:213)
	at org.python.core.PyObject._jcallexc(PyObject.java:3626)
	at org.python.core.PyObject._jcall(PyObject.java:3658)
	at org.python.proxies.__builtin__$MsgListener$28.message(Unknown Source)
	at jmri.jmrix.loconet.LnTrafficController.notify(LnTrafficController.java:96)
	at jmri.jmrix.loconet.LnPacketizer$RcvHandler$1.run(LnPacketizer.java:435)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
```

The script is attached.
[LnScanner38rhwX1.py.txt](https://github.com/JMRI/JMRI/files/326846/LnScanner38rhwX1.py.txt)

rhwood, 2016-06-22:
Changing the method displayMsg in the attached Jython script to force the Llnmon sensorManager to sensors prevents the NPE:
```
def displayMsg(msg):
    log.info("sensors is {}", sensors)
    log.info("msg is {}", msg)
    parseMsg.setLocoNetSensorManager(sensors)
    if timeCheckBox.isSelected():
        scrollArea.setText(scrollArea.getText() + time.strftime('%X:') + " ")
    if rawCheckBox.isSelected():
        scrollArea.setText(scrollArea.getText() + "[" + msg.encode("utf-8") + "] ")
    scrollArea.setText(scrollArea.getText() + parseMsg.displayMessage(msg)) #displayMessage adds a carriage return
    return
```

I can't find any reason the sensorManager should be null at any point in this setup:
- JMRI 4.3.7
- Windows 10
- Single connection: LocoNet Simulator

devel-bobm, 2016-06-22:
> -----Original Message-----
> TMiONI9ks5qOJB9gaJpZM4I7UHl> .

The user's jython code instantiates an llnmon with the line:

parseMsg = jmri.jmrix.loconet.locomon.Llnmon()

, but the jython does not initialize the local sensorManager via a subsequent line like:

parseMsg.setLocoNetSensorManager( aLocoNetSensorManagerObject)

The llnmon itself can automagically figure out which sensor manager to use.  The jython code will have to figure that out.  

LocoMonPane.java finds the sensor manager, starting from the connection memo, as shown here:

        if (memo.provides(jmri.SensorManager.class)) {
            llnmon.setLocoNetSensorManager((jmri.SensorManager) memo.get(jmri.SensorManager.class));
        }

Note that LocoMonPane.java gets the connection memo passed in when the LocoMonPane is created by the menuing system.

As a "hack", the python could do the same thing that "ThrottleBridge.py" does - it just gets the first LocoNet connection memo:

lnMemo=jmri.InstanceManager.getList(jmri.jmrix.loconet.LocoNetSystemConnectionMemo).get(0);

Hope that helps!

devel-bobm, 2016-06-22:
Note that the user's python script will cause similar `Llnmon()` NPEs for LocoNet messages with "turnout" information and LocoNet messages with "reporter" information, because 'Llnmon()' also has local storage of the connection-specific turnout manager and reporter manager.  The script does not initialize either of these.

`Llnmon()` has methods for setting which connection-specific managers will be used for turnouts and reporters.  These methods must be invoked before Llnmon attempts to interpret any Turnout or Reporter "user names".  They are:
`public void setLocoNetTurnoutManager(jmri.TurnoutManager loconetTurnoutManager)`
and
`public void setLocoNetReporterManager(jmri.ReporterManager loconetReporterManager)`

These two methods ought to be invoked immediately after creation of the instance of `Llnmon`, as should the method for setting the sensor manager, as noted in the previous comment.

rhwood, 2016-06-23:
If no messages are received before the user checks any of the bottom checkboxes in the script, the llnmon works as expected without calling ```setLocoNetSensorManager(sensors)```, but checking any of those checkboxes does not trigger any discoverable path that sets the manager for the monitor.

Note also that setting the managers to proxy managers instead of loconet managers also works.

However, this is being closed, since the workaround is in use.

commit 72185415 (2016-10-29): Merge pull request #2437 from rhwood/cleanup

Multiple fixes to Llnmon
Fixes #1556 
Fixes #1550
closed 2016-06-23 by rhwood