JMRI/JMRI #1556: locomon.Llnmon() clarification

devel-bobm, 2016-06-22:
When an `JMRI/java/src/jmri/jmrix/loconet/locomon/Llnmon.java` is created, proper operation requires invocation of `setLocoNetSensorManager()` , `setLocoNetTurnoutManager()` , and `setLocoNetReporterManager()`.

These three methods set local storage of the associated managers, for use when `Llnmon.displayMessage` is invoked and the LocoNet message includes a sensor, turnout, or reporter number which will be used to find a JMRI user name via the manager.

These post-creation initializtion steps are awkward, and have led to at least one reported problem in user-created jython code (reference: Issue #1550 ).

Failure to invoke these methods result in an NPE when `Llnmon.displayMessage` is asked to interpret a LocoNet message which passes a sensor, turnout, or reporter number.

For the purpose of code clean-up, and avoiding jython-related user issues similar to the one noted above, I recommend: 

- Deprecate the existing `Llnmon` constructor method which takes no parameters.
- Create a new `Llnmon` constructor which takes as arguments the sensor, turnout, and reporter managers.  Handle the operations found in the existing "setter" methods as part of the constructor.
- Deprecate the three public methods (listed above) for setting the managers
- Update existing JMRI code to use the new constructor: `locomon.LocoMonPane`, `locomon.LocoMonFrame', etc.

rhwood, 2016-06-23:
These post-creation steps only seem to be needed in some circumstances, and it's not at all clear what they are. Failure to assign the managers *sometimes* leads to failures when a sensor is passed, but not always, and using a ProxySensorManager works (at least for systems with only a single LocoNet connection), so its not clear that the default no argument constructor should not continue to be usable by having it set the managers to the proxy managers and allow that to be overridden later if desired.

devel-bobm, 2016-06-23:
It is incorrect to state that the current `Llnmon.java` implementation's no-argument constructor sets the managers.  It does not.  The current implementation relies on the setter methods being inoked by the code which instantiates the `Llnmon` object.

The only place where it might make sense for `Llnmon` to use managers from a "default" connection would be to support jython scripts, and this would only be useful for the case where the assumption of only a single LocoNet connection applies.

rhwood, 2016-06-23:
> It is incorrect to state that the current `Llnmon.java` implementation's no-argument constructor sets the managers.

Correct or not, the observed behavior is as if the manager has been correctly set when it hasn’t when the sensor is also handled by other other processes.

devel-bobm, 2016-06-23:
I'm missing something, somewhere.

The statement "the observed behavior is as if the manager has been correctly set when it hasn’t when the sensor is also handled by other other processes." baffles me.  How can a java object, instantiated from within a jython process, have its private fields magically set to useful values?  What mechanism does this?

And if the magic really works, I don't understand why the user problem happened in the first place.

rhwood, 2016-06-23:
> On 23 Jun 2016, at 12:56, Bob Milhaupt <notifications@github.com> wrote:
> 
> I'm missing something, somewhere.
> 
> The statement "the observed behavior is as if the manager has been correctly set when it hasn’t when the sensor is also handled by other other processes." baffles me. How can a java object, instantiated from within a jython process, have its private fields magically set to useful values? What mechanism does this?
> 
> And if the magic really works, I don't understand why the user problem happened in the first place.

In the original script, it magically works if and only if either the speak or show sensor checkbox in the script have been clicked before a sensor message is seen. However, we can find no path that sets those managers in the llnmon object in the script as attached to the original bug.

Randall Wood
Alexandria Software
202.683.8604
randall.h.wood@alexandriasoftware.com
http://www.alexandriasoftware.com

devel-bobm, 2016-06-23:
If the debugging which shows this checkbox-related behavior has been done within a single JMRI "instance", using a sequence of executions of different versions of the jython script, it is possible that the debugging is hitting a "feature" of our jython environment (or maybe the underlying jython/python libraries), where one version of a given script do not necessarily replace the code of a previous versoin used within that single "instance" of JMRI.

I have experienced such behavior while debugging some jython scripts, and found that the only reliable way to ensure that my changes took effect was to exit JMRI then re-start, before executing a changed jython script.

If both you and the user are satisfied with the solution, I'll let go now!

rhwood, 2016-06-23:
> it is possible that the debugging is hitting a "feature" of our jython environment

It really is a feature of our scripting environment: See "Can I communicate between scripts?" in the [Scripting FAQ](http://jmri.org/help/en/html/tools/scripting/FAQ.shtml)

commit 72185415 (2016-10-29): Merge pull request #2437 from rhwood/cleanup

Multiple fixes to Llnmon
Fixes #1556 
Fixes #1550
closed 2016-10-29 by rhwood