JMRI/JMRI #1576: Catch JavaDoc warnings in PRs

rhwood, 2016-06-24:
It would be nice to be able to catch JavaDoc warnings in Pull Requests instead of running an after-the-fact test in [Jenkins](http://jmri.tagadab.com/jenkins/job/Development/job/Builds/).

Unfortunately this requires Ant 1.9.4 or newer.

rhwood, 2016-06-24:
One way of fixing the old ant issue would be to wrap ant in maven, and have maven bootstrap and cache ant and then use ant calls for its build and test phases.

bobjacobsen, 2016-08-08:
We've set the `-missing` option in `-Xdoclint` so that we don't currently require those 'missing' @param, @throws, @return tags. I'm not sure whether we're going to require those in the near future or not. There was some discussion (on jmri-developers?) about the utility of those in cases where the obvious and non-obvious.  Note also that the doclet format has changed over the last few versions, e.g. [this example](http://jmri.org/JavaDoc/doc/jmri/Sensor.html#getSensorDebounceGoingActiveTimer--) so that missing tags are no longer being automatically generated.   

In any case, can we implement this test with the `-missing` option so that it detects all the other things that can go wrong?

rhwood, 2016-08-08:
> In any case, can we implement this test with the -missing option so that it detects all the other things that can go wrong?

It's currently implemented with the ```-missing``` option, so I'm not sure what's being asked. If implemented without the ```-missing``` option, there are currently about 5577 warnings thrown, so any other warning would likely be lost in the noise.

bobjacobsen, 2016-08-12:
> It's currently implemented with the -missing option, so I'm not sure what's being asked.

Sorry, I guess I misunderstood the un-checked checklist item at the top.  Thanks for your work to make the JavaDoc check happen.

Can this be closed then?

rhwood, 2016-08-12:
I've been running a local copy of build.xml with the ```-missing``` option removed to get those numbers in the summary. I've been slowly chipping away at this (we are below 5500 missing JavaDoc elements now).

rhwood, 2016-10-06:
Progress to date: down to 5218 missing tags.

rhwood, 2016-11-25:
As of #2584, the code is down to 4699 missing tags.

rhwood, 2017-03-09:
As of #3210, the code is down to 3197 missing tags.

rhwood, 2017-04-07:
As of #3339, we are down to 2750 missing tags (more than 1/2 way from when we started). Thank you to all who have been working on this.

rhwood, 2018-03-16:
OP is no longer concerned.

closed 2019-01-13 by bobjacobsen