JMRI/JMRI #1590: Convert XBee Support to Digi's Official Java Library?

pabender, 2016-06-28:
I just wanted to see if anyone had any opinions as to whether we should convert the JMRI XBee Support to use Digi's official library (  https://github.com/digidotcom/XBeeJavaLibrary )

The current support was written to use Andrew Rapp's XBee-API library ( https://github.com/andrewrapp/xbee-api )

The conversion is going to be non-trivial because the APIs are significantly different, but the Digi library might be more well supported in the long term.

I haven't looked through the features of the Digi library completely yet, but I know this is the basis of the current generation of cross-platform configuration tools.

Paul

pabender, 2016-09-07:
@KenC57 

When you get a chance, can you try pulling down a copy of my DigiXBeeLibrary branch?
( https://github.com/pabender/JMRI/tree/DigiXBeeLibrary )

I have the code compiling with the Digi XBee Library, and at least node discovery and sensors are working again. ( and as of right now, loading devices from the file and then running node discovery (which runs automatically at start up) results in duplicate nodes. )

Paul

pabender, 2016-09-07:
I'll also add it's really nice not having to think about the type of XBee connected.  The Library itself handles the node type detection.

Paul

KenC57, 2016-09-07:
I'll give it a try, maybe this weekend.

 

One quick question about the modules:

 

I've been using those little cards that convert the 3.3 to 5v and give the
pin spacing that fits on a breadboard. What voltage range are they safe
for?? I'm looking to use batteries for testing and a 6V is one option but
open for other ideas. Long term I'm hoping for solar powered with some sort
of battery. I'd either use an off switch or some sort of variable sleep idea
in the end.

 

I've tried with a pair of AA batteries and the radios work for a while but
drop off once the batteries drop below 3. But how much over 5 can that
adapter handle??

 

-ken c

pabender, 2016-09-08:
On Sep 7, 2016, at 4:48 PM, Ken Cameron <notifications@github.com> wrote:

> I'll give it a try, maybe this weekend.

Thanks, by then I may have most things working.

I think i have the Turnouts working again.  I haven't tried lights yet.

I'm trying to figure out a strange problem with the IO stream, it works fine with my SPROG but the connection dies quickly with my RFID reader.

> One quick question about the modules:
>
> I've been using those little cards that convert the 3.3 to 5v and give the
> pin spacing that fits on a breadboard. What voltage range are they safe
> for?? I'm [...]

The XBees are 3.3v modules.  I know many of the sparkfun adapters have 3.3V regulators included in the circuit.  When I use the Sparkfun adapters, I put a 5V voltage regulator in front of them. ( and lately I have been buying the Sparkfun Arduino shields as adapters, with a bridge rectifier and a 5V regulator on the breadboard area). Most of mine are powered by a DCC accessory bus. )

The adafruit adapters have a 5V LM7805 regular included.  

> I've tried with a pair of AA batteries and the radios work for a while but
> drop off once the batteries drop below 3. But how much over 5 can that
> adapter handle??

The little 3.7v lithium ion batteries work pretty well with XBees and come in many different mAh sizes.

I have one in a control panel along with an Arduino Pro-mini and one of these: https://www.sparkfun.com/products/11231
The 2000mAh battery in that lasts 4 to 5 hours. ( without any node sleeping )

Paul

pabender, 2016-09-08:
On Sep 7, 2016, at 8:05 PM, Paul Bender <paul.bender@acm.org> wrote:
> The adafruit adapters have a 5V LM7805 regular included.  

Thinking about this more, that is actually a LM7803, the 3.3V regulator from the same family.

Paul

pabender, 2016-09-08:
@KenC57,

I think I have this working more or less correctly now.

A couple of notes:
1) If the node discovery process goes correctly, and all the identification fields (16 bit address, 64 bit address, and node identifier) all update, then the Node tool works as expected.
2) I haven't fixed the node tool as far as node entry is concerned.
3) Neither the monitor or the send packet tool are expected to work at the moment.  Because of the way the Digi XBee Library works, we don't deal much with messages directly.  
4) I've actually stripped out most of the code that allowed us to use the AbstractMRTrafficController because the Library handles that for us.  Except for in the XBeeIOStream,  we use listener callbacks to get data after sending something to a node.

There's still some clean up left to do, and some classes (like the XBeeMessage and XBeeReply classes) may be eliminated entirely.

(needless to say, there wasn't a way for me to convert to the new library and fully retain the old API...)

KenC57, 2016-09-09:
I made a new workspace and pulled it down. But it is complaining about the
Xbee library being missing. Shouldn't that have been found in the
repository??

 

-ken c

DanielBoudreau, 2016-09-09:
I see xbee-api-0.9.jar in my directories.

Dan

rhwood, 2016-09-09:
Paul does not use Eclipse, so it's likely ant builds this correctly, but the jar is missing from the .classpath file. I know I'm guilty of the same omission myself.

Randall Wood
From my phone

KenC57, 2016-09-09:
Paul,

 

Looking in the directory I don't see a xbee-api-0.9.jar file. It is listed
in the paths, but no file found. It is in a few other workspaces I've got so
I'll just copy it. But figured it worth noting it didn't come over.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

KenC57, 2016-09-09:
Paul,

 

After copying that file into the workspace, it builds. But when I try
running I get an error of a missing import:

 

import com.digi.xbee.api.connection.IConnectionInterface;

 

that's from XBeeAdapter.java line 3. So it seems something else wasn't
coming across from the repository.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

pabender, 2016-09-09:
On Fri, Sep 9, 2016 at 11:14 AM, Ken Cameron <notifications@github.com>
wrote:

> Looking in the directory I don't see a xbee-api-0.9.jar file. It is listed
> in the paths, but no file found. It is in a few other workspaces I've got
> so
> I'll just copy it. But figured it worth noting it didn't come over.
>

xbee-api-0.9.jar is the old library.

The new library is in the file xbjlib-1.1.1.jar

Yes, this is in the branch.

Paul

pabender, 2016-09-09:
On Fri, Sep 9, 2016 at 10:11 AM, Randall Wood <notifications@github.com>
wrote:

> Paul does not use Eclipse, so it's likely ant builds this correctly, but
> the jar is missing from the .classpath file. I know I'm guilty of the same
> omission myself.
>

Yes, ant builds this correctly, and ant is our official build tool.

I also thought our eclipse instructions had users setup the builds to use
the ant builds.  I know eclipse can do that.
(and also some processes like the JavaCC compiler behave better using ant
than building with the eclipse plugin...)

Paul

DanielBoudreau, 2016-09-09:
The most likely issue for Ken is that the .classpath file hasn’t been updated to use the new jar file.

Dan

KenC57, 2016-09-09:
Bingo, what I pulled down did have the new file, but still had the old one
referenced. Changed that and let it rebuild. Tried it with my prior setup,
here is the console:

 

2016-09-09 13:22:17,619 util.Log4JUtil                        INFO  - ******
JMRI log ******* [main]

2016-09-09 13:22:17,641 util.Log4JUtil                        INFO  - This
log is appended to file: C:\Users\kcameron\JMRI\log\messages.log [main]

2016-09-09 13:22:17,642 util.Log4JUtil                        INFO  - This
log is stored in file: C:\Users\kcameron\JMRI\log\session.log [main]

2016-09-09 13:22:17,648 apps.Apps                             INFO  -
PanelPro version 4.5.4ish-201609091519-kcameron starts under Java 1.8.0_101
on Windows 10 amd64 v10.0 at Fri Sep 09 13:22:17 EDT 2016 [main]

2016-09-09 13:22:21,522 apps.Apps                             INFO  -
Starting with profile XbeeDmName.3f6353a5 [main]

2016-09-09 13:22:21,761 node.NodeIdentity                     INFO  - Using
jmri-84A6C8D92D62-3f6353a5 as the JMRI Node identity [main]

2016-09-09 13:22:22,056 xbee.XBeeAdapter                      INFO  - COM5
port opened at 9600 baud with DTR: false RTS: false DSR: false CTS: true
CD: false [main]

2016-09-09 13:22:22,079 api.XBeeDevice                        INFO  -
jmri.jmrix.ieee802154.xbee.XBeeAdapter@517566bOpening the connection
interface... [main]

2016-09-09 13:22:22,079 api.XBeeDevice                        INFO  -
jmri.jmrix.ieee802154.xbee.XBeeAdapter@517566bConnection interface open.
[main]

2016-09-09 13:22:22,121 connection.DataReader                 ERROR - Error
parsing the API packet. [Thread-7]

com.digi.xbee.api.exceptions.InvalidPacketException: Invalid checksum
(expected 0x9D).

               at
com.digi.xbee.api.packet.XBeePacketParser.parsePacket(XBeePacketParser.java:
192)

               at
com.digi.xbee.api.connection.DataReader.run(DataReader.java:409)

2016-09-09 13:22:22,482 api.XBeeDevice                        INFO  -
jmri.jmrix.ieee802154.xbee.XBeeAdapter@517566b0013A20040A20857 (FLLS S1 A1)
- Resetting the local module... [main]

2016-09-09 13:22:22,676 api.XBeeDevice                        INFO  -
jmri.jmrix.ieee802154.xbee.XBeeAdapter@517566b0013A20040A20857 (FLLS S1 A1)
- Module reset successfully. [main]

2016-09-09 13:22:24,682 xbee.XBeeNodeManager                  INFO  -
Starting XBee Node Discovery Process [main]

2016-09-09 13:22:24,766 xbee.XBeeNodeManager                  ERROR -
exception durring discovery process setup [main]

2016-09-09 13:22:26,317 configurexml.ConnectionConfigXml      ERROR -
Exception adding node 0013A20040AAB0C3 - FLLS S1 A2 from configuration file.
[main]

2016-09-09 13:22:27,781 configurexml.ConnectionConfigXml      ERROR -
Exception adding node 0013A20040B11528 - FLLS S1 A3 from configuration file.
[... 695 lines cut ...]
SerialEvent: DATA_AVAILABLE but no data available. [Thread-5]

2016-09-09 13:23:12,544 xbee.XBeeAdapter                      WARN  -
SerialEvent: DATA_AVAILABLE but no data available. [Thread-5]

2016-09-09 13:23:12,640 xbee.XBeeAdapter                      WARN  -
SerialEvent: DATA_AVAILABLE but no data available. [Thread-5]

2016-09-09 13:23:12,688 xbee.XBeeAdapter                      WARN  -
SerialEvent: DATA_AVAILABLE but no data available. [Thread-5]

2016-09-09 13:23:13,920 xbee.XBeeAdapter                      WARN  -
SerialEvent: DATA_AVAILABLE but no data available. [Thread-5]

2016-09-09 13:23:14,273 xbee.XBeeAdapter                      WARN  -
SerialEvent: DATA_AVAILABLE but no data available. [Thread-5]

2016-09-09 13:23:15,223 xbee.XBeeAdapter                      WARN  -
SerialEvent: DATA_AVAILABLE but no data available. [Thread-5]

 

So not sure what some of this means. I'll do some digging later.

-ken c

KenC57, 2016-09-09:
One thing I forgot to ask about, baud rates. So anyone used to these Xbee
devices speak up.

 

If I'm using only the bits on the radio nodes, how much impact does the baud
rate have?? I'm finding for my test, the interaction between to nodes and
the base JMRI running Logix to work a few signals and sensors, takes about 1
second from the sensor changing to the last of the signals (only two nodes)
changing.

 

I'm using the 9600 default. If I changed that I suspect it might help if a
lot of my time is spent in the buffers. That's 6 signal lamps and 4 sensors
(3/2 on a node). Or does that have a big impact on range and power
consumed??

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

KenC57, 2016-09-09:
I'm not sure I said this, but after getting lots of stuff in the console,
once the panel loaded, I had a working system.

 

Paul,

 

I still see the problem in the node config where I can select by 64 bit, and
the other two pulldowns don't change. Same for what selecting by node name,
the others don't change. I suspect some of it might be that for mesh the 16
bit address is the same for everything.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

pabender, 2016-09-09:
On Fri, Sep 9, 2016 at 2:16 PM, Ken Cameron <notifications@github.com>
wrote:

> I'm not sure I said this, but after getting lots of stuff in the console,
> once the panel loaded, I had a working system.
>
>
Great!

> I still see the problem in the node config where I can select by 64 bit,
> and
> the other two pulldowns don't change. Same for what selecting by node name,
> the others don't change. I suspect some of it might be that for mesh the 16
> bit address is the same for everything.
>

Ok.  I'll look into that further.  I think the node config tool is going to
get a redesign
as part of this whole effort.  There's going to be a single selection box
for the
nodes when I'm done.

Paul

pabender, 2016-09-16:
@KenC57 

When you get a chance, can you pull down a current copy of the branch.

I've redesigned large pieces of the node configuration tool.  It now has separate add and edit windows. and a single drop down list that shows the combination of  16 Bit Address, 64 Bit Address, and Node Identifier.

(Function wise, the Add, Edit and Delete buttons now actually behave correctly.  I'm not sure I ever got the Edit and Delete Buttons working before.)

pabender, 2016-09-21:
PR #1982 has the changes here staged to go into JMRI/JMRI master.

pabender, 2016-09-22:
closed via #1982

closed 2016-09-22 by pabender